t6x/reaver-wps-fork-t6x #2: merge request

d8tahead, 2015-04-14:
i have added -P option to loop through up to but not process M4 then start from beginning of the protocol then lopp and repeat. to avoid lockouts.

ive used that latest as of this current post

http://bit.ly/1I9kdYG

t6x, 2015-04-14:
hi, I'll take a look now

d8tahead, 2015-04-14:
Well it seems a commit was made when I had submitted it to you, someone else had also added a -P option.

If you are wanting to see me changes, 

grep -nrI pixie_loop

grep -nrI SEND_M4

Then in the switch info for P and such

And those will lead you to my respective changes

----- Reply message -----

&mdash;

d8tahead, 2015-04-14:
And if crediting, please credit to DataHead
And if you want to add me to the project master branch, add d8tahead on git

----- Reply message -----

d8tahead, 2015-04-14:
Together we can make this reaver the best 
And if you didn't know, I developed the pixie dust poc code back in November last year for ralink and broadcom eCos, and i am the one whom provided it to soxrok and wiire. wires poxiewps code is based off the one I provided them.

So if you want to add me to the project master branch, just thought I'd let you know a little about me and who I am

----- Reply message -----

t6x, 2015-04-14:
the option -P was triggered only in the wash yesterday

In reaver it is still available.

I'm already doing the merge

Not found the function
pixie_loop int (); that declared in globule.h

Can I ignore this declaration?

t6x, 2015-04-14:
Thanks for helping

That's what I wanted, everyone helping to better let the tool

Not needed even talk about the credits, I had already added before you comment

I think more than sure to have help in these projects, only one doing very overloads

I'm sorry the English, I'm using a translator, then ignore errors

d8tahead, 2015-04-14:
Should be find to remove . 
Good to hear, and I can make pretty good sense from the translation, so do not worry :-)

After pixie_loop () removed compile and check with -P it should not reach m4, and then it will loop.

----- Reply message -----

t6x, 2015-04-14:
merge is done

t6x, 2015-04-14:
what is your email or skype?

d8tahead, 2015-04-14:
Skype. feedbaddeadbeef

----- Reply message -----

t6x, 2015-04-15:
I'm trying to add you in this repository but giving error, do not know if now worked, you can check for me?

d8tahead, 2015-04-15:
Yes it works :-) thank you much!

----- Reply message -----

closed 2015-04-15 by d8tahead