t6x/reaver-wps-fork-t6x #4: No output

DarkJester, 2015-04-17:
new compile on kali gives no output old version of wash works fine

root@ikali:~/reaver-wps-fork-t6x/src# ./wash -i mon0 -s

Wash v1.5.1 WiFi Protected Setup Scan Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>
mod by t6_x <t6_x@hotmail.com>
mod by DataHead

BSSID                  Channel       RSSI       WPS Version       WPS Locked        ESSID
---------------------------------------------------------------------------------------------------------------
^C
root@ikali:~/reaver-wps-fork-t6x/src# ./wash -i mon0 

Wash v1.5.1 WiFi Protected Setup Scan Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>
mod by t6_x <t6_x@hotmail.com>
mod by DataHead

BSSID                  Channel       RSSI       WPS Version       WPS Locked        ESSID
---------------------------------------------------------------------------------------------------------------
^C

d8tahead, 2015-04-17:
Try with -C

DarkJester, 2015-04-17:
same result no output. i did get some warnings durring compile but i made sure it install all the deps first

d8tahead, 2015-04-17:
i've done a fresh compile and install with the lastest build. i can not reproduce this error. 

may i reccomend that you either do a restart of the system and put your device into monitor mode and run wash with the following: wash - i <mondevice> -C -s

if that fails, please re compile and make sure dependencies are correctly installed.

if the problem continues after so, please report back with as much info as possible.

-Cheers

DarkJester, 2015-04-17:
please see huge paste below;

root@ikali:~/reaver-wps-fork-t6x/src# apt-get install libpcap-dev aircrack-ng sqlite3 libsqlite3-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
aircrack-ng is already the newest version.
libpcap-dev is already the newest version.
libsqlite3-dev is already the newest version.
sqlite3 is already the newest version.
The following packages were automatically installed and are no longer required:
  libafpclient0 libcrypt-passwdmd5-perl libgadu3 libhackrf libmozjs24d libtsk3-3 linux-headers-3.14-kali1-amd64 linux-headers-3.14-kali1-common
  linux-headers-3.18.0-kali1-amd64 linux-headers-3.18.0-kali1-common linux-kbuild-3.14 python-flask python-itsdangerous python-werkzeug unrar-free xulrunner-24.0
Use 'apt-get autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@ikali:~/reaver-wps-fork-t6x/src# make clean
make: *** No rule to make target `clean'.  Stop.
root@ikali:~/reaver-wps-fork-t6x/src# ./configure
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for pcap_open_live in -lpcap... yes
checking for sqlite3_open in -lsqlite3... yes
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for stdlib.h... (cached) yes
checking for stdint.h... (cached) yes
checking for string.h... (cached) yes
checking pcap.h usability... yes
checking pcap.h presence... yes
checking for pcap.h... yes
checking sqlite3.h usability... yes
checking sqlite3.h presence... yes
checking for sqlite3.h... yes
configure: creating ./config.status
config.status: creating Makefile
root@ikali:~/reaver-wps-fork-t6x/src# make
(cd utils && make)
make[1]: Entering directory `/root/reaver-wps-fork-t6x/src/utils'
  CC  base64.c
common.c: In function ‘pingen_dlink’:
common.c:534:20: warning: variable ‘serial_len’ set but not used [-Wunused-but-set-variable]
common.c:534:9: warning: variable ‘mac_len’ set but not used [-Wunused-but-set-variable]
  CC  common.c
  CC  ip_addr.c
  CC  radiotap.c
  CC  trace.c
  CC  uuid.c
  CC  wpa_debug.c
  CC  wpabuf.c
  CC  os_unix.c
  CC  eloop.c
ar crT libutils.a base64.o common.o ip_addr.o radiotap.o trace.o uuid.o wpa_debug.o wpabuf.o os_unix.o eloop.o
[... 185 lines cut ...]
          TX packets:42 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:3059616 (2.9 MiB)  TX bytes:8276 (8.0 KiB)

root@ikali:~/reaver-wps-fork-t6x/src# ./wash -i wlan3 -C -s

Wash v1.5.1 WiFi Protected Setup Scan Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>
mod by t6_x <t6_x@hotmail.com>
mod by DataHead

BSSID                  Channel       RSSI       WPS Version       WPS Locked        ESSID
---------------------------------------------------------------------------------------------------------------
^C
root@ikali:~/reaver-wps-fork-t6x/src# uname -a
Linux ikali 3.12-kali1-amd64 #1 SMP Debian 3.12.6-2kali1 (2014-01-06) x86_64 GNU/Linux
root@ikali:~/reaver-wps-fork-t6x/src# ./wash -i wlan3 -C -s

Wash v1.5.1 WiFi Protected Setup Scan Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>
mod by t6_x <t6_x@hotmail.com>
mod by DataHead

BSSID                  Channel       RSSI       WPS Version       WPS Locked        ESSID
---------------------------------------------------------------------------------------------------------------

d8tahead, 2015-04-17:
are you doing this before using wash?
airmon-ng start wlan3

or

ifconfig wlan3 down && iwconfig wlan3 mode monitor && ifconfig wlan3 up

and for the sake of: airmon-ng start wlan3    bring down the wlan3 interface before hand
ifconfig wlan3 down

DarkJester, 2015-04-17:
i tried airmon-ng and the manual ifconfig way of putting the device in monitor mode i rebooted etc. the installed version of wash works fine. new version of reaver works just new wash is broken

root@ikali:~/reaver-wps-fork-t6x/src# wash -i wlan3

Wash v1.4 WiFi Protected Setup Scan Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>

BSSID                  Channel       RSSI       WPS Version       WPS Locked        ESSID
---------------------------------------------------------------------------------------------------------------
E0:91:F5:6F:D2:2A       1            -52        1.0               No                Artavia

t6x, 2015-04-17:
What system are you using? Kernel version and what wifi adapter?

Some processes may influence the airmon-ng, as wpa_supplicant and etc, try killall wpa_supplicant and verifies that the wash back to work.

Any application that uses the wifi driver influences the airmon

DarkJester, 2015-04-17:
all proccess that effect wifi killed.

see output below of kernel and kali version

root@ikali:~/reaver-wps-fork-t6x/src# uname -a
Linux ikali 3.12-kali1-amd64 #1 SMP Debian 3.12.6-2kali1 (2014-01-06) x86_64 GNU/Linux
root@ikali:~/reaver-wps-fork-t6x/src# cat /etc/issue
Kali GNU/Linux 1.1.0 \n \l

root@ikali:~/reaver-wps-fork-t6x/src# cat /etc/debian_version 
Kali Linux 1.1.0
root@ikali:~/reaver-wps-fork-t6x/src# 

WIFI adapter 
Bus 002 Device 003: ID 0bda:8187 Realtek Semiconductor Corp. RTL8187 Wireless Adapter

Wash v1.4 WiFi Protected Setup Scan Tool installed from Kali repo works fine,  New wash has no output

t6x, 2015-04-17:
The system is 64-bit may be the problem.
I tested here in a kali 64bit and everything worked normally, but the 64bit is very problematic with the aircrack-ng.
There are many problems with drivers that only compiling reaver not be resolved.
You could test with kali 32bit?

DarkJester, 2015-04-17:
I will install 32 bit and test

kcdtv, 2015-04-19:
@ DarkJester
"root@ikali:~/reaver-wps-fork-t6x/src# wash -i wlan3"
You have to use the interface IN mode monitor 
wash -i wlan3mon  /   wash -i mon3
@ t6x
"The system is 64-bit may be the problem."
No worries with this ;) I only use the last 64amd kali linux / Xubutnu version with rtl8187l and everything goes fine with both them

d8tahead, 2015-04-19:
64bit kali 1.1.0 and all working fine here aswel. Your system might be
broken in more than just the dependencies or there is improper installing
of the new reaver / wash. Instead of using wash directly from the build
directory, please run make install to ensure its placement on the OS is
where it should be.  and run it from global.

d8tahead, 2015-04-20:
I will also be doing an in-depth debug analysis of this issue in a day or two when i have a bit more time

d8tahead, 2015-04-20:
I may need you to participate in some installation of a few tools since i can not replicate this issue. I will be providing a special use case makefile for you in a day or two when i have the time, it will add compiling with debug and 01 or 00 flags and then will need you to install valgrind and do a valgrind --leak-check-full as a prefix to your segfault wash command

DarkJester, 2015-04-20:
This works on a pure 32bit system fresh install. My 64bit system is actually multi arch so maybe im pulling in something I shouldnt durrin compile time. also please note
@kcdtv 
""root@ikali:~/reaver-wps-fork-t6x/src# wash -i wlan3"
You have to use the interface IN mode monitor
wash -i wlan3mon / wash -i mon3"
My interface IS in monitor mode. doing ifconfig down; iwconfig wlan3 mode monitor; ifconfig up places the wlan3 in monitor mode but does not make a psuedo monX interface.
 @d8tahead 
Happy to run any tests you need.

crushedice2000, 2015-07-20:
I'm facing the same problem. I think that this happens because the latest version of `libpcap` is not supported by this reaver (this also happens with the original one)

closed 2015-12-24 by soxrok2212