t6x/reaver-wps-fork-t6x #5: Problem with  TP-LINK TD-W8961ND

Gitussef, 2015-04-18:
I tried this on TP-LINK TD-W8961ND (Ralink chipset:RT2860) and always stuck at M2 after getting the PKr, i think is has something to do with wps version because Wash -i mon0 doesn't show Wps is enabled on the router but other tools on wifislax like WpsScan show that wps is not locked.

t6x, 2015-04-18:
This is because the router was configured to work only as the WPS button, it will respond only when the button is pressedThis not is a standard configuration of the router
Date: Sat, 18 Apr 2015 03:33:23 -0700

Gitussef, 2015-04-18:
Ok thank you,is there a way to get PKe,AuthKey,E-Hash1 and E-Hash2 before M2?

soxrok2212, 2015-04-18:
PKE and Authkey yes, the rest no.

closed 2015-04-19 by soxrok2212