t6x/reaver-wps-fork-t6x #10: error when launching pixiewps:  "pixiewps: invalid option -- 'm'"

kcdtv, 2015-04-27:
[image]

i am using the last pixiewps version and just download the last revision of reaver.
Syntax used :
sudo reaver -i wlan2 -b F8:8E:85:40:84:95 -vv -K 1
stdout :
sudo reaver -i wlan2 -b F8:8E:85:40:84:95 -vv -K 1

Reaver v1.5.2 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>
mod by t6_x <t6_x@hotmail.com> & DataHead & Soxrok2212

[+] Waiting for beacon from F8:8E:85:40:84:95
[+] Switching wlan2 to channel 1
[+] Associated with F8:8E:85:40:84:95 (ESSID: MOVISTAR_8494)
[+] Starting Cracking Session. Pin count: 0, Max pin attempts: 11000
[+] Trying pin 12345670.
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[P] E-Nonce: 29:84:5c:2a:49:53:21:02:cc:69:9b:b5:a9:89:1e:71
[P] PKE: 44:c9:37:e1:12:eb:d0:c4:7c:6b:5a:6e:f0:63:5a:d1:30:04:cb:da:68:1c:92:37:7d:e9:9d:8c:18:7f:65:77:50:ec:66:1a:78:51:2d:98:9e:3f:3f:c2:21:7d:33:b6:01:42:d4:6c:87:01:f6:19:d9:25:52:d0:64:a3:ea:09:38:8a:61:a1:61:6b:af:0d:83:ea:4e:d0:79:f3:b5:6f:83:c0:d1:ff:48:02:0c:48:38:a1:38:d4:98:56:ae:28:fa:8c:23:ca:b6:c1:1a:7b:3e:91:df:c0:4c:3d:6b:ca:c1:02:6c:66:c8:7a:da:47:aa:8b:c8:92:de:a8:d6:e6:72:ab:84:81:11:75:8b:27:f7:33:09:c0:36:6c:7a:7f:3b:c0:f7:29:4e:d8:3d:38:a5:90:b7:3b:d0:5a:dc:b2:18:29:10:70:e8:0a:e4:17:3f:a2:67:12:8f:07:1c:39:58:cf:99:33:3b:58:cf:79:81:d4:32:f2:d6:ae:39:71
[P] WPS Manufacturer: Broadcom
[P] WPS Model Number: 123456
[P] Access Point Serial Number: 1234
[+] Received M1 message
[P] PKR: cf:03:23:93:62:de:32:20:59:6a:f9:3c:0e:3b:5d:89:18:8e:3c:d2:82:0f:c3:b2:3c:b4:2f:a4:f0:ef:4a:9c:04:bf:74:c5:ae:8c:a6:12:f8:b5:a2:e3:44:2f:4a:fe:cd:3e:26:52:5e:75:d9:6b:f1:d0:9a:f5:77:ed:35:33:09:a6:9d:3a:bd:8a:c1:7e:74:d2:89:d7:86:04:e4:27:c9:26:6d:28:aa:8d:88:75:1d:9a:45:de:d8:2e:af:69:6e:06:4c:6a:94:54:07:a6:66:19:7f:b0:75:cd:11:3c:71:70:da:45:0a:ca:f7:8d:8c:4a:f1:8c:df:10:aa:47:31:af:b3:44:02:8e:48:02:60:5c:31:f3:83:05:89:84:b4:99:05:b0:fe:b7:16:e0:e6:00:a1:4a:a5:ab:40:03:99:1b:36:b2:12:79:98:be:bd:3f:78:b4:77:44:c1:f0:79:5c:10:d0:ed:ad:23:c3:bf:11:08:0b:c4:97:8f:1a
[P] AuthKey: 60:ac:8e:2a:b3:f8:2b:34:fb:4e:3e:ae:06:c7:cd:62:18:bc:e9:07:e5:fa:47:9d:9c:dc:28:26:ea:c1:f9:c2
[+] Sending M2 message
[P] R-Nonce: 3b:ae:65:c2:68:bc:ad:63:89:55:8c:9d:9d:1c:6a:2f
[P] E-Hash1: 99:6b:8f:9e:db:8f:dc:b4:00:27:d3:3b:46:99:81:79:17:e7:d2:57:bf:12:42:ad:3e:e9:0f:5d:e6:0d:8c:75
[P] E-Hash2: 58:31:5c:d7:2d:0e:fa:a4:c2:85:2b:59:2f:71:38:b0:2d:b9:b0:e4:8e:17:3e:00:b4:e6:f5:ea:22:c2:e5:ad
pixiewps: invalid option -- 'm'
[Pixie-Dust]  
[Pixie-Dust]   Pixiewps made by wiire
[Pixie-Dust]  
[Pixie-Dust]   Usage: pixiewps <arguments>
[Pixie-Dust]  
[Pixie-Dust]   Required Arguments:
[Pixie-Dust]  
[Pixie-Dust]      -e, --pke      : Enrollee public key
[Pixie-Dust]      -r, --pkr      : Registrar public key
[Pixie-Dust]      -s, --e-hash1  : E-Hash1
[Pixie-Dust]      -z, --e-hash2  : E-Hash2
[Pixie-Dust]      -a, --authkey  : Key used in HMAC SHA-256
[Pixie-Dust]  
[Pixie-Dust]   Optional Arguments:
[Pixie-Dust]  
[Pixie-Dust]      -n, --e-nonce  : Enrollee nonce
[Pixie-Dust]      -S, --dh-small : Small Diffie-Hellman keys (--pkr not needed)
[Pixie-Dust]  
[Pixie-Dust]      -h, --help     : Display this usage screen
[Pixie-Dust]  

i didn't see anything round about -m option... maybe you are working with a unoficial development source of pisie wps that have this "m" option.
If not you might have forgotten a "m" somewhere where you create the syntax for pixiewps
Cheers

soxrok2212, 2015-04-27:
It is for use with pixiewps 1.1 which should be released very very soon :)

closed 2015-04-27 by soxrok2212

nuroo, 2015-04-29:
Im still getting the -m error. And also now falling all associations. Downloaded latest, fork today

soxrok2212, 2015-04-29:
did you remove the old version of reaver then make and make install the new one? are you cd'ing to the right src file?

nuroo, 2015-04-29:
sudo make install??

doesnt that remove old reaver? is there a new version number, 4 prelease?

soxrok2212, 2015-04-29:
It should, but maybe it didn't. navigate to /bin or /local/bin... something like that, and remove the old reaver.

d8tahead, 2015-04-29:
Cd to new reaver release fork src folder

If compiled that fork before, do make distclean first

Then
./configure && make && make install.

nuroo, 2015-04-29:
ok what is the command line for distclean

nuroo, 2015-04-29:
is there a different version number for the mod that fixes -m error? so i can tell

soxrok2212, 2015-04-29:
make distclean

nuroo, 2015-04-29:
ok from the fork directory?

soxrok2212, 2015-04-29:
yes

d8tahead, 2015-04-29:
while in src folder

make distclean

That's if you've compiled in that folder before.

d8tahead, 2015-04-29:
All good now?

nuroo, 2015-04-29:
nope.
tried to delete all versions 
redownloaded
compiled
make distclean
recompiled
same problems as b4
now when reaver ends, crashs
+] Switching wlan1mon to channel 6
[!] WARNING: Failed to associate with 000000000:6B:42 (ESSID: Ony0000)
[!] WARNING: Failed to associate with 000000000:6B:42 (ESSID:  Ony0000)
[!] WARNING: Failed to associate with 000000000:6B:42 (ESSID:  Ony0000)
[!] WARNING: Failed to associate with 000000000:6B:42 (ESSID:  Ony0000)
[+] Associated with E4:F4:C6:0B:6B:42 (ESSID: Onyx123)
[+] Starting Cracking Session. Pin count: 0, Max pin attempts: 11000
^C
[+] Nothing done, nothing to save.
*** glibc detected *** reaver: corrupted double-linked list: 0x00000000015ddcb0 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x75be6)[0x7f04b2890be6]
/lib/x86_64-linux-gnu/libc.so.6(+0x7601d)[0x7f04b289101d]
/lib/x86_64-linux-gnu/libc.so.6(+0x77518)[0x7f04b2892518]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x6c)[0x7f04b289598c]
reaver[0x408ecf]
/lib/x86_64-linux-gnu/libc.so.6(+0x321e0)[0x7f04b284d1e0]
/lib/x86_64-linux-gnu/libc.so.6(close+0x10)[0x7f04b28ea910]
/usr/lib/x86_64-linux-gnu/libpcap.so.0.8(+0xcb53)[0x7f04b34adb53]
/usr/lib/x86_64-linux-gnu/libpcap.so.0.8(+0x7024)[0x7f04b34a8024]
/usr/lib/x86_64-linux-gnu/libpcap.so.0.8(pcap_close+0x1e)[0x7f04b34aec6e]
reaver[0x407f56]
reaver[0x406006]
reaver[0x402fa8]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xfd)[0x7f04b2839ead]
reaver[0x403119]
======= Memory map: ========
00400000-00443000 r-xp 00000000 00:12 428850                             /usr/local/bin/reaver
00643000-0066f000 rw-p 00043000 00:12 428850                             /usr/local/bin/reaver
0066f000-00673000 rw-p 00000000 00:00 0 
015b6000-01650000 rw-p 00000000 00:00 0                                  [heap]
7f04ac000000-7f04ac021000 rw-p 00000000 00:00 0 
7f04ac021000-7f04b0000000 ---p 00000000 00:00 0 
7f04b23e9000-7f04b23fe000 r-xp 00000000 00:12 3428                       /lib/x86_64-linux-gnu/libgcc_s.so.1
7f04b23fe000-7f04b25fe000 ---p 00015000 00:12 3428                       /lib/x86_64-linux-gnu/libgcc_s.so.1
7f04b25fe000-7f04b25ff000 rw-p 00015000 00:12 3428                       /lib/x86_64-linux-gnu/libgcc_s.so.1
7f04b25ff000-7f04b2616000 r-xp 00000000 00:12 2031                       /lib/x86_64-linux-gnu/libpthread-2.13.so
7f04b2616000-7f04b2815000 ---p 00017000 00:12 2031                       /lib/x86_64-linux-gnu/libpthread-2.13.so
7f04b2815000-7f04b2816000 r--p 00016000 00:12 2031                       /lib/x86_64-linux-gnu/libpthread-2.13.so
7f04b2816000-7f04b2817000 rw-p 00017000 00:12 2031                       /lib/x86_64-linux-gnu/libpthread-2.13.so
7f04b2817000-7f04b281b000 rw-p 00000000 00:00 0 
7f04b281b000-7f04b299c000 r-xp 00000000 00:12 1878                       /lib/x86_64-linux-gnu/libc-2.13.so
7f04b299c000-7f04b2b9c000 ---p 00181000 00:12 1878                       /lib/x86_64-linux-gnu/libc-2.13.so
7f04b2b9c000-7f04b2ba0000 r--p 00181000 00:12 1878                       /lib/x86_64-linux-gnu/libc-2.13.so
7f04b2ba0000-7f04b2ba1000 rw-p 00185000 00:12 1878                       /lib/x86_64-linux-gnu/libc-2.13.so
7f04b2ba1000-7f04b2ba6000 rw-p 00000000 00:00 0 
7f04b2ba6000-7f04b2c50000 r-xp 00000000 00:12 12588                      /usr/lib/x86_64-linux-gnu/libsqlite3.so.0.8.6
7f04b2c50000-7f04b2e50000 ---p 000aa000 00:12 12588                      /usr/lib/x86_64-linux-gnu/libsqlite3.so.0.8.6
7f04b2e50000-7f04b2e52000 r--p 000aa000 00:12 12588                      /usr/lib/x86_64-linux-gnu/libsqlite3.so.0.8.6
7f04b2e52000-7f04b2e54000 rw-p 000ac000 00:12 12588                      /usr/lib/x86_64-linux-gnu/libsqlite3.so.0.8.6
7f04b2e54000-7f04b2e55000 rw-p 00000000 00:00 0 
7f04b2e55000-7f04b2ed6000 r-xp 00000000 00:12 2794                       /lib/x86_64-linux-gnu/libm-2.13.so
7f04b2ed6000-7f04b30d5000 ---p 00081000 00:12 2794                       /lib/x86_64-linux-gnu/libm-2.13.so
7f04b30d5000-7f04b30d6000 r--p 00080000 00:12 2794                       /lib/x86_64-linux-gnu/libm-2.13.so
7f04b30d6000-7f04b30d7000 rw-p 00081000 00:12 2794                       /lib/x86_64-linux-gnu/libm-2.13.so
7f04b30d7000-7f04b30d9000 r-xp 00000000 00:12 1880                       /lib/x86_64-linux-gnu/libdl-2.13.so
7f04b30d9000-7f04b32d9000 ---p 00002000 00:12 1880                       /lib/x86_64-linux-gnu/libdl-2.13.so
7f04b32d9000-7f04b32da000 r--p 00002000 00:12 1880                       /lib/x86_64-linux-gnu/libdl-2.13.so
7f04b32da000-7f04b32db000 rw-p 00003000 00:12 1880                       /lib/x86_64-linux-gnu/libdl-2.13.so
7f04b32db000-7f04b32fb000 r-xp 00000000 00:12 1874                       /lib/x86_64-linux-gnu/ld-2.13.so
7f04b349d000-7f04b34a1000 rw-p 00000000 00:00 0 
7f04b34a1000-7f04b34d9000 r-xp 00000000 00:12 3408                       /usr/lib/x86_64-linux-gnu/libpcap.so.1.3.0
7f04b34d9000-7f04b34db000 r--p 00037000 00:12 3408                       /usr/lib/x86_64-linux-gnu/libpcap.so.1.3.0
7f04b34db000-7f04b34dc000 rw-p 00039000 00:12 3408                       /usr/lib/x86_64-linux-gnu/libpcap.so.1.3.0
7f04b34dc000-7f04b34de000 rw-p 00000000 00:00 0 
7f04b34f8000-7f04b34fa000 rw-p 00000000 00:00 0 
7f04b34fa000-7f04b34fb000 r--p 0001f000 00:12 1874                       /lib/x86_64-linux-gnu/ld-2.13.so
7f04b34fb000-7f04b34fc000 rw-p 00020000 00:12 1874                       /lib/x86_64-linux-gnu/ld-2.13.so
7f04b34fc000-7f04b34fd000 rw-p 00000000 00:00 0 
7fff5f577000-7fff5f598000 rw-p 00000000 00:00 0                          [stack]
7fff5f599000-7fff5f59b000 r--p 00000000 00:00 0                          [vvar]
7fff5f59b000-7fff5f59d000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
Aborted

d8tahead, 2015-04-29:
-s y is required, please try the latest updated as of now

d8tahead, 2015-04-29:
Recompile as before and make install

nuroo, 2015-04-29:
ok did it
something is up with assoication code.......any changes?
getting
[!] WARNING: Failed to associate with 00000000:A7:86 (ESSID: 00000000a784)
[!] WARNING: Failed to associate with 00000000:A7:86 (ESSID: 00000000a784)
but router is close in distance
so try
reaver -i wlan1mon -b 00000000:A7:86 -m 0000000:78:6a -vv -S -K1 -s y -t15
[!] WARNING: Failed to associate with 00000000:A7:86 (ESSID: 00000000a784)

if aireplay-ng does assoication it works, no -m error (router not vulernable)
root@kali:~# reaver -i wlan1mon -b 00000000:A7:86 -m 0000000:78:6a -vv -S -K1 -s y -t15 -A

Reaver v1.5.2 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>
mod by t6_x <t6_x@hotmail.com> & DataHead & Soxrok2212

[+] Waiting for beacon from 00000000:A7:86
[+] Switching wlan1mon to channel 6
[+] Associated with 00000000:A7:86 (ESSID: 00000000a784)
[+] Starting Cracking Session. Pin count: 0, Max pin attempts: 11000
[+] Trying pin 12345670.
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[P] E-Nonce: 29:4e:21:03:5a:88:27:f1:e1:d8:6c:ee:09:93:af:5d
[P] PKE: 38:72:fe:0c:a0:24:21:3c:3b:c4:05:d8:fe:93:99:24:aa:93:b7:4c:f7:13:e7:3e:f7:f0:28:e6:f0:de:64:6d:44:2b:35:44:ce:75:0f:88:19:85:1a:fb:9f:90:9a:3d:0b:71:ab:10:80:57:3e:68:79:9a:1a:33:a2:eb:a6:bb:bd:a6:f9:40:85:5e:aa:9c:18:26:31:3b:e9:73:76:55:f0:f5:db:a7:e0:a2:f7:e0:1d:ad:a7:93:87:6f:59:60:34:b1:a7:0f:91:30:1a:49:6a:27:11:7f:f7:eb:a6:29:cc:cc:04:16:9f:26:67:09:61:7d:6e:fa:57:b7:f1:6a:27:63:ba:36:12:34:68:db:92:15:77:f4:e7:a0:78:49:25:87:90:0a:58:19:09:a2:2f:72:e4:f4:7d:d3:b0:4a:1a:f5:e7:41:09:8c:2f:4c:82:81:47:17:8b:02:6b:cd:8d:10:91:b6:78:df:07:19:fe:26:fc:c6:31:49:70:79
[P] WPS Manufacturer: D-Link
[P] WPS Model Name: Verizon Platform
[P] WPS Model Number: DSL-2750B
[P] Access Point Serial Number: PX2M1E5003500
[+] Received M1 message
[P] R-Nonce: 69:61:40:eb:25:4c:d4:a5:e4:ad:27:1b:b9:46:54:36
[P] PKR: 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:02
[P] AuthKey: 31:4d:91:68:32:23:55:68:c3:23:a7:80:24:ee:04:1e:d5:be:82:5b:a7:b8:ab:0e:b8:89:71:c6:72:ae:0d:93

d8tahead, 2015-04-29:
As far as i can tell, no change has been made with association, just
changes in the display of rnonce and model display.

nuroo, 2015-04-29:
ok cool. thanks 4 help. I can still use aireplay-ng......if im only 1 having the issue