t6x/reaver-wps-fork-t6x #13: if received m5 after sent m6,rever can't crack the AP's password

linjy87, 2015-05-18:
hi,
    When I cracked the AP's pin, i could not crack the ap's password when the reaver received m5 after it sent m6. because the wps->state is not RECV_M7 as below. could you help me to solve it.

static enum wps_process_res wps_process_m7(struct wps_data *wps,
        const struct wpabuf *msg,
        struct wps_parse_attr *attr)
{
    struct wpabuf *decrypted;
    struct wps_parse_attr eattr;

    wpa_printf(MSG_DEBUG, "WPS: Received M7");

    if (wps->state != RECV_M7) {
        wpa_printf(MSG_DEBUG, "WPS: Unexpected state (%d) for "
                "receiving M7", wps->state);
        wps->state = SEND_WSC_NACK;
        return WPS_CONTINUE;
    }

t6x, 2015-05-18:
A protocol should be followed.

After sending the M6 protocol demands that you receive the M7.

If you are getting the M5 after sending the M6, then you have a problem with the protocol.

If something like this is happening, it may be that you are too far from the router.

soxrok2212, 2015-05-18:
You can also try adding the M5:M7 timeout to reaver... Add -T 5 to your command.

closed 2015-05-20 by t6x