t6x/reaver-wps-fork-t6x #15: Constant failed to associate

DrDinosaur, 2015-05-30:
I'm getting a "failed to associate with" message every time I run this tool. It's happened with every AP. I use wash to identify WPS APs. Since it's happened with every AP across multiple channels, I think there is an issue. 
Here's the command I try: 
./reaver -i wlan1mon -K 1 -c x -b xx:xx:xx:xx:xx:xx
Adding "-vv" adds no additional helpful information to the output. 
This attack is demonstrated here:
https://www.kali.org/penetration-testing/pixiewps-reaver-aircrack-ng-updates/
It's what I'm trying to replicate.

d8tahead, 2015-05-30:
This is an ongoing known ( to me atleast ) issue that i have yet to figure out if its:
A.) A problem / feature with new aircrack-ng rc2, as that version seems to break reavers funtionality for some reason.

Or
B.) An issue with reaver that collides with either updated or downgraded or incompatible unstable dependencies which interfere with rc2 of aircrack.

Now, i know you didnt state that if you have aircracks new rc2 build or not, but from what you describe, is a symptom of this. 

Also, if you did install aircrack rc2, did you happen to compile it with any extra or unstable experimental flags?

I been looking into code changes to reflect this, but bug reports and symptoms and install steps are scarce

DrDinosaur, 2015-05-30:
I'm on the latest version (all packages) of Kali. 

aircrack-ng --help
Aircrack-ng 1.2 rc2 - (C) 2006-2014 Thomas d'Otreppe

t6x, 2015-05-31:
Some wifi adapters are having problems with the new version of aircrack.

You can try to downgrade the aircrack and test again? So we can begin to identify where is the problem

d8tahead, 2015-05-31:
Also, i've found it important to install aircrack first then reaver second. seems to be a temporary fix for some

DrDinosaur, 2015-06-01:
Okay so I downgraded.

aircrack-ng --help
Aircrack-ng 1.2 rc1 - (C) 2006-2014 Thomas d'Otreppe

Then I complied reaver. I got one successful authentication; it spit out some information about nonce and authkeys. Then it failed to associate again. Same with other APs.

Costanza369, 2015-06-28:
I just ran into with this problem. Using a git checkout revision from 30 days ago seems to have fixed my issue.

Lokke, 2015-07-24:
any updates? - i am using aircrack-ng 1.2rc2-1

soxrok2212, 2015-07-26:
There is nothing we can do about aircrack. You could install a previous version from here: http://archive.aircrack-ng.org/aircrack-ng/ but its not recommended, you don't want to stay on legacy software. Hopefully there will be an update soon that fixes a lot of the issues that have come from the latest update.

closed 2015-07-26 by soxrok2212