t6x/reaver-wps-fork-t6x #20: Instrallation Issue

naflan121, 2015-06-05:
I follow up the instruction to install, Everything Was fine but still reaver is not updated, Any checklists?

soxrok2212, 2015-06-07:
Which operating system are you using?

naflan121, 2015-06-07:
im using kali linux

beddj, 2015-06-07:
Is this for the installation? I'm good. It didn't work at first but I
restarted the terminal and it worked. But if this is for the netgear router
I'm using Kali Linux

soxrok2212, 2015-06-07:
Navigate to /usr/local/bin and remove pixiewps, reaver, and wash if it is in there. The easiest way to install is just through apt. First, just to be sure, remove any existing versions: apt-get remove reaver and apt-get remove pixiewps.

Then, apt-get install reaver and it should add pixiewps in automatically as well. If it doesnt. apt-get install pixiewps.

Close terminal and reopen then everything should work fine and it will automatically update every time you run apt-get update && apt-get upgrade

naflan121, 2015-06-07:
oki thanks ill try and leave the feedbacks

closed 2015-07-26 by soxrok2212