t6x/reaver-wps-fork-t6x #21: OSX installation

fachos, 2015-06-07:
hi... i try install under OSX Yosemite... but error coming..
can you guide me? thanks

"
35 warnings generated.
cc  -I../utils -I ../ wps_ufd.c -c
gmake[1]: Leaving directory '/Users/XXX/Downloads/reaver-wps-fork-t6x-master/src/wps'
(cd lwe && gmake BUILD_STATIC=y libiw.a)
gmake[1]: Entering directory '/Users/XXX/Downloads/reaver-wps-fork-t6x-master/src/lwe'
cp wireless.21.h wireless.h
gcc -Os -W -Wall -Wstrict-prototypes -Wmissing-prototypes -Wshadow -Wpointer-arith -Wcast-qual -Winline -I. -MMD     -fPIC -c -o iwlib.so iwlib.c
In file included from iwlib.c:14:
./iwlib.h:54:10: fatal error: 'linux/types.h' file not found
#include <linux/types.h>                /* for "caddr_t" et al          */
         ^
1 error generated.
Makefile:120: recipe for target 'iwlib.so' failed
gmake[1]: *** [iwlib.so] Error 1
gmake[1]: Leaving directory '/Users/XXX/Downloads/reaver-wps-fork-t6x-master/src/lwe'
Makefile:34: recipe for target 'libiw' failed
gmake: *** [libiw] Error 2
"

soxrok2212, 2015-06-07:
Reaver is a Linux tool, not an OS X tool. I will not work on OS X.

fachos, 2015-06-07:
hi thank you for fast reply, i've try reaver 1.4 run well as it be in os x (macports). it port to os x enviroments. yesterday i compile pixiewps to os x successfully but without reaver 1.5.2 it just useless.
can't get any PKR, PKE, HASH1, HASH2 etc... i hope there will be os x support...

soxrok2212, 2015-06-07:
Were you able to get your wireless card in monitor mode?

fachos, 2015-06-07:
In linux usually it mon0... in os x en0... i test standard version 1.4 with pin 12345670 it take less than 30 minute. That for old dlink router or 24hour with nothing.. Been play in kali lately too with 1.5.2... less than 5 minute.. much more easier to crack... if u want check the source 1.4 for os x they porting here the link "https://docs.google.com/file/d/0B3tHnfnS08GycG1fVENETEdvazA/edit"

thanks..

fachos, 2015-06-07:
In os x don't need airmon-ng start wlan0... does'nt work that way... just en0

soxrok2212, 2015-06-07:
I see, someone made a port from Linux to OS X. This version was made from the original Linux version, not the OS X ported version. Until someone else ports it for OS X, it will not work correctly.

fachos, 2015-06-07:
:+1:  i saw they patch called *.diff the original file so it work under os x before compile... can you guys resolve this? because pixiewps is ready for os x, i sure others are waiting to see 1.5.2 with os x environments :)

closed 2015-06-07 by soxrok2212