t6x/reaver-wps-fork-t6x #22: can't compile reaver ubuntu 14.04

sheikhmunawar, 2015-06-26:
root@Lappi:/home/sheikh/Desktop/temp/reaver-wps-fork-t6x/src# make
(cd utils && make)
make[1]: Entering directory `/home/sheikh/Desktop/temp/reaver-wps-fork-t6x/src/utils'
  CC  base64.c
common.c: In function ‘pingen_zhaochunsheng’:
common.c:597:5: warning: passing argument 1 of ‘snprintf’ makes pointer from integer without a cast [enabled by default]
     snprintf(pin, pin_len, "%08d", (default_pin * 10) + wps_checksum(default_pin));
     ^
In file included from /usr/include/features.h:374:0,
                 from /usr/include/stdlib.h:24,
                 from includes.h:25,
                 from common.c:15:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:61:1: note: expected ‘char * __restrict__’ but argument is of type ‘int’
 __NTH (snprintf (char *__restrict __s, size_t __n,
 ^
common.c: In function ‘pingen_zyxel’:
common.c:614:5: warning: implicit declaration of function ‘wps_pin_checksum’ [-Wimplicit-function-declaration]
     return (pin * 10) + wps_pin_checksum(pin);
     ^
  CC  common.c
  CC  ip_addr.c
  CC  radiotap.c
  CC  trace.c
  CC  uuid.c
  CC  wpa_debug.c
  CC  wpabuf.c
  CC  os_unix.c
  CC  eloop.c
ar crT libutils.a base64.o common.o ip_addr.o radiotap.o trace.o uuid.o wpa_debug.o wpabuf.o os_unix.o eloop.o
make[1]: Leaving directory `/home/sheikh/Desktop/temp/reaver-wps-fork-t6x/src/utils'
(cd tls && make)
make[1]: Entering directory `/home/sheikh/Desktop/temp/reaver-wps-fork-t6x/src/tls'
  CC  asn1.c
In file included from bignum.c:21:0:
libtommath.c: In function ‘mp_exptmod’:
libtommath.c:580:9: warning: variable ‘dr’ set but not used [-Wunused-but-set-variable]
     int dr;
         ^
  CC  bignum.c
  CC  pkcs1.c
  CC  pkcs5.c
  CC  pkcs8.c
  CC  rsa.c
  CC  tlsv1_client.c
  CC  tlsv1_client_read.c
  CC  tlsv1_client_write.c
  CC  tlsv1_common.c
  CC  tlsv1_cred.c
  CC  tlsv1_record.c
  CC  tlsv1_server.c
tlsv1_server_read.c: In function ‘tls_process_client_key_exchange_rsa’:
tlsv1_server_read.c:474:9: warning: variable ‘encr_len’ set but not used [-Wunused-but-set-variable]
     u16 encr_len;
         ^
  CC  tlsv1_server_read.c
  CC  tlsv1_server_write.c
  CC  x509v3.c
ar crT libtls.a asn1.o bignum.o pkcs1.o pkcs5.o pkcs8.o rsa.o tlsv1_client.o tlsv1_client_read.o tlsv1_client_write.o tlsv1_common.o tlsv1_cred.o tlsv1_record.o tlsv1_server.o tlsv1_server_read.o tlsv1_server_write.o x509v3.o
make[1]: Leaving directory `/home/sheikh/Desktop/temp/reaver-wps-fork-t6x/src/tls'
(cd crypto && make)
make[1]: Entering directory `/home/sheikh/Desktop/temp/reaver-wps-fork-t6x/src/crypto'
  CC  aes-cbc.c
  CC  aes-ctr.c
  CC  aes-eax.c
  CC  aes-encblock.c
  CC  aes-internal.c
  CC  aes-internal-dec.c
  CC  aes-internal-enc.c
  CC  aes-omac1.c
  CC  aes-unwrap.c
[... 228 lines cut ...]
wpsmon.c:427:7: warning: pointer targets in passing argument 2 of ‘strcat’ differ in signedness [-Wpointer-sign]
       strcat(cmd_chipset, mac2str(get_bssid(),':'));
       ^
In file included from /usr/include/features.h:374:0,
                 from /usr/include/ctype.h:25,
                 from /usr/include/getopt.h:33,
                 from wpsmon.h:37,
                 from wpsmon.c:34:
/usr/include/x86_64-linux-gnu/bits/string3.h:140:1: note: expected ‘const char * __restrict__’ but argument is of type ‘unsigned char *’
 __NTH (strcat (char *__restrict __dest, const char *__restrict __src))
 ^
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Ilibwps sigint.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 sigalrm.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 keys.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 pins.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 send.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Ilibwps exchange.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Ilibwps session.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Ilibwps cracker.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Ilibwps wpscrack.c *.o libwps/*.o wps/*.o tls/bignum.o tls/libtls.a utils/libutils.a crypto/libcrypto.a lwe/libiw.a -ldl -lm -lpcap -lsqlite3  -o reaver
wpscrack.c: In function ‘main’:
wpscrack.c:40:33: warning: variable ‘r’ set but not used [-Wunused-but-set-variable]
     int ret_val = EXIT_FAILURE, r = 0;
                                 ^
root@Lappi:/home/sheikh/Desktop/temp/reaver-wps-fork-t6x/src#

soxrok2212, 2015-07-26:
You need to run ./configure, then make, then make install.

closed 2015-07-26 by soxrok2212

ifytbairz, 2015-07-27:
hello soxrok2212,

i have this same issue however I am following the readme exactly. i have the prerequisites, i run ./configure, make but make always fails compiling wpscrack. 

please advise. running fresh linux mint 17.2 64bit

tomasjarolim, 2017-04-15:
edit wpscrack.c file and remove variable

- on line 38 was something like 
`int ret_val = EXIT_FAILURE, r=...;`
->
`int ret_val = EXIT_FAILURE;`

- on line 118 starting a condition on few lines
`/* Run user-supplied command */`
->
remove all condition

I'm not sure what is user-supplied command, but it work for me