t6x/reaver-wps-fork-t6x #24: is wpacrack.c supposed to fail ?

ifytbairz, 2015-07-02:
I'm trying to compile this code, during 'make' i always get this error:

gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Ilibwps wpscrack.c *.o libwps/*.o wps/*.o tls/bignum.o tls/libtls.a utils/libutils.a crypto/libcrypto.a lwe/libiw.a -ldl -lm -lpcap -lsqlite3  -o reaver
wpscrack.c: In function ‘main’:
wpscrack.c:40:33: warning: variable ‘r’ set but not used [-Wunused-but-set-variable]
     int ret_val = EXIT_FAILURE, r = 0;
                                 ^
because of this, i don't think wpsmon.c ever gets a chance to compile. i've also noticed that 'wash' doesn't work correctly either. 

sudo ./wash -i wlan1mon

Wash v1.5.2 WiFi Protected Setup Scan Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>
mod by t6_x <t6_x@hotmail.com> & DataHead & Soxrok2212 & Wiire & kib0rg

[X] ERROR: Failed to open 'wlan1mon' for capturing

my wlan1 IS in monitor mode, I can run airodump with no problems. interestingly, after running wlan1mon in airodump, wash doesn't present the previously noted error, but there is no output at all:

sudo ./wash -i wlan1mon

Wash v1.5.2 WiFi Protected Setup Scan Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>
mod by t6_x <t6_x@hotmail.com> & DataHead & Soxrok2212 & Wiire & kib0rg

BSSID              Channel  RSSI  WPS Version  WPS Locked  ESSID
--------------------------------------------------------------------------------------

hope this helps.

closed 2015-07-26 by soxrok2212

chrisonline909, 2015-08-14:
hey guys im getting the same error.. any help would be appreciated 
thank you!

kcdtv, 2015-08-16:
where do you see an "error"? This is a "warning". Your variable is set but not used. That's not a problem. if you can compile and install = don't worry be happy

chrisonline909, 2015-08-17:
your right it was not an error, everythings works great! thanks

tomasjarolim, 2017-04-15:
when still cannot make,
edit wpscrack.c file and remove variable

- on line 38 was something like 
`int ret_val = EXIT_FAILURE, r=...;`
->
`int ret_val = EXIT_FAILURE;`

- on line 118 starting a condition on few lines
`/* Run user-supplied command */`
->
remove all condition

I'm not sure what is user-supplied command, but it work for me