t6x/reaver-wps-fork-t6x #25: reaver-wps-fork will not attempt Pixie or default Pin attacks regardless of parameters passed to it

thesle3p, 2015-07-07:
I have noticed since the most recent push to git your fork of Reaver ignores -K 1 and -W and attempts a full brute force of WPS. I have tested this on two 64-bit Debian Laptops with two different Atheros cards.

kcdtv, 2015-07-08:
Now you need to use -vvv (3 v) to have the "full" stdout with the hashes for pixiedust
-K might have been correctly executed with no positive results but - maybe - that was simply not stdouted at all. 
Could you use -vvv instead of -vv to check that out?

d8tahead, 2015-07-08:
I would assume that would be the solution to their problem, thanks kcdtv

thesle3p, 2015-07-08:
same thing with -vvv it just brutes wps does not send hashes at all it
goes through the standard Reaver brute force. -W does not work either
Here is the args I am passing to reaver sudo reaver -i wlan1mon -b
<BSSID of target> -vvv -K 1 (or -W)  -A -c <channel number> and instead
of the Pixie loop or the default pin brute force it tries to dumb brute
the pin

thesle3p, 2015-08-20:
Can someone take a look at this? This was not the case until I updated to the latest version of reaver-wps-fork.

kcdtv, 2015-08-21:
not before the 27 as i am not home, but i never had this issue. Maybe is due to the use of -A? Are yo correctly associated, could you post a sample of reaver stdout. cheers 
:8ball:  this stupid emotiocone are realy the fucking best shit  :last_quarter_moon:

crushedice2000, 2015-08-23:
@kcdtv: Instead of rant against emoticons,  please put your code between backquotes. In example:

If I write `:8ball:` I'll get this: :8ball: 
If I write `` `:8ball:` `` I'll get this: `:8ball:`

Please read markdown guide [here](https://guides.github.com/features/mastering-markdown/)

soxrok2212, 2015-12-24:
Be sure you are not trying to use both commands at once. Using -W will give you possible PINs, and then you have to manually supply them to Reaver in order for it to work. Using both at once would probably confuse it...

soxrok2212, 2015-12-24:
Try removing Reaver and pixiewps and reinstall them both, ensuring you have all of the dependencies. (Keep in mind that pixiewps has been updated to 1.2, and no longer requires libssl-dev). Not sure if this is the problem, but it has not been reported anywhere else so it might have just been an install/dependency issue.

closed 2015-12-24 by soxrok2212