t6x/reaver-wps-fork-t6x #26: Seems like it does not fork from the latest version of original reaver ?

alex-chan, 2015-07-10:
After I compiled it to OpenWRT, run `reaver -i mon0 -b CC:34:29:3F:XX:XX` It says:

    [+] Waiting for beacon from CC:34:29:3F:48:AE
    Segmentation fault

Then I take a look at source of file : argsparser.c
It's not as new as https://code.google.com/p/reaver-wps/issues/detail?id=120 said.

alex-chan, 2015-07-10:
Update : there is a slight difference between the patch and the forked one.

d8tahead, 2015-07-10:
Hi Alex-Chan, I have this running on my openwrt based router, can you tell me a little but more about the compilation steps you took? And also, what hardware are you using / trying to put it on?

netforhack, 2015-07-10:
@d8tahead can you make package and publish it ?

d8tahead, 2015-07-10:
Yes I can, and there is one stipulation. It's tested / working on ar71xx
mips B.E., openwrt AA

netforhack, 2015-07-10:
@d8tahead can you build for Barier Breaker release ? Hopes it`s cound run on tp-link mr3020

d8tahead, 2015-07-10:
https://github.com/t6x/reaver-wps-fork-t6x/tree/Big_endian/bin/ar71xx/ipk

d8tahead, 2015-07-10:
i have not personally tested it on Barrier Breaker, but from last i heard, reaver is not working on it. i will have to further investigate the issue. maybe if someone would want to compile it with use against valgrind and, run, and send the output logs here from where and why it segfaults

ZeroChaos-, 2015-07-10:
It may be more useful to provide the makefile you used instead.  if it passes quality control it can be included in openwrt proper.

d8tahead, 2015-07-10:
the make file is included

d8tahead, 2015-07-10:
and has been included, as here https://github.com/t6x/reaver-wps-fork-t6x/blob/Big_endian/Makefile

it is to be placed in feeds, and someone had stated that i should be the maintainer for the openwrt based reaver source, and i am fine with that.

netforhack, 2015-07-10:
so test, debug if needed and pull reqest to openwrt packages repo 

@d8tahead i can try to run it, but is there no wps ap in my area

netforhack, 2015-07-10:
@d8tahead on wr1043nd BB i`ve got error "reaver: can't load library 'libpcap.so.1.1' but libpcap installed

d8tahead, 2015-07-10:
you need an older version of libpcap, there are known issues with newer libpcap

d8tahead, 2015-07-10:
@ZeroChaos- If you would like to submit any of this onward to the openwrt repo, feel free. also pixiewps could be added too

netforhack, 2015-07-10:
@d8tahead reaver from repo works fine, i will try on other device downgraded to AA

d8tahead, 2015-07-10:
i will look at the sources / patches for BB and see what is different for the libs

alex-chan, 2015-07-11:
My router is Mercury MAC1200R, the chip is AR9344, However OpenWRT use AR71xx catalog. So I downloaded http://downloads.openwrt.org/snapshots/trunk/ar71xx/  (build7.10) to flash my router. 

And downloaded the relative SDK to compile this reaver( master branch) 

My Makefile for OpenWRT is :

<pre><code>
include $(TOPDIR)/rules.mk

PKG_NAME:=reaver
PKG_RELEASE:=1.5.2czz

PKG_BUILD_DIR := $(BUILD_DIR)/$(PKG_NAME)

include $(INCLUDE_DIR)/package.mk

define Package/reaver
	SECTION:=packages
	CATEGORY:=Utilities
	TITLE:=reaver -- A wireless attack tool
	DEPENDS:=+libpcap +libsqlite3
	#DESCRIPTION:=\
	#If you can't figure out what this program does, \\\
	#you're probably brain-dead and need immediate \\\
	#medical attention.
endef

define Package/reaver/description
	A wireless attack tool. recompiled by sunset 2015/07
endef

define Build/Prepare
	mkdir -p $(PKG_BUILD_DIR)
	$(CP) ./src/* $(PKG_BUILD_DIR)/
endef

define Package/reaver/install
	$(INSTALL_DIR) $(1)/bin
	$(INSTALL_BIN) $(PKG_BUILD_DIR)/reaver $(1)/bin/
endef

$(eval $(call BuildPackage,reaver))
</pre></code>

d8tahead, 2015-07-11:
does the ilk here work for you? https://github.com/t6x/reaver-wps-fork-t6x/tree/Big_endian/bin/ar71xx/ipk

Or if you would like to manually have a go at compiling, place this makefile in /feeds/packages/net/reaver ( preferably an empty reaver directory )
https://github.com/t6x/reaver-wps-fork-t6x/blob/Big_endian/Makefile

d8tahead, 2015-07-11:
Also, You need to grab the Big_endian branch instead of the master

alex-chan, 2015-07-11:
@d8tahead 

1. First I use`opkg install  <the-ipkg-you-build>`,  It doesn't work. 

    root@OpenWrt:~# ls -al /usr/bin | grep reaver
    -rwxr-xr-x    1 root     root        347272 Jun 12 22:18 reaver
    root@OpenWrt:~# /usr/bin/reaver
   -ash: /usr/bin/reaver: not found
Seems like it's not the right SDK compiled for my OpenWRT

2. Then I use the Makefile https://raw.githubusercontent.com/t6x/reaver-wps-fork-t6x/Big_endian/Makefile . which download live code from github. But It still fails: undefined u_char in libwps.h . It said .

3. At last I downloaded the BigEndian brach myself and solve the undefined problem. 
modify the Makefile to  use local copy to build. It shows errors: 

<pre><code>
mips-openwrt-linux-musl-gcc -DCONF_DIR='"/etc/reaver"' -DREAVER_DATABASE='"/etc/reaver/reaver.db"' -Wall -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/sunset/Projects/Router/SDK/OpenWrt-SDK-ar71xx-generic_gcc-4.8-linaro_musl-1.1.10.Linux-x86_64/staging_dir/target-mips_34kc_musl-1.1.10/usr/include -I/home/sunset/Projects/Router/SDK/OpenWrt-SDK-ar71xx-generic_gcc-4.8-linaro_musl-1.1.10.Linux-x86_64/staging_dir/target-mips_34kc_musl-1.1.10/include -I/home/sunset/Projects/Router/SDK/OpenWrt-SDK-ar71xx-generic_gcc-4.8-linaro_musl-1.1.10.Linux-x86_64/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.10/usr/include -I/home/sunset/Projects/Router/SDK/OpenWrt-SDK-ar71xx-generic_gcc-4.8-linaro_musl-1.1.10.Linux-x86_64/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.10/include/fortify -I/home/sunset/Projects/Router/SDK/OpenWrt-SDK-ar71xx-generic_gcc-4.8-linaro_musl-1.1.10.Linux-x86_64/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_musl-1.1.10/include  builder.c -c
{standard input}: Assembler messages:
{standard input}:92: Error: unrecognized opcode `wsbh $2,$2'
{standard input}:103: Error: unrecognized opcode `wsbh $3,$17'
{standard input}:278: Error: unrecognized opcode `wsbh $2,$2'
make[4]: *** [builder.o] Error 1
</code></pre>

I still work on how to overcome  it.

chron0, 2015-07-16:
Same issue on current trunk/CC(RC) configured for a Gl.Inet (ar71xx)

Placed the Makefile

```
  1 #
  2 # Copyright (C) 2012 OpenWrt.org
  3 #
  4 # This is free software, licensed under the GNU General Public License v2.
  5 # See /LICENSE for more information.
  6 #
  7 
  8 include $(TOPDIR)/rules.mk
  9 
 10 PKG_NAME:=reaver-wps-fork-t6x
 11 PKG_REV:=Big_endian
 12 PKG_VERSION:=$(PKG_REV)
 13 PKG_RELEASE:=1
 14 
 15 PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.gz
 16 PKG_SOURCE_URL:=https://github.com/t6x/reaver-wps-fork-t6x.git
 17 PKG_SOURCE_SUBDIR:=$(PKG_NAME)-$(PKG_VERSION)
 18 PKG_SOURCE_VERSION:=$(PKG_REV)
 19 PKG_SOURCE_PROTO:=git
 20 
 21 include $(INCLUDE_DIR)/package.mk
 22 
 23 define Package/reaver
 24   SECTION:=net
 25   CATEGORY:=Network
 26   SUBMENU:=wireless
 27   TITLE:=Efficient online and offline brute force attack against Wifi Protected Setup
 28   URL:=https://github.com/t6x/reaver-wps-fork-t6x
 29   DEPENDS:=+libpcap +libsqlite3
 30 endef
 31 
 32 define Package/reaver/description
 33   Reaver targets the external registrar functionality mandated by the WiFi
 34   Protected Setup specification.
 35   Access points will provide authenticated registrars with their current
 36   wireless configuration (including the WPA PSK), and also accept a new
 37   configuration from the registrar.
 38   New implementations consist of the PixieDust attack and many more.
 39 endef
 40 
 41 CONFIGURE_PATH:=src
 42 
 43 MAKE_PATH:=src
 44 
 45 TARGET_CFLAGS+=$(TARGET_CPPFLAGS)
 46 
 47 define Package/reaver/install
 48     $(INSTALL_DIR) $(1)/usr/bin
 49     $(INSTALL_BIN) $(PKG_BUILD_DIR)/src/{reaver,wash} $(1)/usr/bin/
 50     $(INSTALL_DIR) $(1)/etc/reaver
 51     $(INSTALL_DATA) $(PKG_BUILD_DIR)/src/reaver.db $(1)/etc/reaver/
 52 endef
 53 
 54 $(eval $(call BuildPackage,reaver))
```
in feeds/packages/net/reaver and ran 

    # scripts/feeds install pixiewps
    # scripts/feeds install reaver

Then enabled reaver via make menuconfig

```
make[3]: Entering directory '/home/chrono/src/openwrt/feeds/packages/net/reaver'
. /home/chrono/src/openwrt/include/shell.sh; gzip -dc /home/chrono/src/openwrt/dl/reaver-wps-fork-t6x-Big_endian.tar.gz | tar -C /home/chrono/src/openwrt/build_dir/target-mips_34kc_musl-1.1.10/reaver-wps-fork-t6x-Big_endian/.. -xf - 
[... 448 lines cut ...]
         return (u_char *) FAKE_RADIO_TAP_HEADER;
                         ^
libwps.c:302:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
Makefile:5: recipe for target 'libwps.o' failed
make[5]: *** [libwps.o] Error 1
make[5]: Leaving directory '/home/chrono/src/openwrt/build_dir/target-mips_34kc_musl-1.1.10/reaver-wps-fork-t6x-Big_endian/src/libwps'
Makefile:19: recipe for target 'libwps.o' failed
make[4]: *** [libwps.o] Error 2
make[4]: Leaving directory '/home/chrono/src/openwrt/build_dir/target-mips_34kc_musl-1.1.10/reaver-wps-fork-t6x-Big_endian/src'
Makefile:54: recipe for target '/home/chrono/src/openwrt/build_dir/target-mips_34kc_musl-1.1.10/reaver-wps-fork-t6x-Big_endian/.built' failed
make[3]: *** [/home/chrono/src/openwrt/build_dir/target-mips_34kc_musl-1.1.10/reaver-wps-fork-t6x-Big_endian/.built] Error 2
make[3]: Leaving directory '/home/chrono/src/openwrt/feeds/packages/net/reaver'
package/Makefile:191: recipe for target 'package/feeds/packages/reaver/compile' failed
make[2]: *** [package/feeds/packages/reaver/compile] Error 2
make[2]: Leaving directory '/home/chrono/src/openwrt'
package/Makefile:188: recipe for target '/home/chrono/src/openwrt/staging_dir/target-mips_34kc_musl-1.1.10/stamp/.package_compile' failed
make[1]: *** [/home/chrono/src/openwrt/staging_dir/target-mips_34kc_musl-1.1.10/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/chrono/src/openwrt'
/home/chrono/src/openwrt/include/toplevel.mk:181: recipe for target 'world' failed
make: *** [world] Error 2
```

I guess it's also the libpcap missmatch? It would be great if we could get a working Makefile/Patchset for openwrt feeds for trunk/CC, since it's the perfect platform to play with and deploy wireless related stuff on.

crushedice2000, 2015-07-20:
**Why** the latest version of libpcap is not working with reaver? Is there any fix for that? Can I downgrade the library?

I'm using this OpenWRT version (as the banner says):

    OpenWrt Barrier Breaker 14.07

psyborg55, 2016-10-23:
was able to run it together with pixiewps, tried several APs some got locked and ralink ones turned off wps after about 10 tries. there were many segfaults even with libpcap 1.3

>     [  1423.197749] do_page_fault(): sending SIGSEGV to reaver for invalid read access from 00000004
    [  1423.206432] epc = 0041acf0 in reaver[400000+49000]
    [  1423.211373] ra  = 00407460 in reaver[400000+49000]
    [  1423.216299] 
    [  1438.806961] 
    [  1438.806961] do_page_fault(): sending SIGSEGV to reaver for invalid write access to 00000000
    [  1438.815616] epc = 777b62b0 in libc.so[7778e000+91000]
    [  1438.820907] ra  = 7775a0f3 in libpcap.so.1.3.0[77756000+36000]
    [  1438.826936] 
    [  1446.398974] 
    [  1446.398974] do_page_fault(): sending SIGSEGV to reaver for invalid write access to 00000008
    [  1446.407571] epc = 77ae5e20 in libc.so[77abe000+91000]
    [  1446.412787] ra  = 77ae6c8c in libc.so[77abe000+91000]
    [  1446.417973] 
    [  1451.210009] 
    [  1451.210009] do_page_fault(): sending SIGSEGV to reaver for invalid write access to 00000008
    [  1451.218613] epc = 77a2be20 in libc.so[77a04000+91000]
    [  1451.223849] ra  = 77a2cc8c in libc.so[77a04000+91000]
    [  1451.229013] 
    [  1461.230711] 
    [  1461.230711] do_page_fault(): sending SIGSEGV to reaver for invalid read access from 00000000
    [  1461.239409] epc = 7708dadf in libpcap.so.1.3.0[7708a000+36000]
    [  1461.245634] ra  = 7708e495 in libpcap.so.1.3.0[7708a000+36000]
    [  1461.251644] 
    [  1485.280480] 
    [  1485.280480] do_page_fault(): sending SIGSEGV to reaver for invalid write access to 00000008
    [  1485.289058] epc = 77ef5e20 in libc.so[77ece000+91000]
    [  1485.294263] ra  = 77ef6c8c in libc.so[77ece000+91000]
    [  1485.299427] 
    [  1498.262818] 
    [  1498.262818] do_page_fault(): sending SIGSEGV to reaver for invalid write access to 00000008
    [  1498.271444] epc = 77493e20 in libc.so[7746c000+91000]
    [  1498.276669] ra  = 77494c8c in libc.so[7746c000+91000]
    [  1498.282367] 
    [  1506.140389] 
    [  1506.140389] do_page_fault(): sending SIGSEGV to reaver for invalid write access to 00000008
    [  1506.148967] epc = 776a1e20 in libc.so[7767a000+91000]
    [  1506.154152] ra  = 776a2120 in libc.so[7767a000+91000]
    [  1506.159354] 
    [  1612.802557] 
    [  1612.802557] do_page_fault(): sending SIGSEGV to reaver for invalid write access to 00000008
    [  1612.811182] epc = 774e9e20 in libc.so[774c2000+91000]
    [  1612.816410] ra  = 774eac8c in libc.so[774c2000+91000]
    [  1612.821600] 
    [  1636.746282] 
    [  1636.746282] do_page_fault(): sending SIGSEGV to reaver for invalid write access to 00000000
    [  1636.754911] epc = 77ab42b0 in libc.so[77a8c000+91000]
    [  1636.760099] ra  = 77a580f3 in libpcap.so.1.3.0[77a54000+36000]
    [  1636.766107]

rofl0r, 2017-06-21:
this is probably fixed with release 1.6b.

closed 2017-06-21 by rofl0r