t6x/reaver-wps-fork-t6x #27: wash displays no results

jugganuts420, 2015-07-11:
while using wash it displays no results program runs but works/fails

output as follows

blacktop reaver-wps-fork-t6x # wash -i mon0

Wash v1.5.2 WiFi Protected Setup Scan Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>
mod by t6_x <t6_x@hotmail.com> & DataHead & Soxrok2212 & Wiire & kib0rg

BSSID              Channel  RSSI  WPS Version  WPS Locked  ESSID
--------------------------------------------------------------------------------------

is there a debug option to show more info...

airmon-ng and airodump-ng works flawlessly

currently using a internal network adaptor (chipset-Atheros    driver-ath9k) and external usb network adapter (tp-link722n) on linux mint.

it worked on previously a couple months ago.

let me know if there is anything i can do to help :)

crushedice2000, 2015-07-20:
Me too! It can be a so recent version of libpcap

RED4911, 2015-08-02:
With me also. How to fix?

kcdtv, 2015-08-04:
"Me too! It can be a so recent version of libpcap"
Probably. I don't have linux Mint to "reproduce the error" but it has been an issue on ubutnu (and mint is derivate from ubuntu) 
You should try by installing "manually" libcap0.8-dev <url>http://www.ubuntuupdates.org/package/core/vivid/main/base/libpcap0.8-dev<url/>
Whish will slightly break your dependency sistem (everyupdate tit will be detected as "outdated" and removed )

mikob, 2015-11-02:
@kcdtv awesome. I had this issue on linux mint 17.2. Uninstalling libpcap and libpcap-dev from the apt repos and installing the packages from where you linked to did the trick. Thanks!

closed 2015-11-05 by soxrok2212