t6x/reaver-wps-fork-t6x #28: pkr problem

rsozdance, 2015-08-02:
hello,
i have tried attacking different wifies and using two different laptops. but it just keep looping forever! without any result!

i run reaver with this command:
reaver -b c0:4a:00:XX:XX:XX -vv -S -i wlan1mon -K 1

some of the output:
[+] Sending M2 message
[+] Received M1 message
[+] Sending WSC NACK
[+] Sending WSC NACK
[!] WPS transaction failed (code: 0x03), re-trying last pin
[+] Trying pin 12345670.
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[P] E-Nonce: 28:85:8d:dc:3b:c9:bf:3a:59:2c:a7:bf:b8:e4:a9:93[P] PKE: 26:e9:4f:b5:96:65:ac:f8:3f:1f:7d:11:ac:ce:39:c2:24:b6:bf:62:dd:0a:98:c2:a4:44:12:e9:dd:f0:a4:0b:60:a2:0b:0a:ee:f2:2a:fd:00:e8:fc:3d:f5:c1:2d:70:f4:2c:d5:31:ff:9e:82:43:7a:ab:5b:f0:13:96:59:79:92:89:39:f3:c7:f0:6b:30:1b:84:f4:b6:52:76:5d:92:ec:47:79:14:68:ad:8b:1f:4f:5d:ef:1e:c0:e2:c4:61:68:73:b0:f8:47:d8:c2:b4:92:39:d2:a5:b3:7b:e6:63:3d:81:07:9d:31:7e:f1:24:ed:86:b9:3a:37:99:d8:7e:f8:27:0e:ca:6e:e5:fc:fe:e4:9d:f3:b2:93:a6:89:a1:63:ec:6e:52:e7:f8:1b:6c:97:9a:3c:67:aa:8a:78:e2:9d:9e:40:65:be:23:7c:ce:35:4b:38:4c:d7:8e:08:11:0b:b7:71:a7:11:67:17:57:09:ab:a6:4e:4f:16:53:67
[P] WPS Manufacturer: TP-LINK
[P] WPS Model Name: TL-WR841N
[P] WPS Model Number: 9.0
[P] Access Point Serial Number: 1.0
[+] Received M1 message
[P] R-Nonce: 45:94:c7:6c:7a:2b:15:91:69:90:f8:63:39:37:03:9b[P] PKR: 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:02
[P] AuthKey: 53:01:fe:ca:97:dd:e6:56:8f:0e:06:1f:cb:39:28:3a:d9:30:f3:4f:b1:c0:51:22:9d:6a:d3:22:f3:32:ee:a6

kcdtv, 2015-08-04:
First you should tell us what is your OS; what is your wifi chipset and what is the model and chipset of your acess point
This device has an atehros chipset and pixiewps does not support atheros chipset (so don't use -K1) 
https://wikidevi.com/wiki/TP-LINK_TL-WR841ND_v9.x
Looking at your stdout it seems that you are too far from the AP or that your chipset is crap for reaver.
Or that you did not enable properly WPS on your acess point. 
If your attacking any kind of router around you cannot know what is going on so make the test in a correct distance on your own AP with WPS fuly enabled
This messages of error are "correct" and there is not realy here matter to open an issue.

closed 2015-08-07 by rsozdance

wiire, 2015-08-18:
You get:

`[P] PKR: 00: [...] :02`

when using -S (--dh-small) in Reaver >= 1.3.