t6x/reaver-wps-fork-t6x #29: SQL error on Wash

DrDinosaur, 2015-08-02:
root@kali:~/reaver-wps-fork-t6x/src# ./wash
[X] ERROR: sql_init failed

I'm on the Raspberry Pi, if that matters. 

I did ./configure and then make. 

Reaver can run without error.

kcdtv, 2015-08-02:
* That's not the way to use wash, you have to do:
[code]
(sudo) wash -i <mode_monitor_interface>
[/code]

[quote]I'm on the Raspberry Pi, if that matters. [/quote]
  * The operative system matters too ;)

[quote]
[code]
root@kali:~/reaver-wps-fork-t6x/src# ./wash
[X] ERROR: sql_init failed
[/code]
[/quote]
  * Did you installed all the dependencies? 
check this
[code]
sudo su && apt-get install  libssl-dev &&  apt-get install libnl-3-dev && apt-get install sqlite3 && apt-get install libsqlite3-dev && apt-get install libnl-genl-3-dev
[/code]

DrDinosaur, 2015-08-03:
I'm root, so I don't need to use sudo. 

I tried that, but it still does not work. All depenecies are available.

kcdtv, 2015-08-04:
"I'm root, so I don't need to use sudo" 
That's why i put it between parenthesis :)  
Could you first tell us more about your OS and put here the stdout of 
<code>uname -a</code>
"I tried that, but it still does not work"
Be more precise please
Did you launched wash as i told you instead of how you did it before?.
The error is the same? 
By the way : 
"I did ./configure and then make."
you should do <code>./configure && make && make install</code> 
You just compiled it, you didn't actually  installed it if you didn't finished the sequence with <code>(sudo) make install</code>

closed 2015-12-24 by soxrok2212