t6x/reaver-wps-fork-t6x #33: [Question] Using Reaver v1.4 for Pixie Dust Attack

GeorgeGkas, 2015-08-16:
Is it possible to use reaver v1.4 (default version found in Ubuntu repositories) to generate the info required to run pixie dust attack? I mean not to use the automate `-K ` process ? 
Is it possible? And what commands should I use in reaver v1.4 to find the infos ?

kcdtv, 2015-08-16:
It is not possible because you will miss the "authkey".
Also notice that you would need to capture traffic and use something like wireshark to extract the other datas (PKE, PKR etc...)
I use Xubuntu  as everyday system since ages and i never used the repositories from ubuntu for aircrack-ng or reaver
You should really use "the svn way"
cheers 
PD: I close the topic because there is no possibility to use reaver 1.4 for pixie dust, unless you make the changes ... that have been done in reaver 1.5.2 :p

closed 2015-08-16 by kcdtv

soxrok2212, 2015-08-16:
You can calculate it with pixiewps, but you have to use -S (small dh-keys)

kcdtv, 2015-08-17:
Upsss.... That's true i forgot about this nice feature introduced by wiire... It is hot here :palm_tree: and my brain melt down.
But yeah, reaver 1.4 doesn't shows anything and you will have to check by yourself in the pcap 
cheers

wiire, 2015-08-18:
@GeorgeGkas 
Yes, but you have to gather manually all the data from a .cap manually (or Wireshark). Use the last example pixiewps shows:

```
pixiewps -e <pke> -s <e-hash1> -z  <e-hash2> -n <e-nonce> -m <r-nonce> -b <e-bssid> -S
```

From the pixiewps [README](https://github.com/wiire/pixiewps/blob/master/README.md):

```
    -a, --authkey

        Registration Protocol authentication session key. Although for this parameter a
        modified version of Reaver or Bully is needed, it can be avoided by specifying
        small Diffie-Hellman keys in both Reaver and Pixiewps and supplying --e-nonce,
        --r-nonce and --e-bssid.
```
There's also written where to find that data.

Note that -S works currently only with *2/3 chipsets* so in general it's better to have a modified version of Reaver/Bully wich prints all the data needed, including `--authkey`.