t6x/reaver-wps-fork-t6x #36: Compile warnings

nask0, 2015-08-31:
Here is the log, on Fedora 22 :
```
$ ./configure
--
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for pcap_open_live in -lpcap... yes
checking for sqlite3_open in -lsqlite3... yes
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for stdlib.h... (cached) yes
checking for stdint.h... (cached) yes
checking for string.h... (cached) yes
checking pcap.h usability... yes
checking pcap.h presence... yes
checking for pcap.h... yes
checking sqlite3.h usability... yes
checking sqlite3.h presence... yes
checking for sqlite3.h... yes
configure: creating ./config.status
config.status: creating Makefile

$ make -j 12
--
(cd utils && make)
(cd tls && make)
(cd lwe && make BUILD_STATIC=y libiw.a)
(cd libwps && make)
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 globule.c -c 
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 init.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Ilibwps sql.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 crc.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 keys.c -c
make[1]: Entering directory '/mnt/004A0B094A0AFAE4/reaver-wps-fork-t6x/src/lwe'
make[1]: Entering directory '/mnt/004A0B094A0AFAE4/reaver-wps-fork-t6x/src/utils'
make[1]: Entering directory '/mnt/004A0B094A0AFAE4/reaver-wps-fork-t6x/src/libwps'
gcc -Wall libwps.c -c
make[1]: Entering directory '/mnt/004A0B094A0AFAE4/reaver-wps-fork-t6x/src/tls'
cp wireless.21.h wireless.h
gcc -Os -W -Wall -Wstrict-prototypes -Wmissing-prototypes -Wshadow -Wpointer-arith -Wcast-qual -Winline -I. -MMD     -fPIC -c -o iwlib.so iwlib.c
libwps.c: In function ‘parse_wps_parameters’:
libwps.c:29:12: warning: variable ‘data_len’ set but not used [-Wunused-but-set-variable]
     size_t data_len = 0, offset = 0;
            ^
common.c: In function ‘pingen_zhaochunsheng’:
common.c:597:14: warning: passing argument 1 of ‘snprintf’ makes pointer from integer without a cast [-Wint-conversion]
     snprintf(pin, pin_len, "%08d", (default_pin * 10) + wps_checksum(default_pin));
              ^
In file included from includes.h:26:0,
                 from common.c:15:
/usr/include/stdio.h:386:12: note: expected ‘char * restrict’ but argument is of type ‘int’
 extern int snprintf (char *__restrict __s, size_t __maxlen,
[... 381 lines cut ...]
                 from wps_dev_attr.c:15:
../utils/common.h:448:5: note: expected ‘char *’ but argument is of type ‘int’
 int pingen_dlink(char *mac, int add);        
     ^
wps_dev_attr.c:364:45: warning: implicit declaration of function ‘pingen_zyxel’ [-Wimplicit-function-declaration]
   printf("[Pin Gen] Pin Generated : %08d\n",pingen_zyxel(mac2str(get_bssid(),'\0'), 0));
                                             ^
make[1]: Leaving directory '/mnt/004A0B094A0AFAE4/reaver-wps-fork-t6x/src/wps'
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Ilibwps wpsmon.c *.o libwps/*.o wps/*.o tls/bignum.o tls/libtls.a utils/libutils.a crypto/libcrypto.a lwe/libiw.a -ldl -lm -lpcap -lsqlite3  -o wash
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Ilibwps wpscrack.c *.o libwps/*.o wps/*.o tls/bignum.o tls/libtls.a utils/libutils.a crypto/libcrypto.a lwe/libiw.a -ldl -lm -lpcap -lsqlite3  -o reaver
wpscrack.c: In function ‘main’:
wpscrack.c:40:33: warning: variable ‘r’ set but not used [-Wunused-but-set-variable]
     int ret_val = EXIT_FAILURE, r = 0;
                                 ^
wpsmon.c: In function ‘parse_wps_settings’:
wpsmon.c:438:27: warning: pointer targets in passing argument 2 of ‘strcat’ differ in signedness [-Wpointer-sign]
       strcat(cmd_chipset, mac2str(get_bssid(),':'));
                           ^
In file included from libwps/libwps.h:18:0,
                 from wpsmon.h:41,
                 from wpsmon.c:34:
/usr/include/string.h:137:14: note: expected ‘const char * restrict’ but argument is of type ‘unsigned char *’
 extern char *strcat (char *__restrict __dest, const char *__restrict __src)
              ^
```

crushedice2000, 2015-09-05:
You shouldn't be worried, these warnings will not affect the program
usability, however would be nice to foix them to improve source code
readability.

kcdtv, 2016-08-14:
I close this issue as it is a duplicate with #52

closed 2016-08-14 by kcdtv