t6x/reaver-wps-fork-t6x #37: Speeding up cracking [feature request]

crushedice2000, 2015-09-06:
Hi! Would be interesting to ship a database-like file with contents like the following with the t6x reaver:

    (11:22:33) {Zhao; Pixie; Markov}
    (11:B3:3F) {Zhao; Pixie; 12345670; D-Link; Markov}
    (FA:CE:00) {Zhao; Pixie; 1234567-1235567; Markov}
    (C0:FF:EE) {Zhao; Pixie; 1234567,0102030,1122334; Markov}
    (DE:FE:CA) {Zhao; Pixie; 1234567,0102030,1122334; Markov}
    (BA:DB:07) {Decimal($MAC[0:6])%10000000}

soxrok2212, 2015-11-05:
The problem with this is that different models of APs may come with the same first half of the MAC address - people will get confused when one listed as a vulnerable MAC address doesn't work.

crushedice2000, 2015-11-20:
@soxrok2212 What about adding more data to the database? i.e. some routers broadcast model.

Also, in the worst case, you could say "This AP **seems** to be vulnerable to %s algorythm"

kcdtv, 2017-09-26:
I personally think that this should be done with an external script using the -j wash option. 
Doing and maintaining a world list like this is a 24hour job position, 
With the -j option and the pipe you can easily use a personal script to check on the fly all the wps extended information with a database relevant to your local situation.

kcdtv, 2018-02-23:
Having a "**bssid**" data base approach is **simply wrong**. The correct way is to analyze first the extended WPS parameters contained in the Probe Answers. This should be done with externals scripts as the **-j** option from wash was specially designed to give a online complete stdout for each PA detected, making it very simple to automate the parsing of the probes.  I think it is time to close this issue opened in 2015 as the new option from wash is our solution to adress it.

closed 2018-02-23 by kcdtv