t6x/reaver-wps-fork-t6x #39: reaver problem

christiangnoth, 2015-11-07:
hi,
I have tried reaver 1.4 and now 1.5.2 and I am have problems to get it working. 

attached the generted output

can anyone help

regrads
christian

christiangnoth, 2015-11-07:
[+] Switching wlo1 to channel 11
[+] Waiting for beacon from 34:31:C4:51:93:DC
[+] Associated with 34:31:C4:51:93:DC (ESSID: FRITZ!Box 7312)
[+] Starting Cracking Session. Pin count: 10000, Max pin attempts: 11000
[+] Trying pin 46278197.
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[!] WARNING: Receive timeout occurred
[+] Sending WSC NACK
[!] WPS transaction failed (code: 0x02), re-trying last pin
[+] Trying pin 46278197.
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[!] WARNING: Receive timeout occurred
[+] Sending WSC NACK
[!] WPS transaction failed (code: 0x02), re-trying last pin
[+] Trying pin 46278197.
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[!] WARNING: Receive timeout occurred
[+] Sending WSC NACK
[!] WPS transaction failed (code: 0x02), re-trying last pin
[+] Trying pin 46278197.
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[!] WARNING: Receive timeout occurred
[+] Sending WSC NACK
[!] WPS transaction failed (code: 0x02), re-trying last pin
[+] Trying pin 46278197.
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[+] Received identity request
[+] Sending identity response
[!] WARNING: Receive timeout occurred
[+] Sending WSC NACK
[!] WPS transaction failed (code: 0x02), re-trying last pin
[+] Trying pin 46278197.
[... 2161 lines cut ...]
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[!] WARNING: 25 successive start failures
[+] Sending EAPOL START request
[+] Sending WSC NACK
[!] WPS transaction failed (code: 0x02), re-trying last pin
[+] Nothing done, nothing to save.

binarymaster, 2015-11-17:
Signal quality, or sensitivity of your wireless adapter is not enough to send M2 message.

kcdtv, 2015-11-20:
Hello christiangnoth
You have to understand that github is a code related website.
You don't have problems to "get reaver working" you have problem to lead an attack on a particular access point in a specific context.
This is not a subject to be treated on git hub. You should visit a forum like the one of Kali linux to open a threat to get help.
cheers

closed 2015-11-20 by kcdtv