t6x/reaver-wps-fork-t6x #40: New airmon-ng behavior breaks reaver

Marcool04, 2015-11-20:
Hello.
Just to report that the new version of airmon-ng, which when called against, say for instance wlan0, doesn't create "mon0" but "wlan0mon", and what's more, refuses to create several monitor interfaces, seriously breaks reaver.
This checks out on my end with Reaver 1.5.2, airmon-ng 1.2-rc2, included with Kali Linux 2.
Not too sure how this can be fixed other than by rolling back airmon-ng to the previous version. Will take a closer look at the reaver code and see.
Regards,
Mark.

soxrok2212, 2015-11-20:
Reaver only uses 1 interface. I have had no issues on both HDD installs and VM installs of Kali 2.0 with both v1.5.2 or Reaver and the Aircrack-ng 1.2-rc2 suite. Do you have any error messages? Have you ran airmon-ng check kill before starting? Most wireless problems are due to a conflict between Network Manager and Airmon-ng, running airmon-ng check kill usually fixes it for me.

d8tahead, 2015-11-20:
Hi sox, i know this error and It has always seemed an order of installation that's caused this error for me.

Be sure you completely remove aircrack and reaver, then install aircrack and its dependencies first then install reaver and dependencies.

Marcool04, 2015-11-21:
Oh man! My bad... sorry it's not reaver that is broken, it is a completely different project, ReVdK3... a script that binds reaver and mdk3 to help with WPS bruteforcing when an AP locks up.
I just got the wrong site to report the error :/
Will go and get a strong cup of coffee now...

closed 2015-11-21 by Marcool04