t6x/reaver-wps-fork-t6x #41: Openwrt Mips Compiling Error

well0nez, 2015-11-22:
Hi, i tried to compile this reaver fork with the latest openwrt trunk.

I got the following Problem:

make[5]: Entering directory '/opt/openwrt/build_dir/target-mips_34kc_musl-1.1.11/reaver-wps-fork-t6x-Big_endian/src/utils'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/opt/openwrt/build_dir/target-mips_34kc_musl-1.1.11/reaver-wps-fork-t6x-Big_endian/src/utils'
(cd tls && make)
make[5]: Entering directory '/opt/openwrt/build_dir/target-mips_34kc_musl-1.1.11/reaver-wps-fork-t6x-Big_endian/src/tls'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/opt/openwrt/build_dir/target-mips_34kc_musl-1.1.11/reaver-wps-fork-t6x-Big_endian/src/tls'
(cd crypto && make)
make[5]: Entering directory '/opt/openwrt/build_dir/target-mips_34kc_musl-1.1.11/reaver-wps-fork-t6x-Big_endian/src/crypto'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/opt/openwrt/build_dir/target-mips_34kc_musl-1.1.11/reaver-wps-fork-t6x-Big_endian/src/crypto'
(cd wps && make)
make[5]: Entering directory '/opt/openwrt/build_dir/target-mips_34kc_musl-1.1.11/reaver-wps-fork-t6x-Big_endian/src/wps'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/opt/openwrt/build_dir/target-mips_34kc_musl-1.1.11/reaver-wps-fork-t6x-Big_endian/src/wps'
(cd lwe && make BUILD_STATIC=y libiw.a)
make[5]: Entering directory '/opt/openwrt/build_dir/target-mips_34kc_musl-1.1.11/reaver-wps-fork-t6x-Big_endian/src/lwe'
make[5]: 'libiw.a' is up to date.
make[5]: Leaving directory '/opt/openwrt/build_dir/target-mips_34kc_musl-1.1.11/reaver-wps-fork-t6x-Big_endian/src/lwe'
(cd libwps && make)
make[5]: Entering directory '/opt/openwrt/build_dir/target-mips_34kc_musl-1.1.11/reaver-wps-fork-t6x-Big_endian/src/libwps'
mips-openwrt-linux-musl-gcc -Wall libwps.c -c
cc1: note: someone does not honour COPTS correctly, passed 0 times
In file included from libwps.c:14:0:
libwps.h:50:32: error: unknown type name 'u_char'
 int parse_wps_parameters(const u_char *packet, size_t len, struct libwps_data *wps);
                                ^
libwps.h:129:25: error: unknown type name 'u_char'
 int parse_wps_tag(const u_char *tags, size_t len, struct libwps_data *wps);
                         ^
libwps.h:130:35: error: unknown type name 'u_char'
 unsigned char *get_wps_data(const u_char *data, size_t len, size_t *tag_len);
                                   ^
libwps.h:131:43: error: unknown type name 'u_char'
 unsigned char *get_wps_data_element(const u_char *data, size_t len, uint16_t type, size_t *el_len);
                                           ^
libwps.h:132:32: error: unknown type name 'u_char'
 int libwps_has_rt_header(const u_char *packet, size_t len);
                                ^
libwps.h:133:7: error: unknown type name 'u_char'
 const u_char *libwps_radio_header(const u_char *packet, size_t len);
       ^
libwps.h:133:41: error: unknown type name 'u_char'
 const u_char *libwps_radio_header(const u_char *packet, size_t len);
                                         ^
libwps.c:26:32: error: unknown type name 'u_char'
 int parse_wps_parameters(const u_char *packet, size_t len, struct libwps_data *wps)
                                ^
libwps.c: In function 'parse_wps_parameters':
libwps.c:28:11: error: unknown type name 'u_char'
     const u_char *data = NULL;
           ^
libwps.c:29:12: warning: variable 'data_len' set but not used [-Wunused-but-set-variable]
     size_t data_len = 0, offset = 0;
            ^
libwps.c: At top level:
libwps.c:55:25: error: unknown type name 'u_char'
 int parse_wps_tag(const u_char *tags, size_t len, struct libwps_data *wps)
                         ^
libwps.c: In function 'parse_wps_tag':
libwps.c:89:39: warning: passing argument 1 of 'get_wps_data_element' from incompatible pointer type [-Wincompatible-pointer-types]
             el = get_wps_data_element(wps_ie_data, wps_data_len, elements[i], &el_len);
                                       ^
In file included from libwps.c:14:0:
libwps.h:131:16: note: expected 'const int *' but argument is of type 'unsigned char *'
 unsigned char *get_wps_data_element(const u_char *data, size_t len, uint16_t type, size_t *el_len);
                ^
libwps.c: At top level:
libwps.c:190:35: error: unknown type name 'u_char'
 unsigned char *get_wps_data(const u_char *data, size_t len, size_t *tag_len)
                                   ^
libwps.c:234:43: error: unknown type name 'u_char'
 unsigned char *get_wps_data_element(const u_char *data, size_t len, uint16_t type, size_t *el_len)
                                           ^
libwps.c:269:32: error: unknown type name 'u_char'
 int libwps_has_rt_header(const u_char *packet, size_t len)
                                ^
libwps.c:291:7: error: unknown type name 'u_char'
 const u_char *libwps_radio_header(const u_char *packet, size_t len)
       ^
libwps.c:291:41: error: unknown type name 'u_char'
 const u_char *libwps_radio_header(const u_char *packet, size_t len)
                                         ^
libwps.c: In function 'libwps_radio_header':
libwps.c:299:17: error: 'u_char' undeclared (first use in this function)
         return (u_char *) FAKE_RADIO_TAP_HEADER;
                 ^
libwps.c:299:17: note: each undeclared identifier is reported only once for each function it appears in
libwps.c:299:25: error: expected expression before ')' token
         return (u_char *) FAKE_RADIO_TAP_HEADER;
                         ^
libwps.c:302:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
Makefile:5: recipe for target 'libwps.o' failed
make[5]: *** [libwps.o] Error 1
make[5]: Leaving directory '/opt/openwrt/build_dir/target-mips_34kc_musl-1.1.11/reaver-wps-fork-t6x-Big_endian/src/libwps'
Makefile:19: recipe for target 'libwps.o' failed
make[4]: *** [libwps.o] Error 2
make[4]: Leaving directory '/opt/openwrt/build_dir/target-mips_34kc_musl-1.1.11/reaver-wps-fork-t6x-Big_endian/src'
Makefile:54: recipe for target '/opt/openwrt/build_dir/target-mips_34kc_musl-1.1.11/reaver-wps-fork-t6x-Big_endian/.built' failed
make[3]: *** [/opt/openwrt/build_dir/target-mips_34kc_musl-1.1.11/reaver-wps-fork-t6x-Big_endian/.built] Error 2
make[3]: Leaving directory '/opt/openwrt/customfeeds/reaver'
package/Makefile:191: recipe for target 'package/feeds/customfeed/reaver/compile' failed
make[2]: *** [package/feeds/customfeed/reaver/compile] Error 2
make[2]: Leaving directory '/opt/openwrt'
package/Makefile:188: recipe for target '/opt/openwrt/staging_dir/target-mips_34kc_musl-1.1.11/stamp/.package_compile' failed
make[1]: *** [/opt/openwrt/staging_dir/target-mips_34kc_musl-1.1.11/stamp/.package_compile] Error 2
make[1]: Leaving directory '/opt/openwrt'
/opt/openwrt/include/toplevel.mk:177: recipe for target 'world' failed
make: *** [world] Error 2

whats this all about?

yousong, 2016-07-03:
The issue also exists in the original reaver code and has been fixed by a patch \[1] in the OpenWrt packages feed

\[1] https://github.com/openwrt/packages/blob/master/net/reaver/patches/0100-Include-sys-types.h-for-definition-of-u_char.patch

rofl0r, 2017-06-21:
fixed in 1.6b.

closed 2017-06-21 by rofl0r