t6x/reaver-wps-fork-t6x #43: Not working with WPS-PBC mode

Daniyal-Javani, 2015-12-15:
If reaver does not support this feature, may be it's cool to add it.
I check it with dumpper (a close source software in windows) and easily do that on TP-Link router.
http://sourceforge.net/projects/dumpper/

kcdtv, 2015-12-16:
wpa_cli from wpa_supplicant does it for you and you have it "by default" in your linux distribution.
 (sudo) wpa_cli  wps_pbc <BSSID> 
But it is done with managed mode interface
  you would have to kill network manager to avoid conflict and run dhcpd to get an IP

Daniyal-Javani, 2015-12-16:
No, I want crack it, without pushing the wps button on router. My target have two mode: wps-pin and wps-pbc. reaver can crack it on wps-pin but can't do it in wps-pbc mode. As I said dumpper can do it.
Thanks for your attention

soxrok2212, 2015-12-16:
The point of push button mode is to physically push the button. You can not break PBC without pushing the button. The point of Reaver is to exploit PIN mode where there needs to be no connection or physical access to the AP.

closed 2015-12-16 by soxrok2212