t6x/reaver-wps-fork-t6x #46: Does this has support to Mac OS X?

cusspvz, 2016-01-03:
I've built `reaver` on mac os x but it seems it doesn't gives along with my mac's wifi card.
Is it just because there is no support for Mac or is this kind of a bug with this broadcom card? Any tip out there?

```bash

MacBook-Pro-de-JM:~ root# reaver -b 9C:97:26:xx:xx:xx -i en0 -c 11 -f -vvvv -K 1

Reaver v1.5.2 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>
mod by t6_x <t6_x@hotmail.com> & DataHead & Soxrok2212 & Wiire & kib0rg

[+] Switching en0 to channel 11
[+] Waiting for beacon from 9C:97:26:xx:xx:xx
[+] Associated with 9C:97:26:xx:xx:xx (ESSID: BRISA)
[+] Starting Cracking Session. Pin count: 0, Max pin attempts: 11000
[+] Trying pin 12345670.
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
^C
[+] Nothing done, nothing to save.
```

kcdtv, 2016-01-04:
Reaver requires "monitor mode" to be functional.
 The code seems to work pretty good in your macintosh but  the problem comes from your "en0" wifi interface which is - very probably - not in monitor mode. 
 The only way i knew in mac os to get monitor mode was "Kismac" (i guess it is spelled like that, not sure)  but it works pretty bad and with a few old chipset and the project stopped several years ago.
  It is a dead end...
  So, the "error" you have comes frist from the iOS (it is not gonna be easy to find a way to enable mode monitor in macintosh) and you also have to know that your broadcom chipset may not work properly even-though you find a way to activate monitor mode in macintosh,  because not all of brodacom chipsets are well supported in linux for monitor mode.

cusspvz, 2016-01-05:
> Reaver requires "monitor mode" to be functional.

Reaver already sets monitor mode on my wnic (`en0`)

> The code seems to work pretty well in your macintosh (you get the EAPOL association correctly!) but the problem comes from your "en0" wifi interface which is - very probably - not in monitor mode.
> The only way i knew in mac os to get monitor mode was "Kismac" (i guess it is spelled like that, not sure) but it works pretty bad and with a few old chipset and the project stopped several years ago.
It is a dead end...

It is, also tried with `airport en0 sniff [channel]` which turns on monitor mode and dumps captures entire traffic into a file. Mac users know if they wnic is in monitor mode because wifi appears with an eye on it.

##### PoC:
[image]

I could capture packets with monitor mode using other sniff tools, but with `reaver` (either from the original repo and this fork) I wasn't able to.

Inspite this issue on `reaver`, `wash` seems to receive beacons and work as expected.

soxrok2212, 2016-01-05:
While the Broadcom wireless adapter on many Macs can go into monitor mode, there is no injection support. Therefore, Reaver will not work. Wash will work however, because it is only listening for packets, not sending.

closed 2016-01-05 by soxrok2212

cusspvz, 2016-01-08:
I was thinking on non-supported injection as the culprit but I though association with AP was made through it, so... :/

Thanks for your response @soxrok2212 !

cusspvz, 2016-01-19:
@soxrok2212 I've been searching a little bit, isn't [JamWIFI](https://github.com/unixpickle/JamWiFi) injecting packages on mac?

cusspvz, 2017-01-31:
ping @soxrok2212 
https://github.com/unixpickle/JamWiFi/blob/ad5c319fa941e1fda5e8fe4165193cb1732af745/Wireless/Tapping/ANInterface.m#L61

soxrok2212, 2017-01-31:
I suggest you talk to the author of that project. The project started years ago and a lot has changed in Apple's API since then.

jonasman, 2017-02-24:
Any news here? I also tried on mac and got the same. Do you know if i install linux via virtual machine would work?

cusspvz, 2017-02-24:
@jonasman I've worked around this issue by placing a VM (on VirtualBox) with kali netboot iso install (didn't wanted X11/GNOME) and I carry around one WIFI stick that i use specifically for it.

jonasman, 2017-02-24:
So no luck with native mac card?

cusspvz, 2017-02-24:
Until now, nope. Are you from Portugal?
The wifi pen I'm using is a TP-LINK TL-WN722N and has a good range. You could find easily on a Worten or online for about 10€.

jonasman, 2017-02-24:
Ok thanks. Seems like i wont be able to pull this off without buying a wifi card.
Im from there but i dont live there.

kcdtv, 2017-02-24:
> TL-WN722N 

That is a good choice, not the best but very cheap and better than many devices from "main manufacturers" which are sold for the double of that price.

jonasman, 2017-03-11:
I ended up buying a TL-WN722N
My setup now:

- Virtualbox
- Kali linux as guest
- TL-WN722N connect to the guest via USB
- reaver installed from this repo.

```
airmon-ng check kill
airmon-ng start wlan0
./wash -i wlan0mon
./reaver -i wlan0mon -b BSSID -vvv -K 1
```

cusspvz, 2017-03-11:
@jonasman nice! Use `wifite`, it comes already with kali and it makes everything easy to use.

jonasman, 2017-03-11:
@cusspvz I will give taht one a try.
I also found another attack tool to crack the default WPS pin. https://github.com/devttys0/wps