t6x/reaver-wps-fork-t6x #47: Is there any way to use reaver with chipset RT3572 dual band wifi adapter

zhangbo, 2016-01-26:
The chipset RT3572 works fine with aircrack-ng.  But When using reaver or bully there is timeout or eap failed or wps failed. If this kinda wifi adapter can not well-supported please let me know! Thanks a lot :)

kcdtv, 2016-03-22:
i reopen this issue to post a kind of answer, sorry soxrok, i didn't realy see it before you closed it
The issue affects all ralink USB chipset that uses rt2800 driver (rt3070 - rt3072 - rt3570 and rt 3572)
They work badly with reaver and i don't recommend them for that
One way to get some better results is to activate the mode monitor with iwconfig or with an old version of airmon-ng
If you use an old version of airmon-ng, you have to explicitly turn on the interface in manged mode with ifconfig.
    sudo ifconfig wlan0 up
Take care to not be connected, it has to be up but not connected
You might try with bully, it looks like it is less buggy with this chipset

closed 2016-03-22 by kcdtv

vincenzogianfelice, 2020-02-26:
> riapro questo problema per pubblicare una specie di risposta, mi dispiace soxrok, non l'ho visto prima di averlo chiuso
> Il problema riguarda tutti i chipset USB ralink che utilizzano il driver rt2800 (rt3070 - rt3072 - rt3570 e rt 3572)
> Funzionano male con reaver e non li raccomando per questo
> Un modo per ottenere risultati migliori è quello di attivare il monitor della modalità con iwconfig o con una vecchia versione di airmon-ng
> Se usi una vecchia versione di airmon-ng, devi esplicitamente girare sull'interfaccia in modalità gestita con ifconfig. es:
> `sudo ifconfig wlan0 up`
> fai attenzione a non essere connesso con la tua interfaccia wlanX, deve essere attivo ma non connesso mentre attacchi con la tua interfaccia monX
> Potresti provare con il bullo, sembra che sia meno difettoso con questo chipset

kcdtv so, which chipsets are most compatible for reaver?

(A few years have passed since the question, but despite everything, there are still these compatibility problems ..)

kcdtv, 2020-03-02:
A lot of progresses have been done with ralink USB chipset (rt2800usb) and they now work pretty well.
Using the `-N` option might be helpful with this RT3X7X chipsets. 
Overall they now work pretty well,