t6x/reaver-wps-fork-t6x #49: Allow bruteforce for last digit instead of computing the checksum

adonespitogo, 2016-02-07:
Hi, I stumbled upon this issue where reaver can't get the correct pin. I found [this issue](https://code.google.com/archive/p/reaver-wps/issues/195) on google code archive and would like to know if there has been a fix for this or a feature that can accomodate manual bruteforce on the last digit of the pin. Thanks

adonespitogo, 2016-02-07:
I think this is irrelevant now. I just found out that what ever first 4 digit is used, the AP always sends m5 message tricking reaver that the first 4 digits are correct when in fact it's not. Any idea/workaround on this?

soxrok2212, 2016-02-07:
The only time this would be useful is if a user specifies the PIN. For example, if a user configures the 8 digit pin, it may not follow the checksum rule. I'd make a bet that if someone were smart enough to change the WPS PIN, they'd just disable the feature as a whole as it is a severely flawed protocol. 

If you were to brute force the last 4 digits without a checksum, it would take another 9,000 guesses totaling at 20,000 for the whole PIN. 

The AP will NEVER send a client the M5 message if the first half of the PIN is correct (unless there is a bug in the code or there is no PIN defined and WPS is enabled, but not configured).

adonespitogo, 2016-02-07:
Thanks for the quick reply. I think in my case the AP's WPS is enabled but not configured.

soxrok2212, 2016-02-07:
You can most likely check in Wireshark. Look in a Probe Response from the router and look for a WPS tag. I don't remember exactly where it is, but it may tell you in there.

adonespitogo, 2016-02-07:
I will try. Thanks

closed 2016-03-18 by soxrok2212