t6x/reaver-wps-fork-t6x #51: Reaver Stuck at trying PIN or earlier

PatrickHentschel, 2016-02-18:
Hi guys i am quite new to hacking and was trying to crack the WPS PIN of my own router, first yes WPS is enabled on the router but as seen below he gets stuck when trying a password, i tried with different commands additionally like -N -L but then he gets stuck at waiting for a beacon. Can anybody help the newbie? Thanks

reaver -i wlan0mon -A -b xx:xx:xx:xx:xx:xx

Reaver v1.5.2 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>
mod by t6_x <t6_x@hotmail.com> & DataHead & Soxrok2212

[+] Waiting for beacon from xx:xx:xx:xx:xx:xx
[+] Associated with xx:xx:xx:xx:xx:xx (ESSID: xxxxxxxxxx)
[+] Starting Cracking Session. Pin count: 0, Max pin attempts: 11000

kcdtv, 2016-02-18:
You need to use "verbosity" to understand what's is going  on. 
You have to add **-vv** or **-vvv**
`(sudo) reaver -b  xx:xx:xx:xx:xx:xx -i wlan0mon -vvv`
A question (two in fact :yum: ) :
  - Why do you use the "**-A**" option?
  - Can't you associate with reaver itself?
  Remember that if you use this option you have to associate with another tool (like aireplay-ng) and that this association has to be active and successful during all the process. 
  The threads in Git Hub are only "code oriented" 
  You'd rather visit a "generalist forum" about wifi cracking/hacking to learn and ask for support about the use of the tool
  Take care

PatrickHentschel, 2016-02-19:
Hey thanks for your tip but now it gets stuck at the 2nd pin another question do you know how to get the internal wireless deactivated so I can use my usb wireless?

https://github.com/t6x/reaver-wps-fork-t6x/issues/51#issuecomment-185700715

evict, 2016-02-20:
Try Googling your question first next time: https://askubuntu.com/questions/168032/how-to-disable-built-in-wifi-and-use-only-usb-wifi-card.

PatrickHentschel, 2016-02-20:
I have seen that post but I am a total noob for Kali thanks anyways

https://github.com/t6x/reaver-wps-fork-t6x/issues/51#issuecomment-186542462

closed 2016-03-21 by soxrok2212

JaviStealsLions, 2017-07-01:
This is how I ended up cracking my WiFi: Used command `reaver -i wlan0mon -b XX:XX:XX:XX:XX:XX(BSSID) -c # -vvv`

If prompted with error associating use `aireplay-ng --fakeauth 1 -a XX:XX:XX:XX:XX:XX(BSSID) -h XX:XX:XX:XX:XX:XX(MAC of interface) wlan0mon` in a different Terminal.

Use `wash -i wlan0mon` in another Terminal to determine signal strength. That was my main problem. I was operating at -79ish, and when I grabbed a USB male to female and extended its reach from my PC I now have a signal of -62

The crack took 2 hours and 30 minutes long to complete.

I did get a bunch of  `WPS transaction failed (code: 0x03), re-trying last pin`'s throughout so don't worry like I was

Hope this helps someone! :)

kl3jvi, 2020-04-08:
> This is how I ended up cracking my WiFi: Used command `reaver -i wlan0mon -b XX:XX:XX:XX:XX:XX(BSSID) -c # -vvv`
>
> If prompted with error associating use `aireplay-ng --fakeauth 1 -a [...]

I love you!!