t6x/reaver-wps-fork-t6x #52: Package triggers severe warnings

blshkv, 2016-03-04:


pchk, 2016-05-03:
#63

blshkv, 2016-07-11:
nop, not fixed:

```
 * QA Notice: Package triggers severe warnings which indicate that it
 *            may exhibit random runtime failures.
 * common.c:614:5: warning: implicit declaration of function ‘wps_pin_checksum’ [-Wimplicit-function-declaration]
 * md5-internal.c:185:5: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
 * md5-internal.c:186:5: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
 * md5-internal.c:191:29: warning: argument to ‘sizeof’ in ‘memset’ call is the same expression as the destination; did you mean to dereference it? [-Wsizeof-pointer-memaccess]
 * wps_dev_attr.c:344:3: warning: implicit declaration of function ‘mac2str’ [-Wimplicit-function-declaration]
 * wps_dev_attr.c:364:3: warning: implicit declaration of function ‘pingen_zyxel’ [-Wimplicit-function-declaration]
```

blshkv, 2016-07-11:
Please open the bug so that you won't forget about it

kcdtv, 2016-08-08:
About the "implicit declaration of a function" it is just a warning
It means that for this function the compiler has not seen a declaration ("prototype") yet:
[http://stackoverflow.com/questions/8440816/warning-implicit-declaration-of-function](url)
For example if you use the algorithm from 'pingen_zyxel' you will see that it works even if the declaration before main was not perfect. 
About "dereferencing type-punned " that's more complex (for me who doesn't know C)

blshkv, 2016-08-09:
I like your "just a warning" attitude. Do you ignore all warnings in your life? What's the reason of the warning?

kcdtv, 2016-08-09:
> I like your "just a warning" attitude

:smiley_cat: 

> Do you ignore all warnings in your life?

Very often: Fear and conformity is the enemy and limitations,  living is about taking risks from time to time :smirk_cat: 
I stop joking: You can see a lot of warnings of this kind compiling a lot of stuff and when you install it works, compilation was not perfect but if you can install and the tool works...

>  What's the reason of the warning?

I don't know C so I can't tell much more than what i said before and you'd better read the threads that i gave as a link about thins kind of errors if you want to understand them before. 
C require that you organize your code following a pattern.
when you see 
** warning: implicit declaration of function ‘pingen_zyxel’**
It means that in one of the file created-used  to generate the PIN "zyxel" (Name is not good: Zyxel is a branch from tehcnicolor and this algorithm was first seen on Tenda router by ZaoChesung) the function "pingen-zyxel" was not declared properly for the compiler. 
Which is  not realy a problem, declaration is done implicitly instead of clearly and explicitly, the compiler handle it.  
  Yes, the code would be nicer and compilation more fluent if it was perfectly written, 
No, it is not a bug 
Try the pingen and you wil see that it works... 
`[+] Waiting for beacon from F8:8E:85:40:84:95
[+] Switching wlan0mon to channel 1
[+] Associated with F8:8E:85:4:smiley_cat: (ESSID: MOVISTAR_:cat:)
[+] Starting Cracking Session. Pin count: 0, Max pin attempts: 11000
[+] Trying pin 12345670.
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[P] E-Nonce: 89:bc:5a:57:e2:3d:b4:93:1c:0f:d5:ea:df:a5:2a:39
[P] PKE: 55:3f:a2:5e:12:cf:ec:45:44:ce:31:24:a1:22:01:ba:1f:3a:ec:d1:63:76:cc:c0:3d:39:bc:a7:9a:51:14:20:dd:4d:39:c8:05:6d:d5:6c:f8:0c:45:47:70:50:56:f5:bb:a5:9b:20:fd:49:e0:6c:a4:1a:dc:e1:d8:da:1c:e7:e5:ca:5f:62:21:4b:18:69:f2:5f:2d:26:9e:5f:74:bb:da:fc:82:b8:1c:8c:32:94:67:45:91:f3:ee:4c:62:db:68:48:00:43:fd:6f:ed:d2:98:dd:1b:34:24:96:c4:c9:c2:0c:e8:55:38:b1:12:a9:b1:3c:a0:0c:c3:ac:0c:c9:72:7a:63:76:72:08:bc:70:d7:34:65:f9:23:32:07:3b:4e:11:1c:25:21:96:16:14:f4:e7:98:2b:a6:44:b4:7e:7e:b5:ca:21:c3:66:a5:57:2c:5c:a1:e5:9a:a5:06:43:9c:af:f6:96:5a:37:ff:23:63:33:2b:b3:5b:62:f5:4e
[P] WPS Manufacturer: Broadcom
[P] WPS Model Name: Broadcom
[P] WPS Model Number: 123456
[P] Access Point Serial Number: 1234
[Pin Gen] Zyxel Default Pin Generator
[Pin Gen] Pin Generated : 42282457
[Pin Gen] Pin Generated (+1): 42282464
[Pin Gen] Pin Generated (-1): 42282440
`
Although it will be better to generate the PIN before than after... didn't see that ... but that's a another issue: The function works and the PIN is correctly generated. 
  But you're absolutely  right: It would be better to have a better written code and i have exactly the same warning as you with our without the fix.
   For me neither it didn't fixed this warnings and this issue should remain opened to "not forget about it"

kcdtv, 2016-08-14:
This issue is about the same than this one;  "Compile warning" #36 
I close the other one and we keep this one open: There is no o need to have two issues about theses warnings.

kcdtv, 2016-09-01:
Some patches proposed in  "fix -n argument" are actually  related to this issue.
This are the patches:
https://github.com/t6x/reaver-wps-fork-t6x/pull/72/commits/18e86f4aa7b87301148bcd7dcf0028e0a1c11ec6
https://github.com/t6x/reaver-wps-fork-t6x/pull/72/commits/cd1af2110bd2d6a2078012b622dd35aff086a22b
https://github.com/t6x/reaver-wps-fork-t6x/pull/72/commits/6b39995ecb0f032f2327becac652a1ad39aa987f
https://github.com/t6x/reaver-wps-fork-t6x/pull/72/commits/da955d6f725607df12441043828e5f4da5de62ff

blshkv, 2016-09-01:
confirmed as fixed

kcdtv, 2016-09-05:
As long as the patches are not introduced let's keep it open 
sorry for the time it takes but ... we have a life :smile_cat: 
But be sure that soon we will merge the adecuate patches to fix the issues (which is not an issue, just bullshit without importance to have a compilation without warning)  
cheers.

kcdtv, 2016-09-08:
We can close it now: corrective patch have been merged in last update, clean compilation, not a single warning. GJ vk!

blshkv, 2016-09-09:
tested the latest commit 86a45b5076b0100875de322ec0038337b13bfabc:

NOT fixed in master:

```
 * QA Notice: Package triggers severe warnings which indicate that it
 *            may exhibit random runtime failures.
 * common.c:614:5: warning: implicit declaration of function ‘wps_pin_checksum’ [-Wimplicit-function-declaration]
 * md5-internal.c:185:5: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
 * md5-internal.c:186:5: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
 * md5-internal.c:191:29: warning: argument to ‘sizeof’ in ‘memset’ call is the same expression as the destination; did you mean to dereference it? [-Wsizeof-pointer-memaccess]
 * wps_dev_attr.c:344:3: warning: implicit declaration of function ‘mac2str’ [-Wimplicit-function-declaration]
 * wps_dev_attr.c:364:3: warning: implicit declaration of function ‘pingen_zyxel’ [-Wimplicit-function-declaration]
```

soxrok2212, 2016-09-09:
Are you sure you pulled the latest commit? We had 3 people test (one of which being myself, running Debian 8) and had zero issues.

blshkv, 2016-09-09:
```
ebuild reaver-wps-fork-t6x-1.5.2_p20160909.ebuild compile 
>>> Unpacking source...
 * Fetching https://github.com/t6x/reaver-wps-fork-t6x.git ...
git fetch https://github.com/t6x/reaver-wps-fork-t6x.git +HEAD:refs/git-r3/HEAD
git update-ref --no-deref refs/git-r3/net-wireless/reaver-wps-fork-t6x/0/__main__ 86a45b5076b0100875de322ec0038337b13bfabc
 * Checking out https://github.com/t6x/reaver-wps-fork-t6x.git to /var/tmp/portage/net-wireless/reaver-wps-fork-t6x-1.5.2_p20160909/work/reaver-wps-fork-t6x-1.5.2_p20160909 ...
git checkout --quiet 86a45b5076b0100875de322ec0038337b13bfabc
GIT update -->
   repository:               https://github.com/t6x/reaver-wps-fork-t6x.git
   at the commit:            86a45b5076b0100875de322ec0038337b13bfabc
>>> Source unpacked in /var/tmp/portage/net-wireless/reaver-wps-fork-t6x-1.5.2_p20160909/work
>>> Preparing source in /var/tmp/portage/net-wireless/reaver-wps-fork-t6x-1.5.2_p20160909/work/reaver-wps-fork-t6x-1.5.2_p20160909 ...
 * Running elibtoolize in: reaver-wps-fork-t6x-1.5.2_p20160909/src/
>>> Source prepared.
>>> Configuring source in /var/tmp/portage/net-wireless/reaver-wps-fork-t6x-1.5.2_p20160909/work/reaver-wps-fork-t6x-1.5.2_p20160909 ...
>>> Working in BUILD_DIR: "/var/tmp/portage/net-wireless/reaver-wps-fork-t6x-1.5.2_p20160909/work/reaver-wps-fork-t6x-1.5.2_p20160909/src"
/var/tmp/portage/net-wireless/reaver-wps-fork-t6x-1.5.2_p20160909/work/reaver-wps-fork-t6x-1.5.2_p20160909/src/configure --prefix=/usr --build=x86_64-pc-linux-gnu --host=x86_64-pc-linux-gnu --mandir=/usr/share/man --infodir=/usr/share/info --datadir=/usr/share --sysconfdir=/etc --localstatedir=/var/lib --libdir=/usr/lib64 --docdir=/usr/share/doc/reaver-wps-fork-t6x-1.5.2_p20160909
checking for x86_64-pc-linux-gnu-gcc... x86_64-pc-linux-gnu-gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether x86_64-pc-linux-gnu-gcc accepts -g... yes
checking for x86_64-pc-linux-gnu-gcc option to accept ISO C89... none needed
checking for pcap_open_live in -lpcap... yes
checking for sqlite3_open in -lsqlite3... yes
checking how to run the C preprocessor... x86_64-pc-linux-gnu-gcc -E
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for stdlib.h... (cached) yes
checking for stdint.h... (cached) yes
checking for string.h... (cached) yes
checking pcap.h usability... yes
checking pcap.h presence... yes
checking for pcap.h... yes
checking sqlite3.h usability... yes
checking sqlite3.h presence... yes
checking for sqlite3.h... yes
configure: creating ./config.status
config.status: creating Makefile
>>> Source configured.
>>> Compiling source in /var/tmp/portage/net-wireless/reaver-wps-fork-t6x-1.5.2_p20160909/work/reaver-wps-fork-t6x-1.5.2_p20160909 ...
>>> Working in BUILD_DIR: "/var/tmp/portage/net-wireless/reaver-wps-fork-t6x-1.5.2_p20160909/work/reaver-wps-fork-t6x-1.5.2_p20160909/src"
make -j4 
(cd utils && make)
(cd tls && make)
(cd lwe && make BUILD_STATIC=y libiw.a)
(cd libwps && make)
make[1]: Entering directory '/var/tmp/portage/net-wireless/reaver-wps-fork-t6x-1.5.2_p20160909/work/reaver-wps-fork-t6x-1.5.2_p20160909/src/lwe'
make[1]: Entering directory '/var/tmp/portage/net-wireless/reaver-wps-fork-t6x-1.5.2_p20160909/work/reaver-wps-fork-t6x-1.5.2_p20160909/src/tls'
make[1]: Entering directory '/var/tmp/portage/net-wireless/reaver-wps-fork-t6x-1.5.2_p20160909/work/reaver-wps-fork-t6x-1.5.2_p20160909/src/utils'
make[1]: Entering directory '/var/tmp/portage/net-wireless/reaver-wps-fork-t6x-1.5.2_p20160909/work/reaver-wps-fork-t6x-1.5.2_p20160909/src/libwps'
gcc -Wall libwps.c -c
cp wireless.21.h wireless.h
gcc -Os -W -Wall -Wstrict-prototypes -Wmissing-prototypes -Wshadow -Wpointer-arith -Wcast-qual -Winline -I. -MMD     -fPIC -c -o iwlib.so iwlib.c
libwps.c: In function ‘parse_wps_parameters’:
libwps.c:29:12: warning: variable ‘data_len’ set but not used [-Wunused-but-set-variable]
[... 332 lines cut ...]
             ^
wps_registrar.c:770:13: warning: ‘wps_registrar_pin_completed’ defined but not used [-Wunused-function]
 static void wps_registrar_pin_completed(struct wps_registrar *reg)
             ^
wps_registrar.c:862:12: warning: ‘wps_cb_new_psk’ defined but not used [-Wunused-function]
 static int wps_cb_new_psk(struct wps_registrar *reg, const u8 *mac_addr,
            ^
wps_registrar.c:888:13: warning: ‘wps_cb_reg_success’ defined but not used [-Wunused-function]
 static void wps_cb_reg_success(struct wps_registrar *reg, const u8 *mac_addr,
             ^
wps_registrar.c:1456:24: warning: ‘wps_build_m2d’ defined but not used [-Wunused-function]
 static struct wpabuf * wps_build_m2d(struct wps_data *wps)
                        ^
wps_registrar.c:2252:12: warning: ‘wps_process_e_snonce2’ defined but not used [-Wunused-function]
 static int wps_process_e_snonce2(struct wps_data *wps, const u8 *e_snonce2)
            ^
[... cut ...]
x86_64-pc-linux-gnu-gcc -DCONF_DIR='"/etc/reaver"' -DREAVER_DATABASE='"/etc/reaver/reaver.db"' -Wall -march=native -O2 -pipe -Ilibwps wpscrack.c *.o libwps/*.o wps/*.o tls/bignum.o tls/libtls.a utils/libutils.a crypto/libcrypto.a lwe/libiw.a -ldl -lm -lpcap -lsqlite3 -Wl,-O1 -Wl,--as-needed -Wl,--hash-style=gnu -o reaver
wpscrack.c: In function ‘main’:
wpscrack.c:134:13: warning: ignoring return value of ‘system’, declared with attribute warn_unused_result [-Wunused-result]
             system(get_exec_string());
             ^
>>> Source compiled.
```

soxrok2212, 2016-09-09:
Latest version is 1.5.3

blshkv, 2016-09-09:
Please re-open the bug report ;-)

kcdtv, 2016-09-09:
which system do you use?

soxrok2212, 2016-09-09:
Compiles perfectly fine on Debian 8, latest updates. Fresh clone. 

`root@debian-vm:/home/soxrok2212/reaver-wps-fork-t6x/src|master
⇒  ./configure
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for pcap_open_live in -lpcap... yes
checking for sqlite3_open in -lsqlite3... yes
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for stdlib.h... (cached) yes
checking for stdint.h... (cached) yes
checking for string.h... (cached) yes
checking pcap.h usability... yes
checking pcap.h presence... yes
checking for pcap.h... yes
checking sqlite3.h usability... yes
checking sqlite3.h presence... yes
checking for sqlite3.h... yes
configure: creating ./config.status
config.status: creating Makefile
root@debian-vm:/home/soxrok2212/reaver-wps-fork-t6x/src|master⚡
⇒  make
(cd utils && make)
make[1]: Entering directory '/home/soxrok2212/reaver-wps-fork-t6x/src/utils'
  CC  base64.c
common.c: In function ‘pingen_zhaochunsheng’:
common.c:597:14: warning: passing argument 1 of ‘snprintf’ makes pointer from integer without a cast
     snprintf(pin, pin_len, "%08d", (default_pin * 10) + wps_checksum(default_pin));
              ^
In file included from includes.h:26:0,
                 from common.c:15:
/usr/include/stdio.h:386:12: note: expected ‘char * __restrict__’ but argument is of type ‘int’
 extern int snprintf (char *__restrict __s, size_t __maxlen,
            ^
common.c: In function ‘pingen_zyxel’:
common.c:614:5: warning: implicit declaration of function ‘wps_pin_checksum’ [-Wimplicit-function-declaration]
     return (pin * 10) + wps_pin_checksum(pin);
     ^
  CC  common.c
  CC  ip_addr.c
  CC  radiotap.c
  CC  trace.c
  CC  uuid.c
  CC  wpa_debug.c
  CC  wpabuf.c
  CC  os_unix.c
  CC  eloop.c
ar crT libutils.a base64.o common.o ip_addr.o radiotap.o trace.o uuid.o wpa_debug.o wpabuf.o os_unix.o eloop.o
make[1]: Leaving directory '/home/soxrok2212/reaver-wps-fork-t6x/src/utils'
(cd tls && make)
make[1]: Entering directory '/home/soxrok2212/reaver-wps-fork-t6x/src/tls'
[... 215 lines cut ...]
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 init.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 misc.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Ilibwps sql.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 builder.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 crc.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Ilibwps 80211.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 iface.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Ilibwps wpsmon.c *.o libwps/*.o wps/*.o tls/bignum.o tls/libtls.a utils/libutils.a crypto/libcrypto.a lwe/libiw.a -ldl -lm -lpcap -lsqlite3  -o wash
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Ilibwps sigint.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 sigalrm.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 keys.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 pins.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 send.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Ilibwps exchange.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Ilibwps session.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Ilibwps cracker.c -c
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Ilibwps wpscrack.c *.o libwps/*.o wps/*.o tls/bignum.o tls/libtls.a utils/libutils.a crypto/libcrypto.a lwe/libiw.a -ldl -lm -lpcap -lsqlite3  -o reaver
root@debian-vm:/home/soxrok2212/reaver-wps-fork-t6x/src|master⚡
⇒  make install
rm -f /usr/local/bin/reaver /usr/local/bin/wash /usr/local/bin/walsh
if [ ! -d /usr/local/etc/reaver ]; then mkdir -p /usr/local/etc/reaver; fi
if [ ! -e /usr/local/etc/reaver/reaver.db ]; then cp reaver.db /usr/local/etc/reaver/reaver.db && chmod -R a+rw /usr/local/etc/reaver; fi
if [ -e wash ]; then cp wash /usr/local/bin/wash; fi
if [ -e reaver ]; then cp reaver /usr/local/bin/reaver; fi
`

ZeroChaos-, 2016-09-09:
@soxrok2212 literally has more warnings than reported by @blshkv yet said "it compiles perfectly"

I think the problem is that people don't care about the compiler warnings....

kcdtv, 2016-09-10:
Hi zero Chaos
Do you think that if we didn't care we will loose time trying to find a solution or that i would be at 2 am answering you? :smiley_cat:
  I promise you that if I didn't care I would be sleeping right now :smile:    
Yes, soxrcok2212 says that it compiles perfectly under debian.... Because it does.
* With kali 2016.x: no warning ( i tryed several time)
* With debian 8: no warning (soxrock tried)
* With Xubuntu 16.04 : one warning (at the end, function wpscrack, i just tryed) 
* With slackware: no warning (fixed made by vk496 under this system)  
this is the only fail i get, under ubuntu

> kcdtv@Andtoid4S1M0V:~/reaver-wps-fork-t6x.git/trunk/src$ make
(cd utils && make)
make[1] : on entre dans le répertoire « /home/kcdtv/reaver-wps-fork-t6x.git/trunk/src/utils »
make[1]: rien à faire pour « all ».
make[1] : on quitte le répertoire « /home/kcdtv/reaver-wps-fork-t6x.git/trunk/src/utils »
(cd tls && make)
make[1] : on entre dans le répertoire « /home/kcdtv/reaver-wps-fork-t6x.git/trunk/src/tls »
make[1]: rien à faire pour « all ».
make[1] : on quitte le répertoire « /home/kcdtv/reaver-wps-fork-t6x.git/trunk/src/tls »
(cd crypto && make)
make[1] : on entre dans le répertoire « /home/kcdtv/reaver-wps-fork-t6x.git/trunk/src/crypto »
make[1]: rien à faire pour « all ».
make[1] : on quitte le répertoire « /home/kcdtv/reaver-wps-fork-t6x.git/trunk/src/crypto »
(cd wps && make)
make[1] : on entre dans le répertoire « /home/kcdtv/reaver-wps-fork-t6x.git/trunk/src/wps »
make[1]: rien à faire pour « all ».
make[1] : on quitte le répertoire « /home/kcdtv/reaver-wps-fork-t6x.git/trunk/src/wps »
(cd lwe && make BUILD_STATIC=y libiw.a)
make[1] : on entre dans le répertoire « /home/kcdtv/reaver-wps-fork-t6x.git/trunk/src/lwe »
make[1]: « libiw.a » est à jour.
make[1] : on quitte le répertoire « /home/kcdtv/reaver-wps-fork-t6x.git/trunk/src/lwe »
(cd libwps && make)
make[1] : on entre dans le répertoire « /home/kcdtv/reaver-wps-fork-t6x.git/trunk/src/libwps »
make[1]: « libwps.o » est à jour.
make[1] : on quitte le répertoire « /home/kcdtv/reaver-wps-fork-t6x.git/trunk/src/libwps »
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Ilibwps wpsmon.c *.o libwps/*.o wps/*.o tls/bignum.o tls/libtls.a utils/libutils.a crypto/libcrypto.a lwe/libiw.a -ldl -lm -lpcap -lsqlite3  -o wash
gcc -DCONF_DIR='"/usr/local/etc/reaver"' -DREAVER_DATABASE='"/usr/local/etc/reaver/reaver.db"' -Wall -g -O2 -Ilibwps wpscrack.c *.o libwps/*.o wps/*.o tls/bignum.o tls/libtls.a utils/libutils.a crypto/libcrypto.a lwe/libiw.a -ldl -lm -lpcap -lsqlite3  -o reaver
wpscrack.c: In function ‘main’:
wpscrack.c:134:13: warning: ignoring return value of ‘system’, declared with attribute warn_unused_result [-Wunused-result]
             system(get_exec_string());
             ^

  it seems that blshkv use a distribution called "pentoo" which might be based on gentoo 
So the situation is:  
1) reaver always had this warning.
2) Since this commit warning are solved for debian, kali, salckware and nearly solved for Ubuntu 
3) For pentoo/gentoo: no change (or worst, can't tell, never tried) 
   
Remember that reaver mod is first of all designed for kali linux which is a debian SID based system
   Bringing support to ubuntu and linux mint or other derivative from debian can be pretty straight forward and easy
    We for sure would be happy to introduce any modification that can fix this issues (as long at it does not break kali linux/debian compilation) in gentoo and derivatives
  Remember that this is not a critical issue: Install reaver and you will see that it installs and works properly 
  Cheers

blshkv, 2016-09-10:
Yes, I'm developer of Pentoo Linux. But you are missing the point because It's not about linux distribution. The main difference is that we use hardened linux and more strict GCC parameters by default.

It is obvious from my log if you try try to read it. For example:
```
gcc -Os -W -Wall -Wstrict-prototypes -Wmissing-prototypes -Wshadow -Wpointer-arith -Wcast-qual -Winline -I. -MMD     -fPIC -c -o iwlib.so iwlib.c
```

kcdtv, 2016-09-10:
Hi blshkv
   When I said that "_the linux distribution_" it was a "shortcut" to speak about the packages/elements that composes your Operative System 
And I was actually thinking that it was coming from your gcc version... For one time that I was not equivocated, please don't take that away form me :smiley: 
  May i suggest to you (and your "presidente de por vida" ZeroChaos) next time, to clarify things since the beginning, that you should start by saying that you use a fork from gentoo instead of a supported distribution (debian bazed) ... 
  It would be good to bring much more elements when you report an issue: They actually  don't charge you in github for making sentences or using words, you can be eloquent  :smiley_cat: 
   About your gcc version.... upssssss...  What can I say? .. 
It is more than one year old and have known security issue https://www.cvedetails.com/cve/CVE-2015-5276/
  Backward compatibility with an unsafe version is definitely not generally a priority in a project and perfect "retrocompatibility" is sadly unsustainable.... 
   As I said: I don't know C, don't know the art of compiling, do not use a gentoo derivate, do not downgrade package to break my dependencies system so....

> @soxrok2212 literally has more warnings than reported by @blshkv yet said "it compiles perfectly"
I think the problem is that people don't care about the compiler warnings....

Un momento por favor "_El Presidente Numero Uno. Current lead of Pentoo, second to the great Grimmlin_" : Soxrock said it works perfectly in debian and you are not speaking about debian but about your own SO... That doesn't make any sense for anyone here.
         Do you understand?
   If you want things to go forward bring new elements instead of just complain about "people that doesn't care".
 That's not productive at all and does not actually motivate anyone to try to find a solution for you   
  
   As I said: We, for sure, would be very happy to introduce any possible fix in reaver for your OS and will leave the issue open.   (as long as it doesn't break supported system or systems with an updated gcc version)
  
   Take care and good luck with your project :smile_cat:

blshkv, 2016-09-10:
So pentoo is not a fork. It's an overlay. 

So instead of sayong debian, tell me which version of gcc should i try.

kcdtv, 2016-09-15:
Sorry if i didn't answer before, i was very busy, no time for nothing 

> So Pentoo is not a fork. It's an overlay,

I don't know what is an "overlay" but saying fork was not pejorative in my mind. 
Thanks for the precisions. :wink: 

>Zc is also a Gentoo developer so you might want to take it more serious and slow down with your sarcastic comments. 

:smiley_cat:
So?
 Well, sorry, nothing personal, he could be the Queen of England or the pope himself, that wouldn't change a dam thing... 
To be clear: That was not sarcasm but the normal reaction to expect when dealing with somebody that comes to a thread just to complain. **Whoever he is**. 
I am very surprised that a developer acts with such  an nonconstructive attitude and make such a useless comment .
  I would not say anything more about this subject than a spanish proverb  : "_A llorar a la iglesia_" ( If you want to cry: Go to a church, - not in git hub)
  

>  We use the latest stable gcc but open for any suggestions.
> So instead of saying "debian", "supported" and the rest BS, tell me which version of gcc should i try.

I am not developer, I don't know as much as you, so if you don't like my answers because they are not technical/precise enough:
    1) Sorry: I am not a pro and don't know as much as you.
    2) If you want to know what version i use,  just ask,  that's simple...   instead of telling me "what i should say instead of what I say"...    you and your friend have a **very strange** conception of communication... 
   3) BS? :-1:  Stay polite and change your depreciative (and arrogant) attitude.   
     I'll try to be constructive even if you guys are realy not... 
But I sincerely don't know what you should do or what can be done for you, 
  From what I understood **gcc 6.2** is the last stable release ( and not 4.9.3) 
So maybe try with the last stable release.
  Or at least with a more recent version
I can tell you that with version superior to the one you use in distribution such as ubutnu, kali, slackware and debian there is no problem (but i guess that you understood that, the "bullshit" part) 
This is my stdout of gcc -v with kali (updated) 
`gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/6/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Debian 6.1.1-11' --with-bugurl=file:///usr/share/doc/gcc-6/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-6 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-6-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-6-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-6-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 6.1.1 20160802 (Debian 6.1.1-11)`

> Also, you should try strict options of gcc as i suggested

I don't know how to do that 
 I use the gcc provided by the repositories of the distributions I use as they provide it, no modification, no changes.  
  It is 64bits with multiarch etc... i can post you the stdout of gcc in ubuntu if you want
Sorry if that doesn't help you a lot, I really can't tell you more... 
  We for sure will keep this issue open as long as needed for a better compatibility with pentoo in order to satisfy your request. 
   Take care.

blshkv, 2016-09-16:
@kcdtv you said you dont know C and not a developer. Please dont interrupt technical thread then.

so again, this is not a pentoo bug and not pentoo compatibility.
There are few security options in GCC which might not enabled by default at your favorite distro. These options can detect various problems on the early stage (during compilation).

To enable them all, try this:
```
./configure
make -Wall
```

Make sure to scroll back and read the output.

kcdtv, 2016-09-16:
> @kcdtv you said you dont know C and not a developer. Please dont interrupt technical thread then.

  
You are amazing :smile: 
 My "interruptions" are required here just because **you did not provide the most essential and basic informations when you first exposed your problem**. ( 5 warnings dropped here without any explanation.... What do you want? ) 
 There is no need (at all) for highly technical knowledge in C programing to understand that you did not gave the required details about your configuration/system, that you do not gave at any time the steps you followed to lead to this error... 
That is common sense: A question of logic, not of development or programing.
By the way, I might not be a C programmer or a "overlay" developer or pretend to be "_presidente de por vida_" but i do take part in others projects also since decade and that the way things have to be done  everywhere, for a project in python, bash, moderating a forum or working group or whatever. 
 It is not my fault if you did not take time to expose properly your issue and i would not have to interfere if this fundamental information would have been provided since message one. :wink: 
 

> so, again, it has nothing to do with pentoo or pentoo compatibility.

And **again**: Compilation works perfectly in an updated kali-ubuntu-slackware-arch etc... system following the steps from the  manual 
`./configure && make && make install`
If you do not follow the steps and do not say so at the beginning we just can assume that it comes from pentoo or the gcc version that you use ( which is outdated and is not the last stable release). 
I will now change the tittle as I know understand better what you are doing and why you have this warnings. 
 To sum-up and clarify this messy situation due to your poor and incomplete facts exposure:
 1) You are right:  Compiling with **-W** flag does actually provokes several warning and this regardless the version of gcc or the GNU-Linux distribution used, making the current patches from vk946 "useless".
 2) Nevertheless: **Following the manual and compiling as it is recommended this warning does not show at all** (just one in latest ubuntu - 16.04)which is a good step forward and to my sense legitimate that we maintain the patches of vk946 and do not revert them
 3) It is maybe good to remember that this warnings does not affect at all the good functioning of reaver and are kind of  "cosmetic issue"... For example:  an implicitly declared variable in the "pingen for zyxell" is not a concern at all any time 
  
  Take care and good luck with your projetcs.

blshkv, 2016-09-16:
Some warnings are cosmetic but other are severe as I reported in the very first comment.

blshkv, 2016-09-18:
3) some warnings are "cosmetic" while others are not. I have listed only "severe" ones which might cause memory corruption and lead to a crash or become a security issue.

kcdtv, 2016-11-29:
Issue is closed and warning problems will be debated in other threads where people don't come to demand and complain but to participate with a positive spirit and a non arrogant attitude.

closed 2016-11-29 by kcdtv

blshkv, 2016-11-29:
Absolutely  agree.  It is very difficult to bring to the point when all arguments hit  non technical "debian only"  wall attitude

kcdtv, 2016-11-29:
Mmmmm
   You  sadly totally  missed the point with your so original "debian only" blabla...
     The problem is simply your behavior of arrogant spoiled children. 
       Good luck with that

blshkv, 2016-11-29:
Nop,  I totally  get it.  But this is an arrogant behavior of technical guy talking to a wrong guy who doesn't even know C and trying to act as a protective PM.  Come on,  it's  been two  months  since we settled all missunderstands.  Let it go,  focus on the problem and fix it.

kcdtv, 2016-12-21:
I didn't see that 
Technical  guru? LOL
Guru for script kiddies maybe.... 
i will do you a favour and lock this conversation because you are right now ridiculously arrogant 
Ask to santa claus for some humility dude, 
merry xmas,