t6x/reaver-wps-fork-t6x #54: error: pcap library not found!

eminmammadov, 2016-03-17:
root@root:~/reaver-wps-fork-t6x/src# ./configure
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for pcap_open_live in -lpcap... no
**## error: pcap library not found! **(what is?)

crushedice2000, 2016-03-18:
@eminmammadov:

1. Please, when you paste code or terminal output in GitHub, prepend four spaces to each line. If you do this, the block will be monospaced. [See the markdown reference](https://help.github.com/categories/writing-on-github/). :smile: 

2. Quote from [The Bible of Reaver](https://github.com/t6x/reaver-wps-fork-t6x/blob/master/docs/README)  (lines 41-50): :smiling_imp: 

    > Reaver is only supported on the Linux platform, requires the libpcap and libsqlite3 libraries, and can be built and installed by running:

    >     ./configure
    >     make
    >     sudo make install

    > To remove everything installed/created by Reaver:

    >     sudo make distclean

3. You can install these libraries with these methods:

    + For Debian-based systems:

            sudo apt-get install sqlite3 libsqlite3-dev libpcap0.8-dev

closed 2016-03-18 by soxrok2212

eminmammadov, 2016-03-18:
@crushedice2000 Thanks my friend.

crushedice2000, 2016-03-18:
@eminmammadov @soxrok2212 : You're welcome!

jovanymezura, 2018-01-05:
still working