t6x/reaver-wps-fork-t6x #56: error during make

darkworks, 2016-03-28:
i got error when i run make command after configure , on kali sana 2.0

```
2 -Ilibwps wpscrack.c *.o libwps/*.o wps/*.o tls/bignum.o tls/libtls.a utils/libutils.a crypto/libcrypto.a lwe/libiw.a -ldl -lm -lpcap -lsqlite3  -o reaver
wpscrack.c: In function ‘main’:
wpscrack.c:40:33: warning: variable ‘r’ set but not used [-Wunused-but-set-variable]
     int ret_val = EXIT_FAILURE, r = 0;
                                 ^

```

aanarchyy, 2016-03-28:
You got a warning, not an error.

darkworks, 2016-03-28:
hmmmm then why make process stopped  , i mean after that if i do make install , it not work

kcdtv, 2016-03-28:
Could you put here the exact (full) stdout of both commands?
If you have an error while compiling you will have an error message at the end, so it is important for us to see everything that happens in your shell
Cheers

profedrini, 2016-03-29:
It's a warning. Compilation stopped but you should get the executable on the current directory

Run    ./reaver  after compilation stops

closed 2016-05-24 by soxrok2212