t6x/reaver-wps-fork-t6x #58: Channel 13 not working and Ctrl + C not save

earthGavinLee, 2016-04-03:
# airmon-ng
PHY     Interface       Driver          Chipset
phy0    wlan0mon        ??????          Realtek Semiconductor Corp. RTL8187

#reaver -i wlan0mon -b xx:xx:xx:xx:xx:xx -vv -d9 -t9 -c13

aanarchyy, 2016-04-03:
Does reaver associate on other channels?
Can you get aireplay-ng to associate on channel 13?

earthGavinLee, 2016-04-05:
reaver channel 1 ~11 is ok 

airodump-ng -w xxxxxxxxx -c13 --bssid xx:xx:xx:xx:xx:xx  wlan0mon
CH 13 ][ Elapsed: 2 mins ][ 2016-04-05 11:10                                  
BSSID              PWR RXQ  Beacons    #Data, #/s  CH  MB   ENC  CIPHER AUTH E
xx:xx:xx:xx:xx:xx  -52  40      872        2    0  13  54e. WPA2 CCMP   PSK  xxxxxxxxx
BSSID              STATION            PWR   Rate    Lost    Frames  Probe     
xx:xx:xx:xx:xx:xx  xx:xx:xx:xx:xx:xx   -1    1e- 0      0        2

soxrok2212, 2016-04-23:
Is your driver region set to use channel 13?

closed 2016-05-24 by soxrok2212

strasharo, 2016-06-02:
Try:
iw reg get

This will show your currently used regdom.

earthGavinLee, 2016-06-30:
/ # iw reg get
global
country 00: DFS-UNSET
        (2402 - 2472 @ 40), (N/A, 20), (N/A)
        (2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
        (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
        (5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
        (5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
        (57240 - 63720 @ 2160), (N/A, 0), (N/A)

global
country 00: DFS-UNSET
        (2402 - 2472 @ 40), (N/A, 20), (N/A)
        (2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
        (2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
        (5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
        (5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
        (5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
        (5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
        (57240 - 63720 @ 2160), (N/A, 0), (N/A)

earthGavinLee, 2016-06-30:
lsusb
Bus 001 Device 005: ID 0bda:8187 Realtek Semiconductor Corp. RTL8187 Wireless Adapter

kcdtv, 2016-06-30:
Nooooooo
   This doesn't make any sense... 
If reaver would not be able to use channel 13 you would not even see "failed to associate", nothing would happen...
What happens is that you are attacking a PA too far (probably not your but that's not my concern) and you don't manage to associate.
 That's all.
Saying "reaver doesn't works in channel 13" is over interpretative.    
Obviously Reaver works absolutely the same in every channel,
It  associates and works properly in channel 13  
I am going to show it to you right now using the same chipset as you (RTL8187l)  in channel 13 (kali 2016 rolling) 

[image]

Please, before posting issue, try with your own access point in good conditions and check twice, it would be very appreciated.  :smiley_cat: