t6x/reaver-wps-fork-t6x #59: EAPOL timeouts

Dreadmonkey, 2016-04-05:
Hi Guys,

Thanks very much in advance, I'm getting EAPOL timeouts whenever I try to use reaver on a kali2 installation. At first I thought it was the internal card (Intel Wireless 7260 (rev bb)) so have since bought an Alfa AWUS036NHA and am seeing the same results.

Below is output from the intel card when I first asked this on unixstackexchange but didn't get anywhere. The new Alfa card shows exactly the same output and is installed as follows;

=====================================
Interface             Driver                     Chipset
wlan2                 ath9k_htc                Atheros Communications, Inc. AR9271 800211n
=====================================

Any help would be very much appreciated.

**Output on intel card**
====================================================
root@kali:~# airmon-ng check kill
Killing these processes:
PID Name
776 wpa_supplicant

root@kali:~# airmon-ng start wlan0
No interfering processes found
PHY     Interface   Driver      Chipset
phy0    wlan0       iwlwifi     Intel Corporation Wireless 7260 (rev bb)
        (mac80211 monitor mode vif enabled for [phy0]wlan0 on [phy0]wlan0mon)
        (mac80211 station mode vif disabled for [phy0]wlan0)

root@kali:~#airodump-ng wlan0mon<br>

BSSID              PWR  Beacons    #Data, #/s  CH  MB   ENC  CIPHER AUTH ESSID

xxxxxxxxxxxxxxxxx  -48       48       21    0   1  54e  WPA2 CCMP   PSK  xxx
xxxxxxxxxxxxxxxxx  -57       20        0    0   6  54e. WPA2 CCMP   MGT  xxx
xxxxxxxxxxxxxxxxx  -58       19        0    0   6  54e. WPA2 CCMP   PSK  xxx
xxxxxxxxxxxxxxxxx  -59       24        0    0   6  54e. OPN              xxx
xxxxxxxxxxxxxxxxx  -79       29        1    0   5  54e  WPA2 CCMP   PSK  xxx
xxxxxxxxxxxxxxxxx  -83       17        0    0  11  54e  WPA2 CCMP   PSK  xxx
xxxxxxxxxxxxxxxxx  -83       13        0    0  11  54e  WPA2 CCMP   MGT  xxx
xxxxxxxxxxxxxxxxx  -83       19        0    0  11  54e  OPN              xxx

root@kali:~# reaver -i wlan0mon -b xxxxxxxxxxxxxxxxx -A -S -N -a -vv -K 1

Reaver v1.5.2 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>
mod by t6_x <t6_x@hotmail.com> & DataHead & Soxrok2212

[+] Waiting for beacon from xxxxxxxxxxxxxxxxx 
[+] Switching wlan0mon to channel 1 
[+] Associated with xxxxxxxxxxxxxxxxx (ESSID: xxxxx) 
[+] Starting Cracking Session. Pin count: 0, Max pin attempts: 11000 
[+] Trying pin 12345670.
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[!] WARNING: 25 successive start failures
[+] Sending EAPOL START request
[+] Sending WSC NACK
[!]  WPS transaction failed (code: 0x02), re-trying last pin
[+] Trying pin 12345670.

============================

soxrok2212, 2016-04-05:
Remove -A from your command. -A tells Reaver not to associate (sometimes it's required to do this manually, then you would need -A) but most of the time you'll be fine without -A

Dreadmonkey, 2016-04-05:
I should have been clearer, with the intel card i had to use wash to perform the association otherwise reaver just hung waiting for a beacon. With the Alfa card I can associate but still get the EAPOL timeouts

soxrok2212, 2016-04-05:
Wash does not associate with anything, it just looks in beacon frames to see if WPS is enabled or not. Try without -A and it should work. Internal cards in general are not the best for this purpose, not to mention it's Intel which has terrible support for injection on it's newer cards.

soxrok2212, 2016-04-05:
Reaver will also say it's associated with the AP when you use -A because it's expecting you to manually associate, but in reality it's not associated.

Dreadmonkey, 2016-04-05:
Yeah that's why I bought the Alfa card.

Just running it now without the -A and reaver won't associate, output below with the mac address removed;

root@Kali:~# reaver -i wlan2mon -b 00:21:29:DE:94:8A -vv -K 1

Reaver v1.5.2 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>
mod by t6_x <t6_x@hotmail.com> & DataHead & Soxrok2212

[+] Waiting for beacon from xx:xx:xx:xx:xx:xx
[+] Switching wlan2mon to channel 1
[+] Switching wlan2mon to channel 11
[+] Switching wlan2mon to channel 2
[+] Switching wlan2mon to channel 11
[+] Switching wlan2mon to channel 3
[+] Switching wlan2mon to channel 11
[!] WARNING: Failed to associate with xx:xx:xx:xx:xx:xx

soxrok2212, 2016-04-05:
What is the make and model of the target AP? It would be that WPS is not configured but it is enabled. Does it happen on all targets?

Dreadmonkey, 2016-04-05:
I've tried across multiple AP's, the latest one is an old Linksys WAP4400, previous ones i can't remember off the top of my head as I was testing at a friends. WPS is enabled

soxrok2212, 2016-04-05:
Did it work on other targets?

Dreadmonkey, 2016-04-05:
Nope, same results with EAPOL timeouts

soxrok2212, 2016-04-05:
Try an injection test with aireplay
aireplay-no wlan2mon -9

Dreadmonkey, 2016-04-05:
Yep I get an injection is working according to aireplay

soxrok2212, 2016-04-05:
Have you completely updated your Kali install after installation?

Dreadmonkey, 2016-04-05:
I had so now just done a clean install. Still reaver won't associate without running airodump-ng in another terminal, just switches the the right channel but never gets the beacon. Different behavior once associated though in that it hangs immediately after the line: Starting Cracking Session. Pin count... etc

soxrok2212, 2016-04-06:
Try again with -A and fakeauth with aireplay-ng http://www.aircrack-ng.org/doku.php?id=fake_authentication

Dreadmonkey, 2016-04-06:
Sorry for the slow response, I'll try the -A and fakeauth when I get home. However I have manged to get results. It seems reaver will run successfully if I don't run airodump-ng beforehand to get a MAC. I noticed this late last night after a reboot when I already had the reaver attack saved it ran. Afterwards when I wanted the MAC of my other AP and ran airodump-ng to get it and afterwards reaver was failing and injections tests failed.

I'll let you know if I observe anything else but thank you for all your help, you guys are great.

kcdtv, 2016-04-06:
Sorry to interrupt :smiley_cat:   
Correct me if i am wrong but this is how you use airodump-ng
`airodump-ng wlan0mon`
No filter, nothing... this is not a bug and you can use airodump-ng with reaver **BUT** you need to use at least the channel filter (betteer if you use bssid filter too)... like this
`airodump-ng --bssid <mac_adress_AP> --channel (X)X  <interface>`
Otherwise your interface is doing channel hoping and you cannot properly attack with reaver (or bully) 
WPS transaction is short and fast so you really need your wireless device to be focus on the target...
cheers :cat2:

Hunter5711, 2017-12-20:
can anyone help me out ? I have tried all the wifi hacking tools of kali but sadly None of them worked for me.Can anyone tell me that is Hacking Wi-Fi is really possible coz I'm just fed up with these...plzz help

soxrok2212, 2017-12-20:
This repository is not for “hacking” support.

closed 2017-12-20 by soxrok2212