t6x/reaver-wps-fork-t6x #65: Channel 13 not working

earthGavinLee, 2016-06-30:
airmon-ng
PHY Interface Driver Chipset
phy0 wlan0mon ?????? Realtek Semiconductor Corp. RTL8187

reaver -i wlan0mon -b xx:xx:xx:xx:xx:xx -vv -d9 -t9 -c13
[+] Switching wlan0mon to channel 13
[+] Waiting for beacon from xx:xx:xx:xx:xx:xx
[!] WARNING: Failed to associate with xx:xx:xx:xx:xx:xx (ESSID: xxxxxxxxx)

**reaver channel 1 ~11 is ok , airodump-ng channel 13 is ok, only reaver channel 13 not working**

airodump-ng -w xxxxxxxxx -c13 --bssid xx:xx:xx:xx:xx:xx wlan0mon
CH 13 ][ Elapsed: 2 mins ][ 2016-04-05 11:10

BSSID PWR RXQ Beacons #Data, #/s CH MB ENC CIPHER AUTH E
xx:xx:xx:xx:xx:xx -52 40 872 2 0 13 54e. WPA2 CCMP PSK xxxxxxxxx
BSSID STATION PWR Rate Lost Frames Probe

xx:xx:xx:xx:xx:xx xx:xx:xx:xx:xx:xx -1 1e- 0 0 2

/ # **iw reg get**
global
country 00: DFS-UNSET
(2402 - 2472 @ 40), (N/A, 20), (N/A)
(2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
(2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
(5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
(5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
(57240 - 63720 @ 2160), (N/A, 0), (N/A)

global
country 00: DFS-UNSET
(2402 - 2472 @ 40), (N/A, 20), (N/A)
(2457 - 2482 @ 20), (N/A, 20), (N/A), AUTO-BW, PASSIVE-SCAN
(2474 - 2494 @ 20), (N/A, 20), (N/A), NO-OFDM, PASSIVE-SCAN
(5170 - 5250 @ 80), (N/A, 20), (N/A), AUTO-BW
(5250 - 5330 @ 80), (N/A, 20), (0 ms), DFS, AUTO-BW, PASSIVE-SCAN
(5490 - 5730 @ 160), (N/A, 20), (0 ms), DFS, PASSIVE-SCAN
(5735 - 5835 @ 80), (N/A, 20), (N/A), PASSIVE-SCAN
(57240 - 63720 @ 2160), (N/A, 0), (N/A)

strasharo, 2016-06-30:
Type:
`iw list  | grep MHz`

earthGavinLee, 2016-06-30:
iw list | grep MHz
                        * 2412 MHz [1] (20.0 dBm)
                        * 2417 MHz [2] (20.0 dBm)
                        * 2422 MHz [3] (20.0 dBm)
                        * 2427 MHz [4] (20.0 dBm)
                        * 2432 MHz [5] (20.0 dBm)
                        * 2437 MHz [6] (20.0 dBm)
                        * 2442 MHz [7] (20.0 dBm)
                        * 2447 MHz [8] (20.0 dBm)
                        * 2452 MHz [9] (20.0 dBm)
                        * 2457 MHz [10] (20.0 dBm)
                        * 2462 MHz [11] (20.0 dBm)
                        * 2467 MHz [12] (20.0 dBm) (no IR)
                        * 2472 MHz [13] (20.0 dBm) (no IR)
                        * 2484 MHz [14] (20.0 dBm) (no IR)
                 * short GI for 40 MHz

earthGavinLee, 2016-06-30:
thanks.
**solved**
**iw reg set JP**
iw list | grep MHz
                        * 2412 MHz [1] (20.0 dBm)
                        * 2417 MHz [2] (20.0 dBm)
                        * 2422 MHz [3] (20.0 dBm)
                        * 2427 MHz [4] (20.0 dBm)
                        * 2432 MHz [5] (20.0 dBm)
                        * 2437 MHz [6] (20.0 dBm)
                        * 2442 MHz [7] (20.0 dBm)
                        * 2447 MHz [8] (20.0 dBm)
                        * 2452 MHz [9] (20.0 dBm)
                        * 2457 MHz [10] (20.0 dBm)
                        * 2462 MHz [11] (20.0 dBm)
                        * 2467 MHz [12] (20.0 dBm)
                        * 2472 MHz [13] (20.0 dBm)
                        * 2484 MHz [14] (20.0 dBm)
                 * short GI for 40 MHz

kcdtv, 2016-06-30:
Why do you open a second issue? 
THIS IS NOT A GENERAL HELP FORUM BUT A GITHUB REPOSITORIES
Use kali ofrum or one of the many website about linux to ask foir this kind of questions.

closed 2016-06-30 by kcdtv