t6x/reaver-wps-fork-t6x #67: Kali 2016.1 reaver issues

chazzy1656, 2016-07-10:
hi guys im using kali 2016.1 with everything ou of the box aircrack 1.2 RC3 reaver 1.5.3 etc.. my wifi adapter is a intel 5100 agn with monitor on and packet injection tested with aireplay. But every time i try reaver i keep receiving the M1-M2 message then nothing its says receive time out occur then i tried wifite with no succes at all the only thing that seemed to work was the WPs pins attack but i had 0/12000 success/ttl with no percentage on the side of it so im guessing its not working...then i tries the pixie dust attact but does exactly the same as reaver i kill every pid before airmon-ng to get the wlan0mon on even try with mdk3 attack but it says the device seems to be unvunerable so my guess is that the connection drop somewhere and i cannot get the M3 - M4 M5 m6 and so the get the eapol correctly even wash -i tells me that i got a lot of fcs bad packets so guys plz help me here im trying the same wifi as i used to with ubuntu 12.04 but back then i had the ath9 im really desesperate here

chazzy1656, 2016-07-10:
Update : i tried downgrading the reaver 1.4- 1.3 and then back to 1.5.3 with the libpcap0.8 older but with no luck at all now the tcpdump doesnt work so no pixie dust attack no more

chazzy1656, 2016-07-10:
I tried all the possible command with reaver but still the same message timeout wps failed and cannot pass futher the m2 message

chazzy1656, 2016-07-10:
I only that the process goes faster when i use aireplay to send association but no luck on any m3m4m5 eapol messages

Vegeta0000, 2016-07-10:
Hi. 
I just wondering why you haven't receive list of AP with WPS enabled.
Wash -i <mon> -C ?
Used to have problems with intel 5200 on BackTrack 5. Maybe you are too fare from source AP ?

Vegeta0000, 2016-07-10:
Obviously wash -i monitor -C.

soxrok2212, 2016-07-11:
Intel's drivers for monitor mode are horrible, that is likely your first problem. Second, I'm going to assume the router is yours so I would check inside the router's config and make sure WPS is configured. I have had this happen on an old AP I have where WPS is enabled, but not configured so it stops the protocol after M2 (which is pointless to even go that far IMO).

chazzy1656, 2016-07-11:
Wash -i show me that wps is not protected and is version 1.0

chazzy1656, 2016-07-11:
Yes its enabled and the TX power is -32 so im pretty close to it

soxrok2212, 2016-07-11:
Read my response again.

chazzy1656, 2016-07-11:
Ok is there a way to bypass this? Cause i even tried the mdk3 attack mode and it says that the device seem to be unvulnerable so do i have to move on or there is a solution?

chazzy1656, 2016-07-11:
Thx for your response i mean everybody is looking for a answer for that "issue" but i guess it isnt one since the ap doesnt push the m3 m4 m5 m6 messages only Nacks then drop the connection

soxrok2212, 2016-07-11:
There is nothing to "bypass" here. Either your card does not properly support injection (I've had problems on newer Intel cards but didn't even make it as far as you) or there is a configuration error on the AP. Either that or there's a ton of interference but you seem too close for that to be an issue. There is really no easy way to diagnose a problem. Perhaps try Bully https://github.com/aanarchyy/bully

This may work for you, code is a bit more optimized.

chazzy1656, 2016-07-11:
Thx alot hope people see this :)

soxrok2212, 2016-07-11:
And by the way, FCS issues are usually a hint that theres a driver issue/poor monitor support, at least in my (somewhat limited) testing.

closed 2016-07-11 by soxrok2212

chazzy1656, 2016-07-11:
I tried this morning with another AP around me that i dont know and worked like a charmee so i guess its really a problem with the wps activation or security with the forst AP i was trying

chazzy1656, 2016-07-11:
And no bully does the same as reaver hangs on receiving the M3/M5 message then drop the connection

Thread19, 2022-08-02:
Bro ,
       Please tell me how did you downgrade your reaver to 1.4, please...!!