t6x/reaver-wps-fork-t6x #68: Error in `reaver': double free or corruption, crashes once in a few hours

ptkovsky, 2016-07-24:
Hello,

Host is MC965 with 10.11.6, guest is Kali, hypervisor is Fusion. Monitor is AR7010+AR9287. I did all the get update and upgrade stuff.

It pops up only on this specific AP, so I'm guessing this should not be system related? Not sure. AP seems shitty, despite having decent RSSI it fails to process WPS transactions properly and works very slow, so I'm dancing with -d. It's older Linksys with Atheros CPU, but even high-end Linksys routers are overpriced so don't buy them. I also ended up using -a instead of many other ones here.

Anyway, in my case Reaver dies working like this once in a few hours and it's kinda annoying. Some output looks like this:

```
[+] Received M1 message
[P] R-Nonce: e5:89:5c:b2:9c:49:4b:c6:af:1a:98:0e:b9:aa:d2:d4
[P] PKR: 58:d6:88:39:73:dd:64:dd:bd:4e:61:44:73:44:95:ae:f2:e8:c8:0d:37:5d:ec:1d:52:4d:8a:e2:40:09:19:8a:cb:2b:2a:74:8e:9f:28:d4:70:a0:fd:40:43:cf:27:7c:04:8e:c6:5c:f2:1f:26:c8:39:d1:80:47:dc:13:3e:74:59:7c:e2:b9:aa:87:0f:f6:34:57:40:d3:88:07:41:60:e2:66:04:34:1c:7d:e6:06:ae:af:d4:7f:df:c5:a7:18:73:7b:b8:f3:0e:b1:84:87:0b:1c:eb:ef:6a:19:3e:22:c4:40:d1:1a:05:08:b9:76:9e:d8:ce:c7:95:e9:09:29:90:06:5b:23:b9:6d:ba:91:87:9c:ff:c8:96:00:a7:05:24:a5:6a:5b:f6:9f:71:66:db:ee:ba:08:4f:13:4b:e7:74:1e:eb:23:ac:81:3f:69:9c:8f:85:46:14:f1:d8:b3:d1:f2:e8:72:e9:e6:a1:ff:26:e9:bb:9e:00:73:95:5b
[P] AuthKey: 47:f8:bf:53:af:bd:9b:25:f9:5a:cd:70:a5:89:ac:a6:f1:9e:3a:53:63:5c:c6:37:43:52:fc:d4:14:12:b2:a4
[+] Sending M2 message
[P] E-Hash1: 24:78:12:27:a7:d3:9b:91:06:d3:76:2d:66:77:9e:5f:96:e5:c3:07:ff:1b:ea:fb:ff:9f:87:33:3f:17:91:a7
[P] E-Hash2: d0:f9:ee:4a:4a:2f:79:67:ab:8f:09:41:8c:0e:b6:2a:90:21:df:98:ed:b6:35:ce:bf:68:88:8c:b6:ba:77:8c
[+] Received M3 message
[+] Sending M4 message
[+] Received WSC NACK
[+] Sending WSC NACK
[+] p1_index set to 2964
[+] Pin count advanced: 2964. Max pin attempts: 11000
[+] 26.95% complete. Elapsed time: 0d6h46m42s.
[+] Estimated Remaining time: 3d3h53m44s
[+] Trying pin 29575671.
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[P] E-Nonce: 4c:e0:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[P] PKE: 0d:e9:4d:dc:ad:1d:4c:20:22:a0:da:16:40:0f:58:f5:31:b4:bb:69:0d:c8:cf:04:4b:14:0f:e7:a2:b5:5d:7d:f5:c5:3f:7e:2d:fe:f7:e2:06:05:bf:6c:a5:f4:de:4b:a2:2e:5d:f1:62:a2:fd:12:b1:7a:a4:ef:76:58:58:63:ea:c2:5e:5d:ff:52:86:67:92:88:b3:21:2b:b8:66:df:0b:b4:84:80:d2:2f:de:39:84:55:1e:0c:42:a0:39:6f:42:86:dd:86:69:19:8b:5a:8c:69:46:28:b4:1d:95:60:6f:13:94:43:57:e8:4c:e9:bd:e1:0a:ed:b5:04:bd:a6:19:91:de:2a:a0:e1:39:e3:2c:b0:9f:a3:4f:55:7e:f9:16:5c:f6:97:05:1d:ec:f2:80:ee:15:96:00:e4:22:bf:19:c1:7e:79:0a:e9:d3:a3:e5:59:f0:c2:2f:89:ce:1b:41:48:c7:7d:bc:43:ca:f2:f0:05:73:0c:1d:0d:50:a5
[P] WPS Manufacturer: Linksys by Cisco
[P] WPS Model Name: WRT160NL
[P] WPS Model Number: v1.0.02
[P] Access Point Serial Number: 5
[+] Received M1 message
[P] R-Nonce: 42:43:1c:a2:41:c6:9b:db:78:af:7a:6f:72:0e:f7:22
[P] PKR: c9:9f:eb:3c:89:15:c8:b2:57:b7:c0:84:6a:1f:19:c8:e2:e6:c6:24:09:38:85:45:13:ad:c3:03:ba:94:74:16:89:98:a6:9a:13:67:03:ab:f9:40:dc:b1:73:e4:45:a7:9d:d2:c8:95:48:29:f4:6c:74:a0:47:e0:18:01:f0:80:47:78:ae:82:ae:50:85:0e:33:04:12:bb:84:6d:4d:04:5b:cb:ee:ab:c4:1a:e4:b8:83:56:e1:10:a8:30:94:de:2b:51:25:0e:19:22:3f:89:72:3a:78:9b:c7:cc:9d:f8:5d:ae:a5:91:40:58:f3:1c:71:0a:bc:bd:62:7f:6f:9e:5c:aa:5e:74:20:06:89:7b:4c:6d:79:2f:31:f2:4b:eb:5d:84:76:e7:39:e3:d4:65:2f:7e:37:e6:93:9e:f1:ac:d0:c6:30:2e:f8:70:0a:bd:f1:45:e3:1f:75:26:62:48:61:3a:10:b6:43:49:18:b8:16:4a:b7:37:a3:88:93:fa
[P] AuthKey: d2:73:40:10:a5:66:b9:c0:5d:b0:8d:de:b6:0f:8c:a1:92:bc:dc:13:8d:d0:8d:e6:73:01:cf:b5:70:02:f2:ef
[+] Sending M2 message
[P] E-Hash1: 20:d1:65:38:14:9d:36:12:bb:7c:f6:22:b9:58:b7:b0:fc:c2:4b:37:27:2b:ef:0b:e4:63:af:f9:e9:ad:46:6f
[P] E-Hash2: 57:1b:ee:55:69:84:c5:26:df:d2:38:21:b8:5d:59:47:01:46:37:bd:30:81:0d:cf:ce:e5:2f:2a:75:e9:b7:77
[+] Received M3 message
[+] Sending M4 message
[!] WARNING: Receive timeout occurred
[+] Sending WSC NACK
[!] WPS transaction failed (code: 0x02), re-trying last pin
[+] Trying pin 29575671.
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[P] E-Nonce: 38:c1:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[P] PKE: 50:0f:29:93:e5:d2:8a:25:15:96:14:c6:33:9d:e2:5e:57:0e:77:4c:79:98:31:08:ae:fa:3e:15:4e:91:8e:3d:1e:26:04:d2:90:83:e9:62:71:05:a5:22:c5:bb:31:8d:b5:cc:73:6c:40:b1:fb:ff:a1:fc:76:c1:2b:76:30:1b:40:a9:40:d3:5d:01:42:7a:18:4a:4e:5c:7f:80:3d:26:2b:0e:bc:1c:c7:93:9b:94:58:8c:b6:cd:f3:b6:be:31:74:56:e5:72:54:2f:0d:32:b3:7d:32:0a:7f:ee:47:1f:65:f5:84:2b:8a:12:aa:62:a4:6b:f6:d4:89:c5:4a:6f:3c:e9:44:0b:35:1a:70:af:69:bd:be:b1:f0:c0:cb:ab:d6:8f:6b:20:5d:74:ec:77:c7:b8:e6:ab:1d:1b:7d:41:e9:e2:b6:1c:58:99:a6:5b:69:88:05:5a:de:02:e4:90:76:e1:ea:99:11:e1:a1:1b:77:1a:38:c9:9b:52:81:b3
[P] WPS Manufacturer: Linksys by Cisco
[P] WPS Model Name: WRT160NL
[P] WPS Model Number: v1.0.02
[P] Access Point Serial Number: 5
[+] Received M1 message
[P] R-Nonce: 0b:9c:ad:70:dd:1d:4e:67:e1:bf:85:ff:3f:77:5a:d0
[P] PKR: 79:6b:98:5d:77:66:47:4a:80:79:d5:17:17:50:49:9e:ff:d0:0a:d2:b1:92:76:93:ed:77:c9:cf:4c:7a:d5:e6:b7:64:aa:67:85:e2:29:41:63:7e:12:9a:99:d8:e0:1e:20:6d:78:d1:27:e5:11:44:42:23:64:ec:74:95:0a:59:4f:e8:75:89:01:ca:d8:b3:03:ac:e0:54:b6:72:e4:d4:fa:6c:1d:e4:1e:01:6d:c2:31:a9:f9:22:fa:a2:7b:93:80:d4:be:66:46:a4:b0:02:b7:0f:13:fb:15:e4:89:db:ae:b1:a4:92:4d:c7:25:92:97:39:ee:9a:22:27:74:f9:dc:6a:93:60:06:e5:47:16:9e:cb:7a:5f:ad:72:63:a9:ee:49:27:d0:31:ab:5a:f5:dd:34:27:57:6c:61:97:e6:08:2b:65:77:1f:9c:1b:53:68:e4:26:3a:7c:55:5b:a1:7d:6c:73:ce:c3:19:3f:44:1f:f6:10:a8:04:eb:57:52
[P] AuthKey: 7a:51:7e:43:9e:27:11:29:c0:81:9e:22:93:d5:4d:9f:03:64:51:20:4e:0b:8a:0e:6b:95:cc:73:84:53:af:47
[+] Sending M2 message
[P] E-Hash1: cb:2a:6b:eb:2e:5d:cb:35:d0:94:9d:7e:fa:61:4c:7c:fe:32:5a:5b:38:49:a7:41:6f:58:72:66:ec:0b:68:fe
[P] E-Hash2: c3:26:f0:10:50:b7:40:b4:dc:6f:98:dd:42:56:66:9c:d1:2f:f5:18:73:2e:2f:0d:20:7b:b7:d3:b7:73:24:2c
[+] Received M3 message
[+] Sending M4 message
[!] WARNING: Receive timeout occurred
[+] Sending WSC NACK
[!] WPS transaction failed (code: 0x02), re-trying last pin
*** Error in `reaver': double free or corruption (!prev): 0x00000000010fb640 ***
======= Backtrace: =========
[... 65 lines cut ...]
[+] Switching wlan0mon to channel 5
[+] p1_index set to 2964
[+] p2_index set to 0
[+] Restored previous session
[+] Waiting for beacon from test_bssid
[+] Associated with test_bssid (ESSID: Test)
[+] Starting Cracking Session. Pin count: 2964, Max pin attempts: 11000
[+] Trying pin 29575671.
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[P] E-Nonce: 10:67:00:00:00:00:00:00:00:00:00:00:00:00:00:00
[P] PKE: 96:25:d9:80:40:9d:49:6f:7f:74:9e:fb:fa:f7:f5:0f:74:d3:2e:0c:85:1d:25:95:2e:78:84:ba:33:ad:80:fb:36:a4:78:33:e2:b0:69:74:c9:b8:dc:5a:b1:b7:7a:72:09:0f:07:87:49:ef:84:47:e2:f3:5a:55:9e:a7:d5:0a:b8:3b:73:27:e4:63:73:b1:bc:90:41:d1:f2:4f:aa:c1:a6:33:7e:af:6e:28:49:e4:77:c1:b2:ea:6c:da:fd:31:6a:4a:7a:4c:68:21:89:2a:ed:3f:fe:f2:87:bf:ec:39:28:02:58:b5:68:21:f9:20:12:99:cc:91:44:a9:9f:ee:20:25:ca:5b:7c:32:0e:a3:0c:ad:f0:45:8d:ce:dd:8c:09:95:57:d9:35:07:bb:bf:fc:d6:08:f9:ab:2a:17:ea:97:d4:ab:2a:4a:16:e4:77:2f:51:9f:4b:e1:5f:fc:0a:27:a7:48:54:af:2b:0e:f4:49:8e:03:12:cf:f6:d1:58
[P] WPS Manufacturer: Linksys by Cisco
[P] WPS Model Name: WRT160NL
[P] WPS Model Number: v1.0.02
[P] Access Point Serial Number: 5
[+] Received M1 message
[P] R-Nonce: c7:08:5a:e7:6b:46:7e:a2:5a:66:2b:2d:85:b6:5d:31
[P] PKR: d2:49:81:37:ec:74:0c:e1:5d:29:91:ed:20:8e:45:4f:98:ea:33:e9:a5:e5:ec:cf:ee:f7:e8:3e:05:39:04:cc:c8:ef:6a:da:5b:51:fc:0a:b2:a6:54:19:d7:96:9d:20:4c:48:61:a3:1a:8c:3e:b4:0f:6a:7c:c7:e2:24:de:b3:3b:fb:4a:f5:fc:82:bf:0d:75:54:50:be:fa:3d:96:b2:33:fd:59:6f:d6:d2:73:e6:1d:c2:2a:a6:60:1a:ba:1b:5f:57:26:4d:30:5d:10:d8:43:9e:66:26:c3:f3:47:35:36:7b:53:f7:13:77:f7:86:cb:b6:3f:e6:88:a1:86:ce:5e:21:ef:d9:56:d4:c9:ef:82:5f:b8:3c:6a:d5:f3:b0:bc:2c:67:30:74:3b:56:4c:ff:cb:f2:7c:55:2b:38:c8:b4:87:24:2a:fe:09:c0:57:f2:e7:7c:59:00:54:48:95:b1:60:c6:c3:69:22:39:09:e8:a8:05:ad:32:c2:bf:cc
[P] AuthKey: d3:bf:c6:16:75:42:34:96:64:09:33:02:89:9c:f2:ff:83:d8:c4:1a:af:b7:65:e5:56:88:a9:fd:af:41:df:59
[+] Sending M2 message
[P] E-Hash1: 3a:0d:02:55:8b:7c:d4:3c:b7:c9:bf:5e:d8:15:0e:b8:4f:71:e5:d2:0d:26:f2:7f:a9:7b:2f:b1:4f:91:fe:f1
```
Is this even reaver related?

ptkovsky, 2016-08-17:
It's gone for now. Was looking like a system issue anyway. Closing this atm.

ptkovsky, 2016-08-21:
Was roaming around today and got same error with different Atheros AP. Now I'm kinda confused. So this issue is not really AP, or Broadcom specific. Going back to the software issue theory for now. Can be related firmware or setup issues too. Will be looking forward to any helpful comments about this funky adventures.

soxrok2212, 2016-08-21:
Not sure about the first error you got, but as for the model number, there is no bug in Reaver. The model number is sent in both Probe Responses from the AP and in the WPS MX messages. You can verify this in Wireshark. Just out of curiosity, have you tried completely removing reaver from your system (including the prepackaged version included with Kali) and re-downloading/recompiling? Not sure if the problem is coming from having 2 different versions of Reaver installed.

ptkovsky, 2016-09-01:
Hello again. Thank you for your reply and for the tip, @soxrok2212. I did remove the original version and its dependencies and reinstalled the newest version. Seems like it did the trick and is working perfectly fine for now. I was testing it for about a week and didn't find any problems except for default segmentation fault errors.

closed 2016-09-01 by soxrok2212