t6x/reaver-wps-fork-t6x #71: Reaver does not handle bSSID 00:00:00:00:00:00

kcdtv, 2016-08-09:
This is a bit unrealistic but the code should handle it.
Executing the recommended minimal syntax with 00:00:00:00:00:00 for target bSSID we get in the shell the command description, as if we had a syntax error.
`sudo reaver -i wlan0mon -b 00:00:00:00:00:00 -vvv -K 1
[sudo] password for kcdtv: 
Reaver v1.5.3 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>
mod by t6_x <t6_x@hotmail.com> & DataHead & Soxrok2212 & Wiire & AAnarchYY & KokoSoft
Required Arguments:
-i, --interface=<wlan>          Name of the monitor-mode interface to use
-b, --bssid=<mac>               BSSID of the target AP
Optional Arguments:
-m, --mac=<mac>                 MAC of the host system
-e, --essid=<ssid>              ESSID of the target AP
-c, --channel=<channel>         Set the 802.11 channel for the interface (implies -f)
-o, --out-file=<file>           Send output to a log file [stdout]
-s, --session=<file>            Restore a previous session file
-C, --exec=<command>            Execute the supplied command upon successful pin recovery
-D, --daemonize                 Daemonize reaver
-a, --auto                      Auto detect the best advanced options for the target AP
-f, --fixed                     Disable channel hopping
-5, --5ghz                      Use 5GHz 802.11 channels
-v, --verbose                   Display non-critical warnings (-vv for more)
-q, --quiet                     Only display critical messages
-K  --pixie-dust=<number>       [1] Run pixiewps with PKE, PKR, E-Hash1, E-Hash2 and E-Nonce Ralink, Broadcom & Realtek)
-Z, --no-auto-pass              Do NOT run reaver to auto retrieve WPA password if Pixiewps attack is successful
-h, --help                      Show help
Advanced Options:
-p, --pin=<wps pin>             Use the specified 4 or 8 digit WPS pin
-d, --delay=<seconds>           Set the delay between pin attempts [1]
-l, --lock-delay=<seconds>      Set the time to wait if the AP locks WPS pin attempts [60]
-F, --fake-delay=<seconds>      Set the time to sleep after received fake NACK [0]
-R, --fake-reason=<nack reason> Specifies the reason code for NACK used in the FAKE NACK
-I, --ignore-reason             Ignore change of reason code for NACK
-g, --max-attempts=<num>        Quit after num pin attempts
-x, --fail-wait=<seconds>       Set the time to sleep after 10 unexpected failures [0]
-r, --recurring-delay=<x:y>     Sleep for y seconds every x pin attempts
-t, --timeout=<seconds>         Set the receive timeout period [5]
-T, --m57-timeout=<seconds>     Set the M5/M7 timeout period [0.20]
-A, --no-associate              Do not associate with the AP (association must be done by another application)
-N, --no-nacks                  Do not send NACK messages when out of order packets are received
-S, --dh-small                  Use small DH keys to improve crack speed
-L, --ignore-locks              Ignore locked state reported by the target AP
-E, --eap-terminate             Terminate each WPS session with an EAP FAIL packet
-n, --nack                      Target AP always sends a NACK [Auto]
-w, --win7                      Mimic a Windows 7 registrar [False]
-X, --exhaustive                Set exhaustive mode from the beginning of the session [False]
-1, --p1-index                  Set initial array index for the first half of the pin [False]
-2, --p2-index                  Set initial array index for the second half of the pin [False]
-P, --pixiedust-loop            Set into PixieLoop mode (doesn't send M4, and loops through to M3) [False]
-W, --generate-pin              Default Pin Generator [1] Belkin [2] D-Link [3] Zyxel
-H, --pixiedust-log             Enables logging of sequence completed PixieHashes
Example:
reaver -i wlan0mon -b 00:90:4C:C1:AC:21 -vvv -K 1`

vk496, 2016-08-15:
Yet, it do it.

https://github.com/t6x/reaver-wps-fork-t6x/blob/7ff9107e93dca53cfaa22c620f65234868ab9a9e/src/wpscrack.c#L76-L80

If we have NULL MAC, exit

Salu2

kcdtv, 2016-08-31:
I guess it is better to leave it that way than to change this and break something else :tongue:

closed 2016-08-31 by kcdtv