t6x/reaver-wps-fork-t6x #75: Compile failed

chinalantian, 2016-08-29:
China@LanTian:~/Down/reaver-wps-fork-t6x-master/src$ make
common.c: In function ‘pingen_zhaochunsheng’:
common.c:597:14: warning: passing argument 1 of ‘snprintf’ makes pointer from integer without a cast [-Wint-conversion]
     snprintf(pin, pin_len, "%08d", (default_pin * 10) + wps_checksum(default_pi
              ^
In file included from /usr/include/features.h:367:0,
                 from /usr/include/stdlib.h:24,
                 from includes.h:25,
                 from common.c:15:
/usr/include/i386-linux-gnu/bits/stdio2.h:61:1: note: expected ‘char * restrict’ but argument is of type ‘int’
 __NTH (snprintf (char *__restrict __s, size_t __n,
 ^
common.c: In function ‘pingen_zyxel’:
common.c:614:25: warning: implicit declaration of function ‘wps_pin_checksum’ [-Wimplicit-function-declaration]
     return (pin * 10) + wps_pin_checksum(pin);
                         ^
In file included from bignum.c:21:0:
libtommath.c: In function ‘mp_exptmod’:
libtommath.c:580:9: warning: variable ‘dr’ set but not used [-Wunused-but-set-variable]
     int dr;
         ^
tlsv1_server_read.c: In function ‘tls_process_client_key_exchange_rsa’:
tlsv1_server_read.c:474:9: warning: variable ‘encr_len’ set but not used [-Wunused-but-set-variable]
     u16 encr_len;
         ^
In file included from ../utils/common.h:18:0,
                 from md5-internal.c:17:
md5-internal.c: In function ‘MD5Final’:
md5-internal.c:191:29: warning: argument to ‘sizeof’ in ‘memset’ call is the same expression as the destination; did you mean to dereference it? [-Wsizeof-pointer-memaccess]
     os_memset(ctx, 0, sizeof(ctx)); /* In case it's sensitive */
                             ^
../utils/os.h:413:41: note: in definition of macro ‘os_memset’
 #define os_memset(s, c, n) memset(s, c, n)
                                         ^
wps_dev_attr.c: In function ‘wps_process_manufacturer’:
wps_dev_attr.c:200:26: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
             printf("%c", (char *) str[pixiecnt]);
                          ^
wps_dev_attr.c:200:20: warning: format ‘%c’ expects argument of type ‘int’, but argument 2 has type ‘char *’ [-Wformat=]
             printf("%c", (char *) str[pixiecnt]);
                    ^
wps_dev_attr.c: In function ‘wps_process_model_name’:
wps_dev_attr.c:241:26: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
             printf("%c", (char *) str[pixiecnt]);
                          ^
wps_dev_attr.c:241:20: warning: format ‘%c’ expects argument of type ‘int’, but argument 2 has type ‘char *’ [-Wformat=]
             printf("%c", (char *) str[pixiecnt]);
                    ^
wps_dev_attr.c: In function ‘wps_process_model_number’:
wps_dev_attr.c:280:26: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
             printf("%c", (char *) str[pixiecnt]);
                          ^
wps_dev_attr.c:280:20: warning: format ‘%c’ expects argument of type ‘int’, but argument 2 has type ‘char *’ [-Wformat=]
             printf("%c", (char *) str[pixiecnt]);
                    ^
wps_dev_attr.c: In function ‘wps_process_serial_number’:
wps_dev_attr.c:319:26: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
             printf("%c", (char *) str[pixiecnt]);
                          ^
wps_dev_attr.c:319:20: warning: format ‘%c’ expects argument of type ‘int’, but argument 2 has type ‘char *’ [-Wformat=]
             printf("%c", (char *) str[pixiecnt]);
                    ^
wps_dev_attr.c:344:59: warning: implicit declaration of function ‘mac2str’ [-Wimplicit-function-declaration]
   printf("[Pin Gen] Pin Generated : %08d\n",pingen_belkin(mac2str(get_bssid(),'
                                                           ^
wps_dev_attr.c:344:59: warning: passing argument 1 of ‘pingen_belkin’ makes pointer from integer without a cast [-Wint-conversion]
In file included from ../wps.h:40:0,
                 from ../defs.h:46,
                 from ../globule.h:38,
                 from wps_dev_attr.c:15:
[... 236 lines cut ...]
         ^
iwlib.c:3058:13: warning: called from here [-Winline]
         if((iw_set_ext(skfd, ifname, SIOCSIWSCAN, &wrq) < 0)
             ^
In file included from iwlib.c:14:0:
iwlib.h:522:9: warning: inlining failed in call to ‘iw_get_ext’: call is unlikely and code size would grow [-Winline]
         iw_get_ext(int   skfd,  /* Socket to the kernel */
         ^
iwlib.c:3082:8: warning: called from here [-Winline]
     if(iw_get_ext(skfd, ifname, SIOCGIWSCAN, &wrq) < 0)
        ^
ar: `u' 修饰符被忽略，因为 `D' 为默认（参见 `U'）
libwps.c: In function ‘parse_wps_parameters’:
libwps.c:29:12: warning: variable ‘data_len’ set but not used [-Wunused-but-set-variable]
     size_t data_len = 0, offset = 0;
            ^
wpscrack.c: In function ‘main’:
wpscrack.c:134:13: warning: ignoring return value of ‘system’, declared with attribute warn_unused_result [-Wunused-result]
             system(get_exec_string());
             ^
Failure to stop here, can not continue

operating system:Ubuntu 16.04.1
already have build-essential libpcap-dev sqlite3 libsqlite3-dev aircrack-ng  installed
https://github.com/wiire/pixiewps Download ,  already have Pixiewps install

kcdtv, 2016-08-30:
Hi!
Reaver is designed to be installed in any GNU-Linux distribution :cactus:
I have it installed in Ubuntu 16.04
You need **libpcap0-8.dev** too 
Install it and try again, it should work.

chinalantian, 2016-08-31:
Hi!
Before Compilation libpcap0.8 has been installed , Should not be its relationship .
China@LanTian:~/Down/reaver-wps-fork-t6x-master/src$sudo apt search libpcap
libpcap-dev/xenial,now 1.7.4-2 all [Installed]
  development library for libpcap (transitional package)
libpcap0.8/xenial,now 1.7.4-2 i386 [Installed，automatic]
  system interface for user-level packet capture
libpcap0.8-dev/xenial,now 1.7.4-2 i386 [Installed，automatic]
  development library and header files for libpcap0.8

kcdtv, 2016-08-31:
I don't really know what is going wrong, i have it working in 16.04 
Did you install aircrack-ng from ubuntu repositories or from aircrack-ng repositories? 
The version form ubutnu is outdated. 
I recommend you to uninstall aircrack-ng and its dependencies, than to start form the scratch, 
follow this steps: 
[https://forum.aircrack-ng.org/index.php/topic,1326.msg4199.html#msg4199](url)
than 
`sudo apt-get install libpcap0.8-dev `
(if you removed it) 
and run the installation of reaver... It should work. (it works for me) 
Cheers

chinalantian, 2016-09-01:
According to the steps mentioned above, the same mistake. 
The strange is that no matter whether the Ubuntu comes with or Sources installation Aircrack-ng are compiled failed, did not complete.
But all generated wash reaver file.

kcdtv, 2017-01-19:
I just tried in Ubuntu 16.04 y 16.10 now you just need to intsall libcap-dev that will install libcap0.8-dev automatically (like in Kali)

rofl0r, 2017-06-22:
i think any compile issues have been fixed in the 1.6b release.

closed 2017-06-22 by rofl0r