t6x/reaver-wps-fork-t6x #78: Pixiewps pin not found why ?

5485, 2016-09-16:
Pixiewps pin not found why?

in this file report of reaver

http://rgho.st/86HYVbB9B

http://rgho.st/8LD7qMBBC

I would like to understand why it does not return me the pin...thank

5485, 2016-09-16:
Reaver v1.5.2 WiFi Protected Setup Attack Tool
Copyright (c) 2011, Tactical Network Solutions, Craig Heffner <cheffner@tacnetsol.com>
mod by t6_x <t6_x@hotmail.com> & DataHead & Soxrok2212 & Wiire & kib0rg

[?] Restore previous session for E0:60:66:EE:1B:FB? [n/Y] y
[+] p1_index set to 34
[+] p2_index set to 0
[+] Restored previous session
[+] Waiting for beacon from E0:60:66:EE:1B:FB
[+] Switching mon0 to channel 1
[+] Switching mon0 to channel 2
[+] Switching mon0 to channel 3
[+] Switching mon0 to channel 6
[+] Associated with E0:60:66:EE:1B:FB (ESSID: TIM-76160310)
[+] Starting Cracking Session. Pin count: 34, Max pin attempts: 11000
[+] Trying pin 00235679.
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[P] E-Nonce: 8b:73:ed:fe:15:bb:73:a9:1f:2d:eb:e3:99:42:54:10
[P] PKE: cf:e5:54:6d:86:94:fb:07:1a:ae:c2:9b:5e:3b:4f:c7:60:a3:07:50:b2:ec:91:14:7a:c6:f4:51:9b:3e:20:5c:d2:2b:e4:9c:ff:60:ba:03:9e:f1:67:10:d5:48:e9:46:39:ab:77:fb:5b:52:29:bd:e2:e9:ca:5c:49:d3:c2:88:91:55:23:17:0e:66:b6:29:d0:79:8d:00:da:c4:30:bc:08:f1:63:62:f1:41:8c:e7:4e:0d:c8:63:a3:d9:93:8d:d0:6d:29:16:d8:3b:72:0e:27:58:f6:86:19:f2:9f:48:99:da:b0:67:b2:66:8a:c7:35:fd:7b:f9:10:d2:fe:54:cf:3a:8a:95:5a:a7:d4:45:67:a1:fe:84:5e:eb:3f:09:14:02:e9:88:e2:35:55:8c:1f:1d:79:be:c7:30:5d:62:ca:22:0a:b8:c0:b2:03:df:c1:ee:e1:31:c7:31:21:f4:aa:d2:d0:a6:e1:51:a7:a8:15:5f:25:b6:9c:bb:65:5e
[P] WPS Manufacturer: Broadcom
[P] WPS Model Name: Broadcom
[P] WPS Model Number: 123456
[P] Access Point Serial Number: 1234
[+] Received M1 message
[P] R-Nonce: fe:cd:e8:74:92:72:6a:8f:5c:26:ce:48:eb:8c:00:73
[P] PKR: 61:b7:88:4f:78:62:2e:c0:f3:35:ba:66:4e:ab:cd:8a:6c:f4:35:e0:d5:83:43:ef:a5:07:fe:da:ac:91:e9:16:66:45:8f:80:08:f4:fb:ac:fd:ce:46:55:3c:5c:6b:a0:89:af:52:b5:a3:2e:cc:ff:5f:fa:8e:7d:3e:37:2f:bf:9a:c9:ca:5c:a3:b3:7a:f9:00:46:fe:2e:11:85:45:bd:fb:28:15:bb:3b:7c:7f:58:e1:f8:59:1e:13:42:7a:3f:d2:1c:87:f2:3f:1a:67:65:cb:6f:38:1f:71:07:d0:1a:f0:c9:28:2e:5f:61:5a:0a:7a:1e:8f:31:2a:59:d1:4e:f9:06:38:6a:49:89:8f:df:62:5a:50:8a:0a:2a:15:e8:94:54:9c:aa:b0:b4:5a:82:4d:4e:40:b4:9b:84:c8:67:92:c8:03:6b:0d:fd:f8:f2:b2:6b:cc:c3:43:15:8e:9e:1c:e0:3f:28:44:db:c8:2a:b1:75:1f:22:a0:93:c1:0b
[P] AuthKey: 2c:27:8c:40:d1:74:5c:0e:9c:4d:6e:e9:3f:6a:0b:bd:8f:79:60:f4:90:91:fb:c7:0b:a2:52:3b:5c:e6:19:b2
[+] Sending M2 message
[P] E-Hash1: 68:fe:11:6f:10:24:52:af:aa:a6:05:9c:c4:88:b9:22:82:93:db:eb:d7:32:2d:3a:f1:23:1a:bb:03:11:86:c9
[P] E-Hash2: be:9e:dc:30:a2:98:d4:2e:55:ef:a3:71:dd:26:cd:89:5b:63:3a:28:0e:37:e8:ea:b3:b0:ae:52:df:ff:d8:c8
[+] Received M3 message
[+] Sending M4 message
[+] Received WSC NACK
[+] Sending WSC NACK
[+] p1_index set to 35
[+] Pin count advanced: 35. Max pin attempts: 11000
[+] Trying pin 00245678.
[+] Sending EAPOL START request
[!] WARNING: Receive timeout occurred
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[P] E-Nonce: fc:66:fb:dd:4e:be:2e:ed:d8:03:87:29:db:43:71:4c
[P] PKE: 1e:8f:64:58:3d:98:70:a3:e2:f8:49:44:8f:2c:76:2f:cb:d2:8f:01:89:f3:b1:9a:f6:f2:53:d8:91:26:5f:bf:d4:1b:0d:dc:7a:e7:51:a3:f6:c9:45:90:7c:84:37:4e:d0:e3:18:93:2b:e2:9f:f8:94:37:a9:ed:13:54:db:2b:34:9e:aa:78:0a:35:16:3a:a0:e0:13:32:8c:cc:8b:3d:1b:00:91:f6:d7:6f:1b:2f:6a:79:74:e2:ec:76:1a:9a:28:25:1e:39:4a:bd:29:35:36:fc:00:2a:65:b6:56:a6:ce:42:47:1f:69:61:31:ad:e7:69:65:f2:85:a8:dd:35:4a:f5:0c:91:d4:d4:5a:18:1b:4f:74:b6:c0:6b:48:4c:98:ed:b2:d0:b1:16:b6:4c:bb:b7:cf:bd:6e:7f:6d:8c:a8:f1:60:82:bc:ac:fb:be:b6:74:30:06:d2:2e:80:cd:e3:bb:9a:9e:f8:69:66:f1:da:97:30:71:1c:7c:32:f7
[P] WPS Manufacturer: Broadcom
[P] WPS Model Name: Broadcom
[P] WPS Model Number: 123456
[P] Access Point Serial Number: 1234
[+] Received M1 message
[P] R-Nonce: f5:b2:bf:c4:5e:06:68:04:c0:8c:5a:c1:f6:58:27:60
[P] PKR: c2:68:61:2c:22:b3:1c:de:9c:04:0d:9b:77:86:c8:b8:80:95:15:86:3d:39:22:7b:22:e5:d4:fb:57:06:8c:70:7d:8a:c0:36:61:45:f7:42:9f:a5:46:ba:df:96:7f:b7:c1:04:bb:29:be:bb:ca:e0:dc:be:f5:0f:c1:36:f1:b1:24:81:10:68:c5:8d:d1:33:a9:f5:de:c4:89:a3:35:2e:95:a0:af:ab:7e:be:d7:8f:53:05:14:d3:8b:8f:34:6a:97:e0:de:62:ff:44:11:e4:5a:43:8d:22:03:76:9e:86:12:08:ec:71:6a:35:a7:4a:9d:00:d6:b6:45:8a:58:0d:75:97:6b:d6:09:44:8c:aa:57:20:28:14:4d:65:05:70:d5:61:9b:e2:3f:e2:88:ce:03:4e:b4:5e:b4:5c:7d:0d:27:d4:87:3d:06:ed:81:37:80:c1:bb:57:57:70:34:90:01:95:26:8f:99:77:fc:12:df:a9:f1:be:c9:a1:47:6a
[P] AuthKey: e4:cf:cd:d6:c7:d7:15:90:1c:d1:e4:d7:d3:ad:04:a4:e7:38:2e:de:45:12:40:50:1e:40:ec:67:af:56:f3:94
[+] Sending M2 message
[P] E-Hash1: 35:c3:32:b8:f1:51:d8:af:a5:55:3f:fc:bb:34:85:bc:4a:32:89:06:9d:6f:5c:6c:4a:e3:ca:1b:b7:f5:4e:e3
[P] E-Hash2: 9a:e7:bf:da:fa:30:72:77:44:db:cc:fa:9d:63:4a:02:ee:0d:e0:be:6f:29:94:66:c2:9e:b3:18:09:73:c0:80
[+] Received M3 message
[+] Sending M4 message
[+] Received WSC NACK
[+] Sending WSC NACK
[+] p1_index set to 36
[+] Pin count advanced: 36. Max pin attempts: 11000
[+] Trying pin 00255677.
[+] Sending EAPOL START request
[+] Received identity request
[+] Sending identity response
[P] E-Nonce: 27:3f:c1:e3:8e:43:01:08:70:1d:53:58:06:23:e9:86
[P] PKE: f7:6b:51:07:f0:ee:42:a3:74:e1:0d:1e:fe:b0:4b:ca:b3:2f:45:dc:0a:d6:4e:d3:aa:21:45:bd:9f:cf:b2:d6:f4:c2:44:cc:7f:9b:01:1c:95:b7:38:2a:7b:df:3d:e3:98:9d:60:2a:04:b0:19:51:b9:99:71:ca:ab:d3:c0:dc:fc:b4:48:3c:3c:5e:3b:5c:48:95:24:28:de:a0:d1:9e:1e:45:35:10:b2:ba:8d:7e:58:bb:d1:3c:e3:d6:be:dc:42:0b:18:e6:b5:1d:e2:0c:50:b6:3b:21:fb:e6:09:b9:2c:9f:bf:ff:1a:2e:64:09:46:ab:03:c6:a8:aa:29:38:01:1c:ab:78:83:1e:a2:49:36:02:2b:da:cd:e4:c6:60:af:77:d2:1a:fa:fa [... cut ...]
[P] AuthKey: fd:ed:0c:96:50:e7:50:4e:e9:5e:61:31:68:17:d0:3e:ee:f7:b1:6d:db:c8:b1:ab:be:02:cb:78:24:cc:f8:89
[+] Sending M2 message
[P] E-Hash1: 73:83:d2:96:90:74:26:6f:48:6f:28:0b:38:5f:71:2f:2a:8d:8e:5c:be:15:a1:0c:c0:e8:2e:ed:b9:53:7a:13
[P] E-Hash2: cc:06:28:87:c6:c7:bb:30:39:c9:ca:19:52:1c:aa:62:72:e1:2e:3a:a4:48:c2:68:5d:5c:1b:70:fe:29:76:39
[+] Received M3 message
[+] Sending M4 message
[+] Received WSC NACK
[+] Sending WSC NACK
[+] p1_index set to 37
[+] Pin count advanced: 37. Max pin attempts: 11000

5485, 2016-09-16:
I don't have M5 but I need?

5485, 2016-09-16:
Sorry I was wrong section ....
May I ask how do I unlock the router ?I use  revdk3 r1 but the router remains in lock state

closed 2016-09-16 by soxrok2212

kcdtv, 2016-09-16:
For the record: 

> Pixiewps pin not found why?

-You did not launch a pixiewps attack so you cannot say that "pixiewps" did not recover the PIN :money_with_wings: 
-Just a few broadcom chipset are supported, check the exact model of the chipset from your access point :shipit: 
-if you cannot know which chipset has got your access point it means that this is not your acces point and we do not provide for that :oncoming_police_car: 
-Last but not least: This is a github repository for **reaver**, not a general forum about WPS or revdk3 revdk4 or revdk5

5485, 2016-09-16:
Yes man i use this information of reaver and i use pixiewps off line because the router get to ap limit.
If you take the information of  reaver pke , pkr ecc... and you use pixiewps it s dont found wps pin

soxrok2212, 2016-09-16:
Then it's not vulnerable. Simple as that.

5485, 2016-09-16:
Each attack the infornation must be the same ? Example pke , pkr ect....in a attack i have different pke pke.....

5485, 2016-09-16:
Bad notice soxrok2212 i think that it possibile but i don t good man... :-(
Thank thank all

soxrok2212, 2016-09-16:
This is not a general support forum. And since you can't prove you own the AP or have permission to test I can't help you.

5485, 2016-09-16:
Ok...sorry for this post

warbd, 2017-08-06:
you can see this. its worked for me 
https://www.youtube.com/watch?v=leqBKAEqI1Q