t6x/reaver-wps-fork-t6x #80: This reaver fork has a problem detecting WPS networks...

NovaCygni, 2016-09-28:
Main reaver detects WPS perfectly fine with Wifite, this Reaver seems to have a issue even detecting then, any idea why this is? Hotpatching Wifite to force the WPS also gives no joy as the reaver refuses to work with the WPS networks.

Tested on numerous cards, using Archlinux 4.7.6 and 4.4.6 kernels.

soxrok2212, 2016-09-28:
Which version of Wifite are you using?

NovaCygni, 2016-09-28:
Version 2.0, a.k.a Wifite2 the latest version thats on dervs github page, I think you guys might be behind a update or something has been overlooked due to lack of bug reports on the issue... 
As it stands the latest version of Wifite which is more than a few months old no longer works with this version of reaver which is a shame. Was wondering if there was a known cause for the problem that I could look into and hotpatch myself.

soxrok2212, 2016-09-28:
The version you linked should work, Kali 2 comes default with it. You can try this alternatively: 
 https://github.com/aanarchyy/wifite-mod-pixiewps

closed 2016-09-28 by soxrok2212