t6x/reaver-wps-fork-t6x #81: Add new d-link WPS PIN generation algorithm

lorek123, 2016-10-01:
Here is code for algorithm: https://pierrekim.github.io/advisories/quanta-wps-gen.c
more details:
https://pierrekim.github.io/blog/2016-09-28-dlink-dwr-932b-lte-routers-vulnerabilities.html

kcdtv, 2016-10-01:
Thanks for your suggestion
But read again the full disclosure...., this generator is useless.
The device has static default PIN and this generator is just here as a POC to illustrate how the PIN is generated when the user choose to use the option "generate a new PIN"
The time is used as a seed.
So everytime you will use this generator it will be.... too late
The generator gives you a different reuslt every seconds and you can be sure that when you see the router with wps .activated, the PIN you will generate is not the one that was generated 
Sorry for my English, i am not not sure this is very clear, read again he full disclosure, try the generator a couple of time  ans you will understand what I mean. 
cheers

soxrok2212, 2016-10-01:
@kcdtv is right, the generator is just for the web GUI for users to generate a new pin. Just use the hardcoded pin of 28296607.

closed 2016-10-01 by soxrok2212

kcdtv, 2016-10-02:
By the way; I am pretty sure that this device uses  a Ralink chipset that is  is vulnerable to  pixiewps :smiley_cat:  
So even if the PIN has been changed you will get it after a single try...

soxrok2212, 2016-10-02:
I actually tried looking it up but I am very confident that it is Ralink as well.