t6x/reaver-wps-fork-t6x #82: Compiled failed on Freebsd 11.0

Martinfx, 2016-10-09:
Hello i compiled clang FreeBSD clang version 3.8.0 (tags/RELEASE_380/final 262564) (based on LLVM 3.8.0)
Target: x86_64-unknown-freebsd11.0, os FreeBSD  11.0-RELEASE-p1 but i have problem with compilation.
I instaled libpcap, sqlite3 ok now run configure

```
checking for gcc... no
checking for cc... cc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether cc accepts -g... yes
checking for cc option to accept ISO C89... none needed
checking for pcap_open_live in -lpcap... no
error: pcap library not found!
exit: Illegal number: -1
```
Why, i have installed library and headers. Ok maybe bad paths ..
`CC=clang CFLAGS=-I/usr/local/include LDFLAGS=-L/usr/local/lib ./configure`

```
Compilation is ok but i have warnings 
checking for string.h... (cached) yes
checking pcap.h usability... yes
checking pcap.h presence... yes
checking for pcap.h... yes
checking sqlite3.h usability... yes
checking sqlite3.h presence... no
configure: WARNING: sqlite3.h: accepted by the compiler, rejected by the preprocessor!
configure: WARNING: sqlite3.h: proceeding with the compiler's result
checking for sqlite3.h... yes
configure: creating ./config.status
config.status: creating Makefile
```
[config.txt](https://github.com/t6x/reaver-wps-fork-t6x/files/518325/config.txt)

Do you have any idea ?
thanks :-)

Martinfx, 2016-10-09:
Now compiling with gcc version gcc version 4.8.5 (FreeBSD Ports Collection) 
configure is good but make is  worse.
```
make: "/usr/home/ra/Documents/reaver-wps-fork-t6x/src/Makefile" line 9: Missing dependency operator
make: "/usr/home/ra/Documents/reaver-wps-fork-t6x/src/Makefile" line 12: Need an operator
make: "/usr/home/ra/Documents/reaver-wps-fork-t6x/src/Makefile" line 13: Missing dependency operator
make: "/usr/home/ra/Documents/reaver-wps-fork-t6x/src/Makefile" line 16: Need an operator
make: Fatal errors encountered -- cannot continue
make: stopped in /usr/home/ra/Documents/reaver-wps-fork-t6x/src
```

Martinfx, 2016-10-09:
Ok now i have problem with 
                                                ^
```
33 warnings generated.
cc  -I../utils -I ../ wps_ufd.c -c
gmake[1]: Leaving directory '/usr/home/ra/Documents/reaver-wps-fork-t6x/src/wps'
(cd libwps && gmake)
gmake[1]: Entering directory '/usr/home/ra/Documents/reaver-wps-fork-t6x/src/libwps'
gcc -Wall libwps.c -c
In file included from libwps.c:14:0:
libwps.h:50:1: error: unknown type name 'u_char'
 int parse_wps_parameters(const u_char *packet, size_t len, struct libwps_data *wps);
 ^
libwps.h:129:1: error: unknown type name 'u_char'
 int parse_wps_tag(const u_char *tags, size_t len, struct libwps_data *wps);
 ^
libwps.h:130:1: error: unknown type name 'u_char'
 unsigned char *get_wps_data(const u_char *data, size_t len, size_t *tag_len);
 ^
libwps.h:131:1: error: unknown type name 'u_char'
 unsigned char *get_wps_data_element(const u_char *data, size_t len, uint16_t type, size_t *el_len);
 ^
libwps.h:132:1: error: unknown type name 'u_char'
 int libwps_has_rt_header(const u_char *packet, size_t len);
 ^
libwps.h:133:1: error: unknown type name 'u_char'
 const u_char *libwps_radio_header(const u_char *packet, size_t len);
 ^
libwps.h:133:1: error: unknown type name 'u_char'
libwps.c:26:1: error: unknown type name 'u_char'
 int parse_wps_parameters(const u_char *packet, size_t len, struct libwps_data *wps)
 ^
libwps.c: In function 'parse_wps_parameters':
libwps.c:28:5: error: unknown type name 'u_char'
     const u_char *data = NULL;
     ^
libwps.c:29:12: warning: variable 'data_len' set but not used [-Wunused-but-set-variable]
     size_t data_len = 0, offset = 0;
            ^
libwps.c: At top level:
libwps.c:55:1: error: unknown type name 'u_char'
 int parse_wps_tag(const u_char *tags, size_t len, struct libwps_data *wps)
 ^
libwps.c: In function 'parse_wps_tag':
libwps.c:89:13: warning: passing argument 1 of 'get_wps_data_element' from incompatible pointer type [enabled by default]
             el = get_wps_data_element(wps_ie_data, wps_data_len, elements[i], &el_len);
             ^
In file included from libwps.c:14:0:
libwps.h:131:16: note: expected 'const int *' but argument is of type 'unsigned char *'
 unsigned char *get_wps_data_element(const u_char *data, size_t len, uint16_t type, size_t *el_len);
                ^
libwps.c: At top level:
libwps.c:190:1: error: unknown type name 'u_char'
 unsigned char *get_wps_data(const u_char *data, size_t len, size_t *tag_len)
 ^
libwps.c:234:1: error: unknown type name 'u_char'
 unsigned char *get_wps_data_element(const u_char *data, size_t len, uint16_t type, size_t *el_len)
 ^
libwps.c:269:1: error: unknown type name 'u_char'
 int libwps_has_rt_header(const u_char *packet, size_t len)
 ^
libwps.c:291:1: error: unknown type name 'u_char'
 const u_char *libwps_radio_header(const u_char *packet, size_t len)
 ^
libwps.c:291:1: error: unknown type name 'u_char'
libwps.c: In function 'libwps_radio_header':
libwps.c:299:17: error: 'u_char' undeclared (first use in this function)
         return (u_char *) FAKE_RADIO_TAP_HEADER;
                 ^
libwps.c:299:17: note: each undeclared identifier is reported only once for each function it appears in
libwps.c:299:25: error: expected expression before ')' token
         return (u_char *) FAKE_RADIO_TAP_HEADER;
                         ^
libwps.c:302:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
gmake[1]: *** [Makefile:5: libwps.o] Error 1
gmake[1]: Leaving directory '/usr/home/razor/Documents/reaver-wps-fork-t6x/src/libwps'
gmake: *** [Makefile:27: libwps.o] Error 2

```

kcdtv, 2016-10-12:
All the errors point out to **libwps.c**, **libwps.h** and **libwps.o.** 
My guess is that the _libpcap's_ you have installed are the cause of the problem
Unfortunately I do not know freeBSSD
In debian' based distribution you specifically need to install an additional development package called **libpcap0.8-dev**  
Look for something similar for your system or a way to install the same libraries.

Martinfx, 2016-10-12:
Hello i will look for the problem on FreeBSD and if i fix this problem i will send pull request

rofl0r, 2016-11-27:
should be fixed with one of the commits in #89 (the one about u_char)

rofl0r, 2016-11-29:
my pr was merged. can you try again, @Martinfx ?

rofl0r, 2016-11-29:
@kcdtv is there a specific reason you recommend libpcap version 0.8 instead of the actual version, which would be 1.8.1 ?

kcdtv, 2016-11-29:
Hi there! 
I am speaking about the package called in debian distibutions  **libpcap0.8-dev**
The name is **very** confusing, i think that they could have picked something different....
If you check in details it is not outdated:

>  dpkg -l | grep libpcap
ii  libpcap-dev                               1.8.1-3                              all          development library for libpcap (transitional package)
ii  libpcap0.8:amd64                          1.8.1-3                              amd64        system interface for user-level packet capture
ii  libpcap0.8-dev                            1.8.1-3                              amd64        development library and header files for libpcap0.8
ii  python-pcapy                              0.10.8-1+b1                          amd64        Python interface to the libpcap packet capture library

Reaver requires this **libpcap0.8-dev** which is version 1.8.1-3 (like the others libcap stuff)

Martinfx, 2016-12-02:
Now i have problem with linux files.
```
(cd lwe && gmake BUILD_STATIC=y libiw.a)
gmake[1]: Entering directory '/usr/home/razor/Documents/reaver-wps-fork-t6x/src/lwe'
cp wireless.21.h wireless.h
gcc -Os -W -Wall -Wstrict-prototypes -Wmissing-prototypes -Wshadow -Wpointer-arith -Wcast-qual -Winline -I. -MMD     -fPIC -c -o iwlib.so iwlib.c
In file included from iwlib.h:43:0,
                 from iwlib.c:14:
/usr/include/net/if_arp.h:89:18: error: field 'arp_pa' has incomplete type
  struct sockaddr arp_pa;  /* protocol address */
                  ^
/usr/include/net/if_arp.h:90:18: error: field 'arp_ha' has incomplete type
  struct sockaddr arp_ha;  /* hardware address */
                  ^
In file included from iwlib.c:14:0:
iwlib.h:54:53: fatal error: linux/types.h: No such file or directory
 #include <linux/types.h>  /* for "caddr_t" et al  */
                                                     ^
compilation terminated.
gmake[1]: *** [Makefile:120: iwlib.so] Error 1
gmake[1]: Leaving directory '/usr/home/razor/Documents/reaver-wps-fork-t6x/src/lwe'
gmake: *** [Makefile:52: libiw] Error 2
```

Martinfx, 2016-12-02:
I send pull request for fix data types on FreeBSD and for configure file but now i have problem with struct iwreq
```
gcc -Os -W -Wall -Wstrict-prototypes -Wmissing-prototypes -Wshadow -Wpointer-arith -Wcast-qual -Winline -I. -MMD     -fPIC -c -o iwlib.so iwlib.c
In file included from iwlib.c:14:0:
iwlib.h: In function 'iw_set_ext':
iwlib.h:532:25: error: 'struct iwreq' has no member named 'ifr_name'
             strncpy(pwrq->ifr_name, ifname, IFNAMSIZ);
                         ^
iwlib.h: In function 'iw_get_ext':
iwlib.h:548:25: error: 'struct iwreq' has no member named 'ifr_name'
             strncpy(pwrq->ifr_name, ifname, IFNAMSIZ);
                         ^
iwlib.h: In function 'iw_broad_ether':
iwlib.h:592:49: error: 'ETH_ALEN' undeclared (first use in this function)
             memset((char *) sap->sa_data, 0xFF, ETH_ALEN);
                                                 ^
iwlib.h:592:49: note: each undeclared identifier is reported only once for each function it appears in
iwlib.h: In function 'iw_null_ether':
iwlib.h:603:49: error: 'ETH_ALEN' undeclared (first use in this function)
             memset((char *) sap->sa_data, 0x00, ETH_ALEN);
                                                 ^
iwlib.c: In function 'iw_sockets_open':
iwlib.c:156:26: error: 'AF_AX25' undeclared (first use in this function)
         AF_INET, AF_IPX, AF_AX25, AF_APPLETALK
                          ^
iwlib.c: In function 'iw_set_basic_config':
iwlib.c:758:20: error: 'struct iwreq' has no member named 'ifr_name'
         strncpy(wrq.ifr_name, ifname, IFNAMSIZ);
                    ^
iwlib.c:758:9: warning: passing argument 1 of 'strncpy' from incompatible pointer type [enabled by default]
         strncpy(wrq.ifr_name, ifname, IFNAMSIZ);
         ^
In file included from iwlib.h:28:0,
                 from iwlib.c:14:
/usr/include/string.h:101:7: note: expected 'char *' but argument is of type 'const int *'
 char *strncpy(char * __restrict, const char * __restrict, size_t);
       ^
iwlib.c: In function 'iw_get_stats':
iwlib.c:1275:20: error: 'struct iwreq' has no member named 'ifr_name'
         strncpy(wrq.ifr_name, ifname, IFNAMSIZ);
                    ^
iwlib.c:1275:9: warning: passing argument 1 of 'strncpy' from incompatible pointer type [enabled by default]
         strncpy(wrq.ifr_name, ifname, IFNAMSIZ);
         ^
In file included from iwlib.h:28:0,
                 from iwlib.c:14:
/usr/include/string.h:101:7: note: expected 'char *' but argument is of type 'const int *'
 char *strncpy(char * __restrict, const char * __restrict, size_t);
       ^
iwlib.c: In function 'iw_check_mac_addr_type':
iwlib.c:1940:21: error: 'SIOCGIFHWADDR' undeclared (first use in this function)
     if((ioctl(skfd, SIOCGIFHWADDR, &ifr) < 0) ||
                     ^
iwlib.c:1940:5: warning: passing argument 2 of 'ioctl' makes integer from pointer without a cast [enabled by default]
     if((ioctl(skfd, SIOCGIFHWADDR, &ifr) < 0) ||
     ^
In file included from /usr/include/sys/ioctl.h:45:0,
                 from iwlib.h:21,
                 from iwlib.c:14:
/usr/include/sys/ioccom.h:75:5: note: expected 'long unsigned int' but argument is of type 'const int *'
 int ioctl(int, unsigned long, ...);
     ^
iwlib.c:1941:18: error: 'struct ifreq' has no member named 'ifr_hwaddr'
             ((ifr.ifr_hwaddr.sa_family != ARPHRD_ETHER)
                  ^
iwlib.c:1941:29: error: request for member 'sa_family' in something not a structure or union
             ((ifr.ifr_hwaddr.sa_family != ARPHRD_ETHER)
                             ^
iwlib.c:1941:40: warning: comparison between pointer and integer [enabled by default]
             ((ifr.ifr_hwaddr.sa_family != ARPHRD_ETHER)
[... 51 lines cut ...]
/usr/include/string.h:101:7: note: expected 'char *' but argument is of type 'const int *'
 char *strncpy(char * __restrict, const char * __restrict, size_t);
       ^
iwlib.c:2262:25: error: 'SIOCGARP' undeclared (first use in this function)
         if((ioctl(skfd, SIOCGARP, &arp_query) < 0) ||
                         ^
iwlib.c:2262:9: warning: passing argument 2 of 'ioctl' makes integer from pointer without a cast [enabled by default]
         if((ioctl(skfd, SIOCGARP, &arp_query) < 0) ||
         ^
In file included from /usr/include/sys/ioctl.h:45:0,
                 from iwlib.h:21,
                 from iwlib.c:14:
/usr/include/sys/ioccom.h:75:5: note: expected 'long unsigned int' but argument is of type 'const int *'
 int ioctl(int, unsigned long, ...);
     ^
iwlib.c: In function 'iw_process_scan':
iwlib.c:3046:17: error: 'ETIME' undeclared (first use in this function)
         errno = ETIME;
                 ^
iwlib.c:3046:15: warning: assignment makes integer from pointer without a cast [enabled by default]
         errno = ETIME;
               ^
gmake[1]: *** [Makefile:120: iwlib.so] Error 1

```

rofl0r, 2016-12-02:
that looks pretty broken then. look at Makefile.in, there's some stuff under `ifdef Darwin` or `ifdef MACOSX` oslt, maybe it'll work when you put `FreeBSD` in there (it seems the OS X ifdef prevents compilation of one subdir).

rofl0r, 2017-06-22:
we pushed out a new release: can you try if it works on freebsd ?

Martinfx, 2017-06-23:
This is good news but now not works on bsd.

rofl0r, 2017-06-23:
oh yeah, sorry. i forgot that the big-endian fix commit used a linux-specific header for the __le16_to_cpu() etc macros... i'll open another issue for that.

rofl0r, 2017-06-23:
opened #145.

rofl0r, 2017-11-26:
@Martinfx with #145 closed, can you try if it works now ? (current master pls)
if there are further issues, please report them here

Martinfx, 2017-11-26:
Thank you for fix endian issues now is only : 
```
cc -DCONF_DIR='"/usr/local/var/lib/reaver"' -Wall -g -O2 -Wno-unused-variable -Wno-unused-but-set-variable -Wno-unused-function -Wno-pointer-sign -Ilibwps -I. -Ilwe -DCONFIG_IPV6   -c -o 80211.o 80211.c
warning: unknown warning option '-Wno-unused-but-set-variable'; did you mean '-Wno-unused-const-variable'?
      [-Wunknown-warning-option]
In file included from 80211.c:34:
In file included from ./80211.h:40:
In file included from ./argsparser.h:40:
In file included from ./iface.h:40:
./lwe/iwlib.h:508:17: error: no member named 'ifr_name' in 'struct iwreq'
  strncpy(pwrq->ifr_name, ifname, IFNAMSIZ);
          ~~~~  ^
./lwe/iwlib.h:524:17: error: no member named 'ifr_name' in 'struct iwreq'
  strncpy(pwrq->ifr_name, ifname, IFNAMSIZ);
          ~~~~  ^
```

rofl0r, 2017-11-26:
build is fixed, now the question is whether it works ;)

closed 2017-11-26 by rofl0r

Martinfx, 2017-11-26:
Thank you M.

rofl0r, 2017-11-26:
were you able to test ? let us know if it works, thanks.

commit e3772a53 (2017-11-26): fix build on freebsd. closes #82