t6x/reaver-wps-fork-t6x #84: installation fails

squiddymute, 2016-10-14:
squid@kali:~/reaver-wps-fork-t6x/src# sudo make install
rm -f /usr/local/bin/reaver /usr/local/bin/wash /usr/local/bin/walsh
if [ ! -d /usr/local/etc/reaver ]; then mkdir -p /usr/local/etc/reaver; fi
if [ ! -e /usr/local/etc/reaver/reaver.db ]; then cp reaver.db /usr/local/etc/reaver/reaver.db && chmod -R a+rw /usr/local/etc/reaver; fi
if [ -e wash ]; then cp wash /usr/local/bin/wash; fi
cp: cannot create regular file '/usr/local/bin/wash': No such file or directory
Makefile:96: recipe for target 'install' failed
make: *** [install] Error 1

kali distro
Linux version 4.3.0-kali1-amd64 (debian-kernel@lists.debian.org) (gcc version 5.3.1 20160121 (Debian 5.3.1-7) ) #1 SMP Debian 4.3.3-7kali2 (2016-01-27)

kcdtv, 2016-11-13:
I just tryed to install in kali rolling updated and no problems... 

> if [ -e wash ]; then cp wash /usr/local/bin/wash; fi
if [ -e reaver ]; then cp reaver /usr/local/bin/reaver; fi

Could you please gives us more details by copying pasting the complete stdout of all the steps you follow to install reaver? 
Did you installed all the dependencies?

xhebox, 2016-12-04:
@kcdtv There's a patch for [reaver](https://raw.githubusercontent.com/alpinelinux/aports/master/testing/reaver-wps-fork-t6x/0200-makefile-path-fix.patch) in upstream of alpinelinux. Hope this is useful.

Ghostboy-287, 2017-02-16:
same thing for me 
[image]

i followed all your steps to install it but i get this error every time 

Os: kali rolling 2016.2 x86

kcdtv, 2017-02-16:
I have just tried tried with Kali...
[image]
Does it works with the patch proposed previously by **xhebox**?

Ghostboy-287, 2017-02-16:
this is the result of `ls /usr/local` and `ls /usr/`
[image]

and we noticed that `/bin/` is not present on `/usr/local` why i had the error: cannot create regular file... No such file or directory
but the `/bin` folder is present on `/usr`

so to fix this i have two options:
1- create a folder `/bin` on `/usr/local`
2-or editing the script by changing `/usr/local/bin` by `/usr/bin`

Ghostboy-287, 2017-02-16:
you can set this as resolved 
it works after creating `bin` folder on `/usr/local`

rofl0r, 2017-02-16:
this is expected behaviour, all autoconf using software default to install to /usr/local, unless you override it by calling `./configure --prefix=/usr`. the reasoning behind is that in usual corporate scenarios stuff in /usr is supplied by the distro package manager, and stuff in /usr/local is what the users compile themselfes.

rofl0r, 2017-02-16:
what is unexpected though is that leading directories are not created. above PR ^ fixes the issue.

closed 2017-02-16 by soxrok2212

kcdtv, 2017-02-20:
Hi! Sorry for my absence, hardcore week....

> noticed that /bin/ is not present on /usr/local

That what I thought... I made a modif adding a line to create with mkdir the directory it it doesn't exist  but rofl0r found a far better solution. 
Great job!

MasterJi24, 2018-03-06:
How to fix

Makefile121: receip for target 'reaver' failed
make: *** [reaver] Error 1

please help me.

rofl0r, 2018-03-06:
@MasterJi24 please open a new issue and provide the lines displayed above that error message.

PR #123 (2017-02-16): Makefile.in: fix behaviour of make install
Makefile didn't honour DESTDIR, which is a standard behaviour,
and it failed to produce leading directories.

should fix #84