t6x/reaver-wps-fork-t6x #85: reaver stuck in [+] Starting Cracking Session for all my routers (even with -vv)

chelinho139, 2016-10-20:
Hey guys reaver is giving me real trouble. I am using the latest version from kali apt-get sources.

My router is WPS ENABLED and I tried several routers, same result. I get stuck and no more prints. Also if I ctrl+c and start again it stats from pin 0.

I have a WN722N the recommended wifi dongle. Running Linux Kali. My commands are:

sudo reaver -i wlan0mon -b xx:xx:xx:xx:xx:xx -vv

[+] Waiting for beacon from xx:xx:xx:xx:xx:xx
[+] Associated with xx:xx:xx:xx:xx:xx (ESSID: xxxxxxxxxx)
[+] Starting Cracking Session. Pin count: 0, Max pin attempts: 11000

It get to there and no more prints. Left it for hours and nothing..

What is this problem? can it be fixed?

Thanks in advanced.

kcdtv, 2016-10-20:
Did you properly configure a PIN?
Are you in decent conditions?
Nothing wrong about your command, nothing wrong about reaver neither with this command it really looks like you are not in good condition for the attack or WPS is not fully enabled.

> I have a WN722N the recommended wifi dongle.

I do recommend it but the best chispet is rtl8187l , it is the most efficient with reaver.
Maybe an issue with injection has it hapened for some ralink with kernel around 4.3 
Do you get correct results preforming an injection test?
`sudo aireplay-ng -9 wlan0mon`  

cheers

chelinho139, 2016-10-20:
Im sorry, what would decent conditions be?

When running sudo aireplay-ng -9 wlan0mon i get:

root@kali:~# aireplay-ng -9 wlan1mon
16:14:44  Trying broadcast probe requests...
16:14:44  Injection is working!
16:14:45  Found 1 AP 

16:14:45  Trying directed probe requests...
16:14:46  00:02:2A:F0:56:D7 - channel: 11 - 'test hack'
16:14:46  Ping (min/avg/max): 1.250ms/10.372ms/22.318ms Power: -53.73
16:14:46  30/30: 100%

soxrok2212, 2016-10-20:
Can we get some information about the target? Are you sure WPS is enabled an configured? Are you sure WPS is not locked?

chelinho139, 2016-10-20:
@soxrok2212 sure,
Information:

WPS: Enabled
Status: Set up
PIN Number: 1280618
Auth: WPA PSK
Connection: Enable PA 802. 11n

I Have attached the screenshots from my router but the info is in spanish.

[b.pdf](https://github.com/t6x/reaver-wps-fork-t6x/files/542844/b.pdf)
[a.pdf](https://github.com/t6x/reaver-wps-fork-t6x/files/542843/a.pdf)

kcdtv, 2016-10-20:
¿What the fuck is this ISP? :smiley_cat: 
The logo looks like the supermarket Eroski...

About conditions: 
  injection is working and you are in good conditions (100% injection, -50dBm, that's perfect) 

About WPS;:
 The PIN is "configurado" -> OK
 WPS is enable -> OK

Maybe you need to reboot the router to fully enable the changes?
With my router i have to reboot it to have the WPS fully operational
I don't know your ISP, it could be, but I doubt it, that they decided to "break" the WPS in an update. 
In Spain some company did it, (jazztell) but it is very unlikely to happen.
I do not believe ( at all) that it is an issue with reaver as i can perfectly run the same command with the same WiFi chipset against my PA and it works without problems.
I believe there is something fishy with your access point.
What is the model? 
As you have the PIN you should try with wpa_cli to connect with it normally. 
Cheers

amazer-bil, 2017-01-07:
@kcdtv...... Hi brother....i can say i have the same issue !

i have latest rooters around me such as ' technicolor and broadcom and dlink and ralink.

when i ran reaver it stuck as ' waiting for beacon from ' the mac address " and after little while it shows to me ' warning about not associated with the rooter i'm trying to hack.
help if that possible and thanks :)

soxrok2212, 2017-03-10:
inactive

closed 2017-03-10 by soxrok2212