t6x/reaver-wps-fork-t6x #86: Timeouts?

pspecter, 2016-10-22:
Guys congratulations for the reaver modification!!!

I have a question/problem that u might know how to solve it. 

I installed Kali on my MacBook and I have been running it and reaver without any problem. I have three Cards atheros ar9002u, rt2870 and the mac internal Broadcom.

Recently I changed internet provider and they gave me a new router. I tried to check if it was secure. I have a signal of -30 and when I try to run reaver usually happens to create a cycle between sending identity response and failing replying with the message that a timeout as occurred. It never comes to the point of sending/receiving M1 messages.

I have been googling it and I have tried reaver with the -A and aireplay at the same time, with the -d 15 -t 60 and even 100 with the -N and -n and even -a or -S. 

I also tried reaver with some friends and the funny thing happens. Different internet providers reaver works like a charm. The same internet provider as I have but different routers, reaver keeps having the same behaviour. Is it possible that the internet provider developed something that forbids the routers of sending M1 messages???

Does any of you have a solution for this problem?

kcdtv, 2016-10-22:
Hi **pspecter** :smiley_cat:  !
  To answer your question  _"Is it possible that the Internet provider developed something that forbids the routers of sending M1 messages???"_ 
  Although there is  not technically "counter measure that forbids the routers of sending M1 messages" , It is usual that  ISP take measures to disable-break WPS in PIN mode.
 For example jazztell in Spain did an update to all their firmware remotely and WPS that was enabled with PIN 12345670 is now "broken" and the option to activate or disactivate PIN is removed from the administration interface. They basically disabled the PIN  mode by not configuring any PIN 
So you will maybe get some M1 and M2 message but northing more. 
And there is no way for the user to change that (no more WPS related option in the interface)   

If a PIN is not properly configured or the PIN mode is disabled there is nothing that reaver  can do. 
  You will never be able to properly check a single first half of PIN.
 Even if you get an M3, you will never go further and the strings that you got are useless for pixiewps 
As you can imagine there is not any method known to configure remotely a router with raw WiFipacket injection...    Never say never but i don't see that happens tomorrow, or the day after...

   If you are not 100% sure about if you are safe or not,  check carefully your web interface, 
With reaver use the **default PIN with -p option**, don't forget to use maximum verbose (**-vvv**)
  Try with wpa_cli in managed mode
  Or directly ask your ISP: You want to connect with the PIN provided. You cannot. What is going on?

pspecter, 2016-10-22:
Thanks kcdtv for taking the time to answer to me so fast. 

If reaver is not a solution what would you advise?

kcdtv, 2016-10-22:
This is not really a good place to ask this kind of questions, 
This github branch is to speak about reaver focusing on code related aspects.
My advices are:
  1) Carefully check your web interface to check if WPS is enabled and if a PIN is configured
   2) Once this is checked (you can double check with wash) launch reaver with the default PIN and full verbose 
`sudo reaver -i <interface_mode_monitor> -b <bssid_target> -p <PIN_target> -vvv`
  3) If you cannot get the PIN you can try again in managed mode with wpa_cli or from windows, use the PIN when you are pormpted too (you need to delete the profile, PIN is proposed at first connection)
  If you tell us first what is going at this 3 stages  we could eventually answer you. 
cheers

pspecter, 2016-10-28:
I found my answer. the routers from my isp have wps-pbc that is why reaver was getting timeouts in getting the m1 messages. 

thanks anyway!

closed 2016-10-28 by kcdtv